schreibfaul1/ESP32-MiniWebRadio #451: Recent Changes in common.h

cschwick, 2024-06-09:
Dear Wolle, 

Recently there was a change in common.h (class dlnaList). This class had a function "hasClicked" in which the variable m_browseOnRelease was set (for example to '3' if the user clicked on a folder). 
Now I see that this functionality has been moved to "hasReleased". 

Why am I asking? 

I was just implementing a change for browsing the DLNA server so that the user could long-press on a folder and this would then on the fly create a playlist and play it. (I like listening to entire albums which I have organised in tracks in a folder on my server...) My change was just working (still some rough corners...) and I wanted to prepare it to propose it to you, however, when moving to the latest version in the master branch I saw the recent change mentioned above. With the change above the m_browseOnRelease is '0' in the longReleased callback function which I added to the dlnaList class. I was trying to check in this function if the long press/release of the user was done on a folder, and if so the creation of the On-the-fly playlist was triggered. 

I might find a work-around, but was there a specific reason to move this functionality to the onRelease function? (...I guess so... but I do not see it since I am missing the overview....) 

Thanks a lot, Chris

cschwick, 2024-06-10:
Hi Wolle, I am trying your recent changes concerning long-pressing objects in the dlna list:

It works great if I press objects in a list which are not the FIRST item in the list (e.g. not the first song in a folder, not the first folder in a list of folders). But If I long-press the first item in the list I get a crash: one of these:
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
The crash comes before the message about the long press is printed out.

The reason for the crash seems to be that if you longpress on the first item in a subfolder, the m_itemListPos is calculated to be '0', but in the check for a file the vector m_srvContent.itemURL[m_itemListPos - 1] is indexed effectively with -1 which leads to a read of a "forbidden" memory region. 

Does it work flawless in your case?

cschwick, 2024-06-10:
Dear Wolle, 

The first version of the dlna-folder-player is in a forked github repo:
https://github.com/cschwick/ESP32-MiniWebRadio-fork/tree/dlna-directory-play
You have to use the branch: dlna-direactory-play
If I long-click on a folder in my dlna server the radio plays the contents (and shows the titles & the number in the playlist)

However, there are flaws: I cannot change the bottom row of buttons to play the next/previous song. If I stop the playing the Player goes to the default playlist of the SD card. This needs further work so it sticks to the playlist created on the fly. But already like this it is usable (to some extent)...

Best wishes Chris

schreibfaul1, 2024-06-11:
Of course the array index must not be negative, [m_itemListPos - 1] is the problem, the -1 is rubbish.
I'll have a look at your fork in the next few days.

schreibfaul1, 2024-06-11:
I have adopted your suggestions, and added two small changes:
1) the currently selected entry turns magenta if you press and hold it,
2) it waits until the DLNA client has the complete result and then the playlist is created. 

This is not yet perfect, but it works. Many thanks for your help.

closed 2024-07-26 by github-actions[bot]