schreibfaul1/ESP32-MiniWebRadio #452: Just a few things

appollokillers1, 2024-06-10:
i have a few things that i must say about this project,talking about the 3.1 version,first : you must check how many months you have actually in the shorten months,i count 11,not 12,august is missing,second :why did you actually chosed to make the updates available as OTA ? isn't much easier thru the usb ?,many for sure have problems with upload an eventual update thru OTA,myself too,we don't know how, third : why didn't you used for ili9341 as driver bodmer's tft_espi driver ? for me is working without problems,but only when i activate 1 important line from user_setup.h file,it is at the end of it :  #define SUPPORT_TRANSACTIONS,if i comment that line,nothing is showing on the screen,fourth : what is the point of having all those defined for tft pins and audio pins if they are not taken automaticaly in the other parts,u put your own pins in the common.h,and no matter of what i have put it there,in the tft.h has other that i have to change it too and for the sound ,the same thing (for this part i had to dig deep to find out),i love the project,even tho it didn't worked for me,maybe it was my esp32s3 board that i have (lilygo t7s3) that for my surprise it even connect at internet at all,it wasn't showing in my router,and i know i checked a few times if i put the right information for that,and i found something else deep in the code,there is a main.h or cpp in the DLNA_Client,after building the project,where i found something,there were 2 lines : String ssid =     "Wolles-FRITZBOX"; and String password = "40441061073895958449";,just under WiFiMulti wifiMulti; line,is that supposed to be there ?

schreibfaul1, 2024-06-13:
I have made some corrections that you have pointed out. I'm not quite sure what you mean by #define SUPPORT_TRANSACTIONS and setup.h.

appollokillers1, 2024-06-15:
i was talking about the bodmer's tft-espi driver,it has a file user_setup.h where is that line,if i use hes driver,i have to activate it to make my screen actually show anything,don't ask me what it does or not,but for my screen is working,i didn't yet manage to look into the program of yet,as i don't have another kit bought,is not available yet,the first as i said befour,i can't access it anymore after i uploaded the code with the radio,as is not conecting to my router,i thing it is the chip foulty,and i can't upload another program through the usb either,windows don't see it and neither linux,i tried with both.On the chip it does say n16r8 but i have no ideea how to make it to be seen again in windows or linux again,so i have to wait to buy a new 1 and test the new version.

schreibfaul1, 2024-06-16:
If
````
Connecting .............
A fatal error occured:
````
appears in the terminal during the upload, it may be that the last upload did not work and the ESP32 is busy with itself and is blocking.
If you manage to get the ESP to query the serial interface while the points are being written in the terminal, the ESP could work again. The boot routine must fall within the time of the ......., which usually requires several attempts. By this I mean the time between pressing the upload button in VSCode and the reset button on the ESP32

appollokillers1, 2024-06-24:
The thing is that i didn't got any errors at uploading,neither after,when i conect it to the pc,i hear the new conected sound and after 1 sec the disconected sound,but that's it,and i use linux for this things 'coz in linux i can conect and siconect the divice as many times i want and is still see it,not like in windows that after 3 times...is not showing anymore,besides the fact that i don't have to install any driver at all :),anyway,i'm going to wait to buy the esp32s3 that u have it in your list,the last in the list as i don't know if the 32 mb esp32s3 is going to work with the program for the 16 mb esp32s3,i mean this 1 : ESP32-S3-DEVKITC-1-N32R8V

schreibfaul1, 2024-06-25:
That's right, under Linux you don't need any additional drivers, but sometimes you need to 'dialout' the user and give them rights, for example: 
````
sudo usermod -a -G dialout $USER
sudo chmod 666 /dev/ttyACM0 
````
You can use 32MB, for this project 4MB is enough.

appollokillers1, 2024-06-29:
yes,you are right,but now with the new version that i have,i don't need to do that anymore,as linux ubuntu does come with CODE installed,so i don't need to do that again,are a few things that need to be done to make it work and i had to search on internet for a long time for the solution,but i managed to make it work,but i didn't got hit about the old problem with not beeing able to conect to a device thru USB as it was befour,and that did made me very happy too :)

closed 2024-08-13 by github-actions[bot]