schreibfaul1/ESP32-MiniWebRadio #453: ESP-IDF based build really needed?

maxgerhardt, 2024-06-11:
When I change the `platformio.ini` from to just use

```ini
platform = espressif32@6.5.0
framework = arduino
```
and not `arduino,espidf`, the only build failure is in `common.h` trying to use `uxTaskGetSystemState()` in the `GetRunTimeStats()` function to get generate the runtime statistics, which is not actiavted in regular Arduino-ESP32. When I change the code to not print any statistics, the code compiles fine, and a lot faster than with a ESP-IDF build.

```cpp
RAM:   [==        ]  22.7% (used 74220 bytes from 327680 bytes)
Flash: [======    ]  55.9% (used 2126109 bytes from 3801088 bytes)
Building .pio\build\esp32\firmware.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 26 ELF sections
Successfully created esp32 image.
=== [SUCCESS] Took 46.74 seconds ===
```

vs
```cpp
==================== [SUCCESS] Took 134.05 seconds ====================
```

These runtime stats are printed only to the serial anyways when the `s` command is received.

It would also accellerate the downloading of the project's dependencies (no more git clone of Arduino-ESP32 2.0.14 from Github, but using official packages instead).

schreibfaul1, 2024-06-12:
The pre-compiled Arduino version is the optimum for versatile applications. It also works with this project. However, the data throughput via WiFi is reduced. This is sufficient for most radio stations.  There are dropouts with large data packets or high bit rates. This is not ideal for a radio project. It is better to make your own settings in LWIP, WIFI and IRAM.

maxgerhardt, 2024-06-12:
Ah interesting! Then that makes sense.