schreibfaul1/ESP32-MiniWebRadio #457: Pin of sleep mode

valera39, 2024-06-16:
Hi Dear Wolle!
Is it possible to allocate a separate pin for ESP32 S3 on which a high level would appear in sleep mode. I would like to use it to cut off the power supply to the audio power amplifier when the radio is asleep. And also it can be used to turn on or off any devices.
Thank you!

valera39, 2024-06-21:
I have the amplifier speakers continue to make a little noise when the radio goes into sleep mode. In this case a separate pin with a high level in sleep mode would really help me a lot.

schreibfaul1, 2024-06-22:
You can assign a pin to AMP_ENABLED for this purpose.

valera39, 2024-06-23:
Hi Dear Wolle!
Thank you very much for realizing the possibility to control the amplifier. But I noticed a small bug, which I hope you will not have difficulties with and can quickly fix.
If only the radio is enabled in the wakeup settings, the high level signal from the AMP_ENABLED pin works fine and turns on the station. But if both the alarm clock and the radio are on, when I wake up, the tune is playing, but I can't hear it because the AMP_ENABLED pin hasn't turned on yet, and it only turns on when the station is turned on.  That is, I would like the high level on the AMP_ENABLED pin to appear, along with the alarm tune
Thanks for your attention and participation!

schreibfaul1, 2024-06-23:
You are right, AMP_ENABLED is set too late. I'll change that soon.

valera39, 2024-06-23:
Thanks, it's working properly now. But I can't set the volume of the alarm melody. I have set different values of "ringvolume" in settings.json file, also deleted this file, but the volume of the alarm clock melody is invariably very quiet, by ear about the second volume level. What am I doing wrong?
And what not to open another topic I want to ask: in the documentation folder of the project there is a file menuconfig.pdf. In the file sdkconfig.esp32s3 according to it parameters are not changed
CONFIG_LWIP_WND_SCALE, CONFIG_LWIP_TCP_RCV_SCALE and another value CONFIG_LWIP_TCPIP_TASK_STACK_SIZE. What is the correct way?

valera39, 2024-06-25:
I also tried changing the alarm volume in the sound editor, but the sound is still quiet.
Maybe there are some other settings?

schreibfaul1, 2024-06-25:
The **_ringvolume** variable has no meaning at the moment (still to come) The ringtone is always played at the maximum volume.

valera39, 2024-06-25:
So if I understand correctly, you are planning to make the ringer louder?
And what do you say about the menuconfig file?

closed 2024-08-09 by github-actions[bot]