schreibfaul1/ESP32-MiniWebRadio #460: BT implementation

ve2cfb, 2024-06-24:
Hello Wolle,

I need your expertise for the modifications on my radio about been able to do a POWER  ON/OFF with a switch. 
In a previous post i did explain the purpose of this Switch.
Here is a Schematic of my setup.
[image]

 I am using GPIO35 R3 R4 as a voltage divider to check if the switch is ON/OFF.
I have been using the Connect and POWER_Off  code  **( fall_asleep()** and **wake_up() )** .
I tried using different way of using AT+POWER_OFF but still problems. 
After a while, i get lost very easily.

Many Thanks for your time.

schreibfaul1, 2024-06-25:
You could assign a GPIO to "AMP_ENABLED". This is intended to switch on external audio amplifiers.  This will work as an indicator for your purpose.

ve2cfb, 2024-06-25:
Hello Wolle,
Thanks for the the comment.

The AMP_ENABLED is an OUtput pin where as my GPIO35 is an input.

Here is my modified Schematic. As you can see, the BT module has constant 5v connected to it.

[image]

Since you already have the code in the **void fall_asleep()**   and the **void** fall_asleep()** functions, 

**if(BT_EMITTER_CONNECT != -1){digitalWrite(BT_EMITTER_CONNECT, LOW); vTaskDelay(100); digitalWrite(BT_EMITTER_CONNECT, HIGH);} // POWER_ON**

and....

**if(BT_EMITTER_CONNECT != -1){digitalWrite(BT_EMITTER_CONNECT, HIGH); bt_emitter.cmd_PowerOff();}**

Its just a manner to copy those  2 **Power_On and Power_Off** , somewhere else and use the GPIO35  input pin to control them.

That is why i am asking you where would it be the best place in the LOOP() function to incorporate those 2 functions without interffering with your code.

Again, thanks for your time.

schreibfaul1, 2024-06-25:
I think I misinterpreted your first post.
You want to control the BT_Emitter (on/off) externally via a GPIO?

ve2cfb, 2024-06-26:
Helo again,

Yes,  using GPIO35 as INPUT, and the switch will toggle between CONNECT and POWER_OFF.
I was able to make it partially work but somehow,  i lost it.
I will try using an other approch and see what happen.
 To be followed...

valera39, 2024-06-26:
[schreibfaul1](https://github.com/schreibfaul1) I support you, this button is necessary

ve2cfb, 2024-06-26:
Yes Wolle, you are right.
the only time i got it working was with the UART command mode.
If you do the change, that would be super.
Thanks again.

ve2cfb, 2024-06-30:
Hello Wolle...
Concerning your previous comment:
**_I am considering whether it would be useful to add a BT of/off button to the display, similar to the BT button on the smartphone.
This would be easy to control externally (via a GPIO or the IR remote control)._**

I would definitly go with an external button which include a Blue Led, just like my previous Schematic.

Many thanks...

closed 2024-08-14 by github-actions[bot]