schreibfaul1/ESP32-MiniWebRadio #461: HLS Packet 'chunk' skipped

RobinMosedale, 2024-07-06:
Wolle,

Recall the intent was receiving BBC HLS M3u8 playlist of TS segments

Consistent HLS Packet skipped, all BBC channels, resulting in for a voice broadcast: half a sentence skipped regularly, and on BB3 a portion of a bar of music skipped.

Is my PCM5012  running slow?

I've built two identical radios'
Configuration:
ESP32 Devkit S3 N32R8
ILI9341
PCM5012
Max 98306
12V to 5V 5A DC-DC Power Down Module Power Supply
There is no Bluetooth, but left the pins defined
Bit Clock (BCK) is connected to ESP32 pin 3
The PCM 5012A SCK-> ESP32 GND, should this be otherwise?

[image]
[image]

The relevant config.h elements are:

```
#ifdef CONFIG_IDF_TARGET_ESP32S3
    // Digital I/O used
        #define TFT_CS            8
        #define TFT_DC           12
        #define TFT_BL           10 // at -1 the brightness menu is not displayed
        #define TP_IRQ           39
        #define TP_CS            15
        #define SD_MMC_D0        11
        #define SD_MMC_CLK       13
        #define SD_MMC_CMD       14
        #define IR_PIN            4  // IR Receiver (if available)
        #define TFT_MOSI         18  // TFT and TP (FSPI)
        #define TFT_MISO          2  // TFT and TP (FSPI) ******TFT MISO Not connected
        #define TFT_SCK          17  // TFT and TP (FSPI)

        #define I2S_DOUT          9  // DIN
        #define I2S_BCLK          3 //BCK
        #define I2S_LRC           1 //LRCL-LCK
        #define I2S_MCLK          0

        #define I2C_DATA         -1  // some DACs are controlled via I2C
        #define I2C_CLK          -1
        #define SD_DETECT        -1  // some pins on special boards: Lyra, Olimex, A1S ...
        #define HP_DETECT        -1
        #define AMP_ENABLED      -1

        #define BT_EMITTER_RX    45  // TX pin - KCX Bluetooth Transmitter (-1 if not available)
        #define BT_EMITTER_TX    38  // RX pin - KCX Bluetooth Transmitter (-1 if not available)
        #define BT_EMITTER_LINK  19  // high if connected                  (-1 if not available)
        #define BT_EMITTER_MODE  20  // high transmit - low receive        (-1 if not available)
```
and TFT RST -> ESP32/RST/EN
PCM5012 SCK-> ESP32 GND
PCM5012 DMP, FLT, FMT, XMT all left floating although experimented with FMT and DMP set high and low

A portion of the serial log is below:-

The message may be significant:-
at 18:24:59
E (25878) i2s_std: i2s_channel_reconfig_std_clock(265): invalid state, I2S should be disabled before reconfiguring the clock
Should PCM5012A SCK not be set to GND?
ESP32 Chip: ESP32-S3
Arduino Version: 3.0.0
ESP-IDF Version: 5.1.3
MiniWebRadio     Version 3.00o Mar 02/2024 
[... 394 lines cut ...]
18:53:28 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268795064.ts"
18:53:35 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268795065.ts"
18:53:42 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268795066.ts"
18:53:49 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268795067.ts"
18:53:56 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... 96000.norewind.m3u8"
[1750230][W][Audio.cpp:2744] parsePlaylist_M3U8(): mediaseq 268795068 skipped
18:53:56 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268795069.ts"
18:54:03 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268795070.ts"
18:54:10 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268795071.ts"
18:54:17 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268795072.ts"
18:54:24 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268795073.ts"
18:54:31 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... 96000.norewind.m3u8"
[1785051][W][Audio.cpp:2744] parsePlaylist_M3U8(): mediaseq 268795074 skipped
18:54:31 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268795075.ts"
18:54:38 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268795076.ts"
18:54:45 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268795077.ts"
18:54:52 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268795078.ts"
18:54:59 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268795079.ts"
18:55:06 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... 96000.norewind.m3u8"
18:55:06 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268795080.ts"
18:55:13 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268795081.ts"
18:55:20 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268795082.ts"
18:55:27 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268795083.ts"
18:55:34 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... 96000.norewind.m3u8"
18:55:34 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268795084.ts"

RobinMosedale, 2024-07-16:
On closer inspection and tracing back, this seems to be an issue with decoding the playlist within your Auio.cpp, and nothing to do with a slow que to the PCM5012. Let me know if you need any further diagnostics, perhaps the M3U8 record

schreibfaul1, 2024-07-19:
This is a problem that occurs with some HLS stations. Each ts packet is always 188 bytes in size (payload 184 bytes). Sometimes a ts packet has to be skipped to ensure synchronisation. Every ts packet and every m3u8 playlist are files that are downloaded via the WiFiClient. Unfortunately, the WiFiClient does not work without blocking, so unfortunately there is a lot of waiting time during which nothing can be done.

RobinMosedale, 2024-07-20:
That's a real shame, and more than noticeable with both music and voice broadcast.
The cyclic buffer is not even partially filled and plenty space. Perhaps only skip when buffering becomes an issue after several hours.
As it seems a shame to waste two perfectly decent builds, I may in slow time to see if there are other WifiClients that do not block. or experimentally disable the skip line

schreibfaul1, 2024-07-27:
I have changed a few things in the audioI2S library. The WiFiClient is not blockade-free. To get around this, I have added an extra task for decoding. This ensures that the I2S_DMA is sufficiently filled despite the blockade, http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d96000.norewind.m3u8 now runs much better.

RobinMosedale, 2024-07-27:
Thank you so much, Wolle, it's more than appreciated you taking the trouble.
After re-cloning the project, and reconfiguring, it seems as though I've broken PIO, IDF tools don't work, menuconfig etc.

I'll investigate later as I'm keen to get this to work

Very many thanks again, your efforts are very much appreciated

Robin

schreibfaul1, 2024-07-27:
Previously, the Arduino loop task did everything, fetched the data via WiFi, decoded it and transferred it to the I2S. Now the output and decoding is done by another task. The loop task now has more time for WiFi so that additional settings in menuconfig are no longer necessary.

RobinMosedale, 2024-07-28:
This is looking very promising indeed. No sign of chunk skipped as yet. I'll leave it as soak test for an hour and report back.
Very many thanks indeed.
At the moment it's located in ideal conditions, less than 30 cm from the Wifi beacon.
I'll then expose it to less ideal signal, and see if the connection recovery works better
Remarkable assistance.

RobinMosedale, 2024-07-28:
Well, that's a resounding success.

Left from 17:00 BST until now 19:45 BST, no chunks skipped and one packet rx failure recovered as below.

Well done, Wolle, an excellent result. Now to test with a weaker signal to see if packet loss recovery has improved.

Robin
18:41:51 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269091957.ts"
18:41:58 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269091958.ts"
18:42:04 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d96000.norewind.m3u8"
18:42:04 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269091959.ts"
18:42:10 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269091960.ts"
18:42:17 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269091961.ts"
18:42:23 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269091962.ts"
18:42:30 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d96000.norewind.m3u8"
18:42:30 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269091963.ts"
18:42:36 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269091964.ts"
[5703489][E][Audio.cpp:3785] parseHttpResponseHeader(): timeout
18:42:40 StationName:
18:42:40 AUDIO_info:  Connect to new host: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d96000.norewind.m3u8"
18:42:40 AUDIO_info:  buffers freed, free Heap: 185852 bytes
[5710518][E][NetworkManager.cpp:130] hostByName(): DNS Failed for 'as-hls-ww-live.akamaized.net' with error '-54'
[5713031][I][NetworkClient.cpp:253] connect(): select returned due to timeout 2500 ms for fd 53
18:42:50 AUDIO_info:  Request http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_four ... orewind.m3u8 failed!
[5713053][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 8 - ASSOC_LEAVE
[5713055][W][main.cpp:2819] audio_info(): disconnected, wait 35s
[5748067][W][main.cpp:2821] audio_info(): try reconnection
18:43:25 StationName:
18:43:25 StreamTitle:
18:43:25 RECONNECTION to http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d96000.norewind.m3u8, try 1
[5748128][I][WiFiMulti.cpp:86] addAP(): [WIFI][APlistAdd] add SSID: TP-LINK_EB850C
[5748149][I][WiFiMulti.cpp:86] addAP(): [WIFI][APlistAdd] add SSID: TP-LINK_EB850C_Ext
[5748158][I][WiFiMulti.cpp:86] addAP(): [WIFI][APlistAdd] add SSID: Rose
[5748166][I][WiFiMulti.cpp:86] addAP(): [WIFI][APlistAdd] add SSID: Roise_4
[5748174][I][WiFiMulti.cpp:86] addAP(): [WIFI][APlistAdd] add SSID: TP-LINK_EB850C
[5748183][I][WiFiMulti.cpp:86] addAP(): [WIFI][APlistAdd] add SSID: Rose
[5748190][I][WiFiMulti.cpp:86] addAP(): [WIFI][APlistAdd] add SSID: Rosie_3
[5748198][I][WiFiMulti.cpp:86] addAP(): [WIFI][APlistAdd] add SSID: Rosiem
18:43:31 setup: ....  4 WiFi networks found
18:43:31 setup: ....  TP-LINK_EB850C (-27)
18:43:31 setup: ....  TP-LINK_EB850C_Ext (-46)
18:43:31 setup: ....  Roise_4 (-64)
18:43:31 setup: ....  Rose (-80)
[5760239][I][WiFiMulti.cpp:129] run(): [WIFI] scan done
[5760245][I][WiFiMulti.cpp:134] run(): [WIFI] 4 networks found
[5760252][I][WiFiMulti.cpp:249] run(): [WIFI] Connecting BSSID: 60:E3:27:EB:85:0C SSID: TP-LINK_EB850C Channel: 1 (-27)
[5760264][W][STA.cpp:537] disconnect(): STA already disconnected.
[5760694][I][WiFiMulti.cpp:272] run(): [WIFI] Connecting done.
18:43:38 WiFI_info:   Connecting WiFi...
[5760701][E][ESPmDNS.cpp:181] addService(): Failed adding service esp32.tcp.

18:43:38 WiFI_info:   mDNS name: MiniWebRadio
18:43:38 AUDIO_info:  Connect to new host: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d96000.norewind.m3u8"
18:43:38 AUDIO_info:  buffers freed, free Heap: 184872 bytes
18:43:38 AUDIO_info:  Connection has been established in 62 ms, free Heap: 184416 bytes
18:43:38 lastURL: ..  http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_r ... d96000.norewind.m3u8
18:43:38 AUDIO_info:  media sequence number: 269091972
18:43:38 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269091972.ts"
18:43:38 AUDIO_info:  AACDecoder has been initialized, free Heap: 182688 bytes , free stack 4888 DWORDs
18:43:38 Audiocodec:  AAC
18:43:39 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269091973.ts"
18:43:40 AUDIO_info:  stream ready
18:43:40 AUDIO_info:  syncword found at pos 0
[... cut ...]
18:52:19 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269092055.ts"
18:52:25 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269092056.ts"
18:52:32 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d96000.norewind.m3u8"

RobinMosedale, 2024-07-29:
A few tests on 2 identical units, with only the default WiFi beacon changed, in 2 different locations. 
1. Within 2 metres of the fibre hub WiFi beacon, good signal. It experienced fetch dropouts, with several minutes attempting to recover. During which time, both the Web server became unresponsive, and the touch screen also.
2. In a more challenging location with a weaker signal but where my android phone happily receives BBC Sounds broadcast. Spends most of the time recovering with lock out
It's also noticed that the Web interface and touch screen is much less responsive even when not recovering and suspect that these function are on the same task as the WiFi rx function. Would it help if these functions were also separated as individual tasks?

These units are remote from the test rig, so it'll be a few days before I can gain diagnostic logs

RobinMosedale, 2024-08-04:
Some observations:
Improvements, no chunks skipped, and it resolved my impression that the Ts segments were played slowly, and with a slightly lower pitch. All normal now.
Side effects.
The touch screen is now very unresponsive, with new displays as a result of a touch, appearing at a very slow rate.
Web server is also very slow to the point of being unresponsive.
If a fetch fails, and wifi retry is in operation, anything other than the Wifi function is locked out. Screen input, webpage access.

There follows a single instance of a wifi retry, but this one was successful, if a little tortuous in its operation. More often the unit becomes stuck in a retry/reconnect loop permanently, whereas other devices are happily connecting. The only way to recover is to recylce power.

There appears to be two issues.
1. Splitting out the audio function into a separate task is successful in maintaining speed of audio decoding, however, it is suspected that all remaining tasks are in one task stream: Wifi, Web Server, touch screen, and if there is wifi recovery, nothing else operates. Perhaps further separation of tasks, with appropriate priority of scheduling.
2. The wifi recovery seems to be failing. I've read there needs to be a disconnect before retry or reconnect.
3. Perhaps we are running out of processing
4. Further tracing hopefully with a permanent lockout shown.
`20:37:56 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269120045.ts"
20:38:02 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269120046.ts"
20:38:08 action: ...  switch to station 7
20:38:08 AUDIO_info:  Connect to new host: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_three/bbc_radio_three.isml/bbc_radio_three-audio%3d96000.norewind.m3u8"
20:38:08 AUDIO_info:  buffers freed, free Heap: 184732 bytes
[4071390][E][NetworkManager.cpp:130] hostByName(): DNS Failed for 'as-hls-ww-live.akamaized.net' with error '-54'
[4073903][I][NetworkClient.cpp:253] connect(): select returned due to timeout 2500 ms for fd 53
20:38:17 AUDIO_info:  Request http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_thre ... orewind.m3u8 failed!
[4073926][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 8 - ASSOC_LEAVE
[4073928][W][main.cpp:2819] audio_info(): disconnected, wait 35s
[4108939][W][main.cpp:2821] audio_info(): try reconnection
20:38:52 StationName:
20:38:52 StreamTitle:
[4108947][E][NetworkClient.cpp:423] write(): fail on fd 55, errno: 113, "Software caused connection abort"
[4108994][W][common.h:1157] GetImageSize(): file /logo/s/BBC Radio 3.jpg not exists, objName: pic_RA_logo
[4109012][W][common.h:1157] GetImageSize(): file /logo/s/BBC Radio 3.jpg not exists, objName: pic_RA_logo
20:38:53 RECONNECTION to http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d96000.norewind.m3u8, try 1
[4109195][I][WiFiMulti.cpp:86] addAP(): [WIFI][APlistAdd] add SSID: Rose
[4109215][I][WiFiMulti.cpp:86] addAP(): [WIFI][APlistAdd] add SSID: Rose
[4109223][I][WiFiMulti.cpp:86] addAP(): [WIFI][APlistAdd] add SSID: Roise_4
[4109231][I][WiFiMulti.cpp:86] addAP(): [WIFI][APlistAdd] add SSID: Rosiem
[4109239][I][WiFiMulti.cpp:86] addAP(): [WIFI][APlistAdd] add SSID: Rosie_3
20:38:59 setup: ....  4 WiFi networks found
20:38:59 setup: ....  TP-LINK_EB850C (-53)
20:38:59 setup: ....  Rose (-57)
20:38:59 setup: ....  TP-LINK_EB850C_Ext (-67)
20:38:59 setup: ....  Roise_4 (-71)
[4121281][I][WiFiMulti.cpp:129] run(): [WIFI] scan done
[4121287][I][WiFiMulti.cpp:134] run(): [WIFI] 4 networks found
[4121293][I][WiFiMulti.cpp:249] run(): [WIFI] Connecting BSSID: 00:0F:94:D5:A2:14 SSID: Rose Channel: 9 (-55)
[4121304][W][STA.cpp:537] disconnect(): STA already disconnected.
[4121384][I][WiFiMulti.cpp:272] run(): [WIFI] Connecting done.
20:39:05 WiFI_info:   Connecting WiFi...
[4121391][E][ESPmDNS.cpp:181] addService(): Failed adding service esp32.tcp.

20:39:05 WiFI_info:   mDNS name: MiniWebRadio
20:39:05 AUDIO_info:  Connect to new host: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d96000.norewind.m3u8"
20:39:05 AUDIO_info:  buffers freed, free Heap: 185848 bytes
20:39:05 AUDIO_info:  Connection has been established in 27 ms, free Heap: 185232 bytes
20:39:05 lastURL: ..  http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_r ... d96000.norewind.m3u8
20:39:05 AUDIO_info:  media sequence number: 269120054
20:39:05 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269120054.ts"
20:39:05 AUDIO_info:  AACDecoder has been initialized, free Heap: 183876 bytes , free stack 5060 DWORDs
20:39:05 Audiocodec:  AAC
20:39:06 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269120055.ts"
20:39:06 AUDIO_info:  stream ready
20:39:06 AUDIO_info:  syncword found at pos 0
20:39:06 AUDIO_info:  Channels: 2
20:39:06 AUDIO_info:  SampleRate: 48000
20:39:06 AUDIO_info:  BitsPerSample: 16
20:39:06 AUDIO_info:  BitRate: 129000
20:39:06 AUDIO_info:  AAC HeaderFormat: ADTS
20:39:06 AUDIO_info:  AAC Codec: MPEG-4 LowComplexity
[... cut ...]
20:39:52 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269120063.ts"
20:39:58 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269120064.ts"
20:40:04 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269120065.ts"
`

closed 2024-08-08 by RobinMosedale