schreibfaul1/ESP32-MiniWebRadio #463: Refetch chunk hangup

RobinMosedale, 2024-07-08:
Wolle,

Recall the intent was receiving BBC HLS M3u8 playlist of TS segments

I've built two identical radios'
Configuration:
ESP32 Devkit S3 N32R8
ILI9341
PCM5012
Max 98306
12V to 5V 5A DC-DC Power Down Module Power Supply
There is no Bluetooth, but left the pins defined

This occurs on BBC HLS Chunked sites. I haven't observed this on old 'shout-out' sites
WIFI isn't perfect, but I've established a reliable beacon with reasonable power.
This occurs on both of my units.
A timeout can occur on fetching one of the .ts files defined in an M3U8 playlist.
Occasionally I can see it recover with a reconnect. However, more often it gets stuck, and can only be released by manually switching station, and then switching back.
Wifi connection hasn't been broken as I can receive data from the webpage with stack and que size responses from a ?. Switching station can be made either via the touchscreen or the WebServer.
The hang ups can occur several times in a hour hour programme.

The attached shows the trace from a reasonably long successful connection, then a 20 minute hang up, followed by intervention via the touch screen, switching to another station stream, and then back.

I'm mindful that the BBC specification is very strict. Requests are strictly time limited in each playlist chunk, to prevent excessive requests. I'm not sure whether this is occurring.

Let me know if there are further diagnostics which may help pinpoint what is going on during the break.

```
*  Executing task: C:\Users\robin\.platformio\penv\Scripts\platformio.exe device monitor --port COM4 

git version 2.45.2.windows.1
Cloning into 'C:\Users\robin\.platformio\.cache\tmp\pkg-installing-hjj26idh'...
warning: Could not find remote branch idf-v5.1.3 to clone.
fatal: Remote branch idf-v5.1.3 not found in upstream origin
Esp32ExceptionDecoder: disabling, exception while looking for addr2line: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', '--branch', 'idf-v5.1.3', 'https://github.com/espressif/esp32-arduino-libs.git', 'C:\\Users\\robin\\.platformio\\.cache\\tmp\\pkg-installing-hjj26idh']

Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

--- Terminal on COM4 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29d8
entry 0x403c9880
[   618][I][esp32-hal-psram.c:92] psramInit(): PSRAM enabled

ESP32 Chip: ESP32-S3
Arduino Version: 3.0.0
ESP-IDF Version: 5.1.3
MiniWebRadio     Version 3.00o Mar 02/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 33554432 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 20 MHz
TFT speed 40 MHz
PSRAM total size: 8388608 bytes
RESET_REASON: Reset due to power-on event

00:00:00
[... 220 lines cut ...]
14:20:42 Script:      index.js
14:20:44 webSrv: ...  GetFolder "/"
14:20:44 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268819507.ts"
14:20:50 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268819508.ts"
14:20:57 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268819509.ts"
14:21:04 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... 96000.norewind.m3u8"
14:21:04 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268819510.ts"
14:21:11 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268819511.ts"
14:21:18 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268819512.ts"
14:21:25 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268819513.ts"
14:21:32 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268819514.ts"
14:21:39 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... 96000.norewind.m3u8"
[5525143][W][Audio.cpp:2744] parsePlaylist_M3U8(): mediaseq 268819515 skipped
14:21:39 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268819516.ts"
14:21:46 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268819517.ts"
14:21:53 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268819518.ts"
14:22:00 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268819519.ts"
14:22:07 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268819520.ts"
14:22:14 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... 96000.norewind.m3u8"
14:22:14 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268819521.ts"
14:22:21 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268819522.ts"
14:22:28 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268819523.ts"
14:22:35 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radi ... =96000-268819524.ts"

```

schreibfaul1, 2024-07-16:
I have also observed this, I suspect the name is not resolved, this is only since the last Arduino versions, "DNS failed for..." I hope the error is gone with Arduino V3..., but PlatformIO (still) causes problems with V3.

RobinMosedale, 2024-07-16:
Thank you, Wolle.
I have only ever built with your recommended method of Platform IO. I can't find Arduino v3, the latest is 
Arduino IDE 2.3.2. Do I have to install a specific version of Arduino as an extension within Platform IO? Could you point me to where I might find Arduino V3 and how to build with the parameters required under Platform IO

RobinMosedale, 2024-07-20:
Seems quite a significant flaw, and requires manual intervention several times during a broadcast.
Is it possible to revert to a previous Arduino and build with that? and how would one do it?
Seems such a shame to waste to perfectly good builds

RobinMosedale, 2024-08-08:
Some tests:
A few tests on 2 identical units, with only the default WiFi beacon changed, in 2 different locations.

Within 2 metres of the fibre hub WiFi beacon, good signal. It experienced fetch dropouts, with several minutes attempting to recover. During which time, both the Web server became unresponsive, and the touch screen also.
In a more challenging location with a weaker signal but where my android phone happily receives BBC Sounds broadcast. Spends most of the time recovering with lock out
It's also noticed that the Web interface and touch screen is much less responsive even when not recovering and suspect that these function are on the same task as the WiFi rx function. Would it help if these functions were also separated as individual tasks?

Some tests have been performed.
An esp32 NAT repeater was programmed for two purposes.
1. The repeater can be moved to vary the quality of the signal, both from its origin hotspot AP, and varying the quality of the signal to the Webradio.
2. Both the Webradio and the NAT repeater can be monitored via serial link simultaneously to observe what the NAT AP is seeing the Radio doing in a recovery.

The NAT repeater was positioned where the signal to the radio was OK, and was coping with  distant origin Wifi AP.
Once a packet was lost, the radio seems to be stuck attempting a reconnect making the same connect error. It can e recovered by cycling the repeater to force loss of the DNS address, and rassigning.

Here is the radio trace.
`--- Terminal on COM20 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
00:00:07 setup: ....  8 WiFi networks found
00:00:07 setup: ....  TP-LINK_EB850C_Ext (-36)
00:00:07 setup: ....  TP-LINK_EB850C (-45)
00:00:07 setup: ....  ESP32_NAT_Router (-47)
00:00:07 setup: ....  DIRECT-276C4D1D (-52)
00:00:07 setup: ....  Roise_4 (-71)
00:00:07 setup: ....  Rose (-77)
00:00:07 setup: ....  podpoint (-87)
00:00:07 setup: ....  Rosie_3 (-90)
[ 14315][I][WiFiMulti.cpp:129] run(): [WIFI] scan done
[ 14320][I][WiFiMulti.cpp:134] run(): [WIFI] 8 networks found
[ 14327][I][WiFiMulti.cpp:249] run(): [WIFI] Connecting BSSID: FC:B4:67:74:59:A1 SSID: ESP32_NAT_Router Channel: 9 (-43)
[ 14339][W][STA.cpp:537] disconnect(): STA already disconnected.
[ 18373][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
[ 18381][W][STA.cpp:537] disconnect(): STA already disconnected.
[ 18668][I][WiFiMulti.cpp:272] run(): [WIFI] Connecting done.
00:00:18 WiFI_info:   Connecting WiFi...
00:00:18 WiFI_info:   mDNS name: MiniWebRadio
00:00:18 setup: ....  connected to ESP32_NAT_Router, IP address is 192.168.4.2
[ 18708][I][ArduinoOTA.cpp:129] begin(): OTA server at: MiniWebRadio.local:3232
00:00:18 ftpServer:   Buffers allocated: 4622 bytes
00:00:18 rtime_info:  Initializing SNTP
00:00:18 rtime_info:  Waiting for system time to be set... (1/10)
00:00:20 rtime_info:  Waiting for system time to be set... (2/10)
16:41:34 rtime_info:  16:41:34
16:41:34 audiotask:   is pinned to core 0
16:41:34 audiotask:   priority is 2
16:41:34 AUDIO_info:  commFMT = Philips
16:41:34 setup: ....  Number of saved stations: 13
16:41:34 setup: ....  current station number: 8
16:41:34 setup: ....  current volume: 3
16:41:34 setup: ....  last connected host: http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d96000.norewind.m3u8
16:41:34 setup: ....  connection timeout: 2500 ms
16:41:34 setup: ....  connection timeout SSL: 3500 ms
16:41:34 action: ...  switch to station 8
16:41:34 AUDIO_info:  PSRAM found, inputBufferSize: 638965 bytes
16:41:34 AUDIO_info:  buffers freed, free Heap: 190144 bytes
16:41:34 AUDIO_info:  connect to: "as-hls-ww-live.akamaized.net" on port 80 path "/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d96000.norewind.m3u8"
16:41:34 AUDIO_info:  Connection has been established in 61 ms, free Heap: 189700 bytes
16:41:34 lastURL: ..  http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_r ... d96000.norewind.m3u8
16:41:35 AUDIO_info:  media sequence number: 269239328
16:41:35 AUDIO_info:  connect to: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269239328.ts"
16:41:35 AUDIO_info:  AACDecoder has been initialized, free Heap: 187252 bytes , free stack 5116 DWORDs
16:41:35 AUDIO_info:  connect to: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269239329.ts"
16:41:35 WiFI_info:   RSSI is -40 dB
16:41:35 Audiocodec:  AAC
16:41:35 ftpServer:   Ftp server waiting for connection on port 21
16:41:36 AUDIO_info:  stream ready
16:41:36 AUDIO_info:  syncword found at pos 0
16:41:36 AUDIO_info:  Channels: 2
[... 296 lines cut ...]
I (1428637) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (1446547) wifi_event_handler: station disconnected - 1 remain
I (1446577) wifi_event_handler: 2. station connected
I (1446667) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (1464537) wifi_event_handler: station disconnected - 1 remain
I (1464567) wifi_event_handler: 2. station connected
[... cut ...]
if(bestNetwork.ssid) {
    DEBUG_WIFI_MULTI("[WIFI] Connecting BSSID: %02X:%02X:%02X:%02X:%02X:%02X SSID: %s Channal: %d (%d)\n", bestBSSID[0], bestBSSID[1], bestBSSID[2], bestBSSID[3], bestBSSID[4], bestBSSID[5], bestNetwork.ssid, bestChannel, bestNetworkDb);
```

    WiFi.disconnect(true); // <= Added this line
    WiFi.begin(bestNetwork.ssid, bestNetwork.passphrase, bestChannel, bestBSSID);
see:-

https://esp32.com/viewtopic.php?t=37331

https://github.com/espressif/arduino-esp32/issues/2269#top

schreibfaul1, 2024-08-09:
Thank you very much for your detailed experience report.

Currently 3 tasks are relevant for the execution. These are 
1) Arduino loop task, controls the audio library (fills the InBuffer, takes care of the metadata, playlists, audio headers and events), the web server and the display including the touchpad 
2) Audio task, fetches the data from the InBuffer within the audio library, decodes it and feeds I2S 
3) Audio control task controls the audio library (fills the InBuffer, takes care of the metadata, playlists, audio headers and events) and provides sufficient stack All three tasks are assigned a core in the platformio.ini is assigned to all three tasks. 

The Arduino loop task and the audio task should not run on the same core, otherwise the display will react very slowly. 

I am amazed that you are still experiencing the DNS problem so often. With Arduino 3.0.4 this is no longer recognizable to me.

If I understood correctly in the last part, you have several APs distributed around the house. If the signal of the connected AP becomes weaker, the connection is disconnected, but no connection is established to another AP.

RobinMosedale, 2024-08-09:
"If I understood correctly in the last part, you have several APs distributed around the house. If the signal of the connected AP becomes weaker, the connection is disconnected, but no connection is established to another AP."

Almost Wolle, but in this test all of the AP's in the network list had been commented out except this ESP NAT AP, so I could weaken the signal in a controlled manner.  The second log in the above is the diagnostics from the ESP32 NAT which clearly shows the repeated disconnect by the radio, and reconnect with ip address. However, the RadioWifi reconnect seems not to realise that it has been successful on the original IP address.

HOWEVER

You've been really busy, I mean astonishingly busy. I checked my build and it had arduino_espressif_32 3.0.2
I've just rebuilt with 3.0.4, and my goodness the speed at start up is astonishing.

However, I'm still experiencing getting stuck on a reconnect loop, even though the signal seems good.

Here is the ESP32 NAT Log:
`I (983567) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (983847) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (992427) wifi_event_handler: station disconnected - 2 remain
I (1022807) wifi_event_handler: 3. station connected
I (1024797) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (1032007) wifi_event_handler: station disconnected - 2 remain
I (1032097) wifi_event_handler: 3. station connected
I (1034207) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (1038207) wifi_event_handler: station disconnected - 2 remain
I (1057127) wifi_event_handler: 3. station connected
I (1059887) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (1070227) wifi_event_handler: station disconnected - 2 remain
I (1070347) wifi_event_handler: 3. station connected
I (1071137) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (1071467) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
`
Here is the radio log (nice touch):-
`20:58:25 AUDIO_info:  connect to: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269255236.ts"
20:58:32 AUDIO_info:  connect to: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269255237.ts"
20:58:39 AUDIO_info:  connect to: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269255238.ts"
20:58:44 AUDIO_info:  connect to: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269255239.ts"
20:58:49 AUDIO_info:  connect to: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269255240.ts"
20:58:55 AUDIO_info:  connect to: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d96000.norewind.m3u8"
20:58:55 AUDIO_info:  connect to: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269255241.ts"
20:59:03 AUDIO_info:  connect to: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269255242.ts"
20:59:08 AUDIO_info:  connect to: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269255243.ts"
20:59:14 AUDIO_info:  connect to: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269255244.ts"
20:59:22 AUDIO_info:  connect to: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d96000.norewind.m3u8"
20:59:22 AUDIO_info:  connect to: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269255245.ts"
20:59:32 AUDIO_info:  connect to: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269255246.ts"
20:59:39 AUDIO_info:  connect to: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269255247.ts"
20:59:46 AUDIO_info:  connect to: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269255248.ts"
20:59:51 AUDIO_info:  connect to: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269255249.ts"
20:59:53 Time: ...... Announcement at 21 o'clock is silent
20:59:56 AUDIO_info:  connect to: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d96000.norewind.m3u8"
20:59:56 AUDIO_info:  connect to: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269255250.ts"
21:00:00 AUDIO_info:  connect to: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269255251.ts"
[... cut ...]
[1007785][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 8 - ASSOC_LEAVE
21:13:45 AUDIO_info:  Request http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d96000.norewind.m3u8 failed!
21:13:45 StationName:
21:13:45 StreamTitle:
21:13:47 AUDIO_info:  buffers freed, free Heap: 199972 bytes
21:13:47 AUDIO_info:  connect to: "as-hls-ww-live.akamaized.net" on port 80 path "/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d96000.norewind.m3u8"`

RobinMosedale, 2024-08-10:
Some more diagnostics and observations this afternoon.
This using the latest build, with the separated tasks on both cores.

The test harness is the radio on PIO serial link, the ESP32 NAT adjacent to the radio, which in turn is logged on to an AP which derived from ethernet adjacent to the desk, and yet another Wifi repeater 60 cm away by the desk.

The radio netlist contains the log on details of all 3 AP's, and does successfully log on to each.

[image]

The ESP32 NAT is also monitored on a serial link in order to monitor the allocation and release of IP address to the radio.

The radio is booted, and successfully logs on to the ESP32 NAT, and fetching of HLS chunks proceeds.
The ESP32 NAT attached to a laptop is moved away, until the radio loses synch and connection failure. The ESP32 NAT logs the disconnect. The NAT is quickly brought back to the radio, and it can be observed that:-
1. The NAT repeatedly logs IP allocation to the Radio, followed by release as disconnected. It's the same IP as originally allocated.
2. The radio is stuck attempting to reconnect. The STA task is active believing already disconnected. The NAT allocation and release can clearly bee seen.
3. Repeated failures to connect to the m3u8 playlist, however the radio is still connected as can be seen at:-
4. 17:14:09 Webpage:     index.html as the webserver responds.
5. The NAT is clearly showing the repeated reconnection and re-assigning the IP, followed by release. The Web page does respond but slow, so it is connected. However, the Audio task seems to fail fetching the audio playlist.
6. In any event, the radio WP task has 3 good AP beacons loud and clear, which it ignores of fails to log on.
7. Once it gets into this state, the only method of freeing the loop is reset and repower up.

Hopefully these logs are less cluttered than previous data

Here is the ESP32 NAT log showing the assigning IP, and release repeated.
`

esp32> show
I (34068) cmd_router: ssid TP-LINK_EB850C
I (34078) cmd_router: ent_username
I (34078) cmd_router: passwd blanked
I (34078) cmd_router: ap_ssid ESP32_NAT_Router
I (34088) cmd_router: ap_passwd blanked
STA SSID: TP-LINK_EB850C Password: blanked Enterprise: <not active>
AP SSID: ESP32_NAT_Router Password: blanked
AP IP address: 192.168.4.1
Uplink AP connected
IP: 192.168.1.176
2 Stations connected
esp32> I (46478) wifi_event_handler: station disconnected - 1 remain
I (50528) wifi_event_handler: 2. station connected
I (50748) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (111558) wifi_event_handler: station disconnected - 1 remain
I (115358) wifi_event_handler: 2. station connected
I (116958) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (117128) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (120248) wifi_event_handler: station disconnected - 1 remain
I (136038) wifi_event_handler: 2. station connected
I (136148) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (142608) wifi_event_handler: station disconnected - 1 remain
I (142638) wifi_event_handler: 2. station connected
I (142708) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (154518) wifi_event_handler: station disconnected - 1 remain
I (154548) wifi_event_handler: 2. station connected
I (154578) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (158198) wifi_event_handler: station disconnected - 1 remain
I (158288) wifi_event_handler: 2. station connected
I (158478) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (162208) wifi_event_handler: station disconnected - 1 remain
I (162258) wifi_event_handler: 2. station connected
I (162618) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (162768) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (166668) wifi_event_handler: station disconnected - 1 remain
I (166778) wifi_event_handler: 2. station connected
I (167588) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (171188) wifi_event_handler: station disconnected - 1 remain
I (171218) wifi_event_handler: 2. station connected
I (171568) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (171768) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (176228) wifi_event_handler: station disconnected - 1 remain
I (177558) wifi_event_handler: 2. station connected
[... 221 lines cut ...]
17:13:02 AUDIO_info:  buffers freed, free Heap: 200924 bytes
17:13:02 AUDIO_info:  connect to: "as-hls-ww-live.akamaized.net" on port 80 path "/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d96000.norewind.m3u8"
17:13:04 AUDIO_info:  Connection has been established in 1591 ms, free Heap: 201184 bytes
17:13:04 lastURL: ..  http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_r ... d96000.norewind.m3u8
17:13:05 AUDIO_info:  media sequence number: 269266623
17:13:05 AUDIO_info:  connect to: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=96000-269266623.ts"
17:13:05 AUDIO_info:  AACDecoder has been initialized, free Heap: 199364 bytes , free stack 4796 DWORDs
17:14:09 Webpage:     index.html
[263919][E][websrv.cpp:98] show(): write error in webpage
17:14:34 Webpage:     index.html
[... cut ...] oximity.

Nevertheless, this is testing connection recovery. When the Radio  experiences no connection dropouts, it performs superbly, as does the one in the kitchen (as long as I don't turn up the Induction cooker to full power which swamps the spectrum).

Are the two task out of sync with registering the connection, STA.cpp and Audio_info?

Let me know if I can perform further diagnostics, Wolle.

Best wishes, and hope that we can overcome this otherwise excellent piece of HLS decoding.

Robin

RobinMosedale, 2024-08-20:
Bump.
Still a frequent issue

closed 2024-11-04 by github-actions[bot]