schreibfaul1/ESP32-MiniWebRadio #466: Important issues

valera39, 2024-07-16:
Dear Wolle!
I have some very important questions piling up for me, some of which I have asked before:
1. Please fix the ability to select the volume of the alarm tone when waking up the radio.
2. When idle for a long time in the sleep mode, the radio does not always wake up, it happens that the screen backlight slightly lights up, but there is no image and sound.
Also sometimes when power is applied there is a similar situation, you have to turn the receiver on and off several times I will try to catch this moment in the monitoring log.
3. Can I make it so that when the receiver wakes up the pre-selected station with a predetermined volume?
4. Is the state of the bluetooth module is preserved at power off. I mean the state of the module, transmitter or receiver. 
5. The web page used to display the station name here, now it is missing.
[image]

6. The volume step is 21, is there no possibility to increase it to fine tune the sound?
Thank you for your consideration and responsiveness!

valera39, 2024-07-16:
I forgot to add one more very important thing: on the latest version of ESP32-audioI2S library I don't play some streams, for example http://pub0302.101.ru:8000/stream/pro/aac/64/46.
The log shows that there are no errors, but instead of sound there is noise, the volumeter stands still. I had to use version 3.0.11e? Everything is fine with it.

valera39, 2024-07-17:
I also noticed that if the radio is in sleep mode for a long time, the time displayed on the display runs forward for about 10 hours for 7 minutes. maybe there is a sense to apply a real-time clock module RTC?

steve6375, 2024-07-17:
Or it woke up every 10 minutes, got the time, and then went back to sleep?

valera39, 2024-07-17:
The radio was in sleep mode almost all day, I happened to look at the time and saw that it didn't match the real time. Maybe it's an isolated case, I don't know, but I read somewhere that ESP32 has a built-in mediocre clock module. I don't usually keep the radio on all the time.

valera39, 2024-07-17:
And now the clock in sleep mode has become very laggy. I also caught an interesting moment: if you wake up the receiver with the remote control button, the time in the upper right corner of the display is real. Then if you turn the receiver off again with the remote control, the screen goes out, there is only a weak backlight and instead of hours and minutes on the screen only their separator of two dots, then the last digit appears. I have to reset the receiver's power supply.
[image]
I was planning to use the radio as an alarm clock, but this is no longer possible. I really hope Wolle will fix this problem

valera39, 2024-07-19:
I don't want to guess, but it looks like the sleep time problem has been solved. I replaced the SD card with a new class 10 16GB card. I also soldered all connections between the card reader and ESP32. I also soldered three pull-up resistors from the card reader board and now during the day the time in sleep mode is displayed correctly, the menu on the display switches quickly. 
[image]
Perhaps my actions will be useful to someone in the future. 
I delete my previous comments and leave only questions to the respected Wolle and look forward to their answer.

Translated with DeepL.com (free version)

schreibfaul1, 2024-07-19:
Many points, I'll start with the last one: 
- The resistors on the SD card adapter (~10KOhm) do not interfere, there is only one exception: The resistor on D0 must be removed only on the ESP32 (not necessary on ESP32-S3) otherwise there will be problems at startup.
- http://pub0302.101.ru:8000/stream/pro/aac/64/46 (Radio 101) seems to work, but the metadata is weird, maybe json coded, I'll take a closer look soon
- the audioI2S library allows up to 100 volume steps. 21 is still from the old days
- What do you mean by "5. The web page used to display the station name here, now it is missing?"
- The ringtone is currently always at maximum volume, but could be made adjustable.
-The BT status is currently not saved, but I could include it in the "settings.json".

valera39, 2024-07-19:
Here is a screenshot of version 2 of the radio. 
[image]
Here you can see that the line has the name of the station playing at the moment.
On the alarm clock, which is more important to me: how to set the volume of the alarm clock? I have it when waking up plays constantly quiet, I changed the volume value in "settings.json" but the volume does not change.
Could it be because I have both alarm clock and radio on when waking up?
It would be nice if the volume step was 100 logarithmically for smooth adjustment. And also save BT values in "settings.json".
And you didn't say anything about suggestion number 3. It would be very nice if it was possible.

valera39, 2024-07-19:
For the alarm clock, the following solution is logical: when you wake up, the alarm sound is played at a high enough volume to wake you up  :) and then starts playing the previously selected station at the previously set volume.

schreibfaul1, 2024-07-27:
The station name and URL are now displayed correctly again

valera39, 2024-07-27:
What do you say to my other questions?
I look forward to hearing what you have to say

valera39, 2024-07-28:
Dear Wolle!
Perhaps you don't have time or don't see my posts, but I'll allow myself to reiterate my concerns and wishes:
Please fix the ability to select the alarm volume when the radio wakes up.

When staying in sleep mode for a long time, the radio does not always wake up, it happens that the screen backlight lights up slightly, but there is no picture or sound.
Also sometimes when power is applied there is a similar situation, you have to turn the receiver on and off several times I will try to catch this moment in the monitoring log.

Is it possible to make it so that when the receiver wakes up the pre-selected station with a preset volume is turned on?

Make it possible to save the bluetooth module state in a json foil

Volume step 100, what should be changed to increase it?

schreibfaul1, 2024-07-28:
Hello valera39,
an adjustable ringtone is coming.
The issue that the radistation does not run when waking up has something to do with an error `hostbyname() DNS fail for....`, I wrote something about it in the Espressif Arduino forum. I have set a workaround that neutralizes the DNS problem, the wake-up should now work reliably.
Now the last set volume is used when waking up.
Volume levels up to 255 can be preset in the audioI2S lib. But I still have no idea how to implement this.
What is still to come is a control of the display brightness by the ambient light, but only for the ESP32-S3. For this purpose, an I2C bus is used to query the BH170 sensor. The library for this is already available.

valera39, 2024-07-28:
Thanks for the reply, I'll wait for the update and test it out

closed 2024-09-11 by github-actions[bot]