schreibfaul1/ESP32-MiniWebRadio #467: No OTA Upload since Versionupdate to 3.2h Jun 28/2024

HWH53, 2024-07-25:
Hallo Wolle,
nach dem Update auf Version 3.2h Jun 28/2024 kann ich das MiniWebRadio nicht mehr über OTA updaten. Habe ich eine erforderliche Einstellung übersehen?
Aus dem Uploadprotokoll  entnehme ich, dass die IP-Adresse für den ESP32S3 beim Upload anscheinend nicht bekannt ist??
upload_port     = "MiniWebRadio.local" und upload_protocol = espota sind wie in den vorherigen Versionen in der Platform.ini aktiviert. Die Portkonfiguration von VSCODE steht auf "Auto".
Nach dem Update über Serial - COM / Kabel hat sich das Webradio hat sich ins Wifi eingebucht und funktioniert. Ebenfalls ist die Weboberfläche über den Webbrowser zu erreichen, nur OTA funktioniert nicht mehr.

Das Uploadprotokoll habe ich angehängt.

kannst Du oder die Community mir weiterhelfen?

` *  Task wird ausgeführt: C:\Users\HWH\.platformio\penv\Scripts\platformio.exe run --target upload --environment esp32s3 

Processing esp32s3 (board: ESP32-S3-DevKitC-1-N16R8; platform: https://github.com/platformio/platform-espressif32.git#v6.4.0; framework: arduino, espidf)
--------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ESP32-S3-DevKitC-1-N16R8.html
PLATFORM: Espressif 32 (6.4.0+sha.516520f) > Espressif ESP32-S3-DevKitC-1-N16R8V (16 MB QD, 8MB PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 2.0.14+sha.44da992
 - framework-espidf @ 3.40405.230623 (4.4.5)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - tool-ninja @ 1.9.0
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0)
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Warning! Arduino framework as an ESP-IDF component doesn't handle the `variant` field! The default `esp32` variant will be used.
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 44 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP32-audioI2S @ 2.0.0+sha.d7a2007
|-- ESP32-KCX-BT-EMITTER @ 0.0.0+20240128094011.sha.b766d32
|-- ESP32-DLNA-Client @ 0.0.0+20231205185914.sha.3130951
|-- ESP32-IR-Remote-Control @ 1.0.10+sha.2bc531e
|-- FFat @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|-- ac101
|-- es8388
|-- ftpsrv
|-- rtime
|-- tftLib
|-- websrv
|-- ESPmDNS @ 2.0.0
|-- FS @ 2.0.0
|-- Preferences @ 2.0.0
|-- SD_MMC @ 2.0.0
|-- SPI @ 2.0.0
|-- Ticker @ 2.0.0
|-- WiFi @ 2.0.0
|-- Wire @ 2.0.0
Building in release mode
Retrieving maximum program size .pio\build\esp32s3\firmware.elf
Checking size .pio\build\esp32s3\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  21.9% (used 71812 bytes from 327680 bytes)
Flash: [=====     ]  54.2% (used 2095449 bytes from 3866624 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = espota
Uploading .pio\build\esp32s3\firmware.bin
12:32:34 [DEBUG]: Options: {'esp_ip': 'MiniWebRadio.local', 'host_ip': '0.0.0.0', 'esp_port': 3232, 'host_port': 42243, 'auth': '', 'image': '.pio\\build\\esp32s3\\firmware.bin', 'spiffs': False, 'debug': True, 'progress': True, 'timeout': 10}
12:32:34 [INFO]: Starting on 0.0.0.0:42243
12:32:34 [INFO]: Upload size: 2095808
Sending invitation to MiniWebRadio.local ..........
12:34:16 [ERROR]: No response from the ESP
*** [upload] Error 1
`

schreibfaul1, 2024-07-25:
Das stimmt, ich hatte OTA deaktiviert weil es einigen Usern Probleme bereitet hatte. Die Ursache war mir nicht ganz klar, zumal in anderen Projekten, wie im [ESPuino](https://forum.espuino.de/) OTA problemlos läuft. Ich habe OTA wieder eingeschaltet, aber mit einigen Änderungen in der platformio.ini. Sonst bleibt alles gleich, die Auswahl über "Switch PIO Project Environment" unten in der Taskleiste
[image]
und der "Upload Port" muss auf "Auto" stehen.
Du müsstest dann einmal noch mit USB uploaden, danach ist OTA wieder auf dem ESP32 aktiv. Und OTA geht nur auf ESPs mit 8MB Flash oder größer.

HWH53, 2024-07-26:
Hallo Wolle,
danke für die schnelle Unterstützung.
Nach dem Austausch der platformio.ini, der Neukompilierung und dem Upload über den Comport war kein Upload über OTA möglich. Fehlermeldungen siehe Uploadprotokoll.
Erst das erneute Clonen des Repositorys konnte ich OTA wieder funktionsfähig bekommen.😅

` *  Task wird ausgeführt: C:\Users\HWH\.platformio\penv\Scripts\platformio.exe run --target upload --environment esp32s3_OTA 

Processing esp32s3_OTA (board: ESP32-S3-DevKitC-1-N16R8; platform: https://github.com/platformio/platform-espressif32.git#v6.4.0; framework: arduino, espidf)
--------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ESP32-S3-DevKitC-1-N16R8.html
PLATFORM: Espressif 32 (6.4.0+sha.516520f) > Espressif ESP32-S3-DevKitC-1-N16R8V (16 MB QD, 8MB PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 2.0.14+sha.44da992
 - framework-espidf @ 3.40405.230623 (4.4.5)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - tool-ninja @ 1.9.0
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0)
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Warning! Arduino framework as an ESP-IDF component doesn't handle the `variant` field! The default `esp32` variant will be used.
Reading CMake configuration...
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
Processing 1 dependencies:

[1/1] idf (4.4.5)

-- Project sdkconfig file C:/Users/HWH/Documents/PlatformIO/Projects/Radio/ESP32-MiniWebRadio_V3/ESP32-MiniWebRadio/sdkconfig.esp32s3_OTA   
-- App "ESP32-MiniWebRadio" version: 1
-- Adding linker script C:/Users/HWH/Documents/PlatformIO/Projects/Radio/ESP32-MiniWebRadio_V3/ESP32-MiniWebRadio/.pio/build/esp32s3_OTA/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Users/HWH/.platformio/packages/framework-espidf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script C:/Users/HWH/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script C:/Users/HWH/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script C:/Users/HWH/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script C:/Users/HWH/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script C:/Users/HWH/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script C:/Users/HWH/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib-time.ld
-- Adding linker script C:/Users/HWH/.platformio/packages/framework-espidf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Configuring incomplete, errors occurred!
See also "C:/Users/HWH/Documents/PlatformIO/Projects/Radio/ESP32-MiniWebRadio_V3/ESP32-MiniWebRadio/.pio/build/esp32s3_OTA/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository: C:/Users/HWH/.platformio/packages/framework-espidf/components/openthread/openthread/../../../.git/modules/components/openthread/openthread
CMake Error at C:/Users/HWH/.platformio/packages/framework-arduinoespressif32-src-1bce0ea9e393cde368b46969d581065f/CMakeLists.txt:218 (message):
  esp32-arduino requires CONFIG_FREERTOS_HZ=1000 (currently 100)

======================================================== [FAILED] Took 5.08 seconds ========================================================
Environment    Status    Duration
-------------  --------  ------------
esp32s3_OTA    FAILED    00:00:05.082
================================================== 1 failed, 0 succeeded in 00:00:05.082 ================================================== 

 *  Der Terminalprozess "C:\Users\HWH\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload', '--environment', 'esp32s3_OTA'" wurde mit folgendem Exitcode beendet: 1. 
 *  Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste. `

danke nochmals für Unterstützung!

closed 2024-07-26 by HWH53