schreibfaul1/ESP32-MiniWebRadio #468: different IR remote conflict !!!

clear-sky, 2024-07-26:
hi dear wolle!
i tested irmp(https://github.com/IRMP-org/IRMP)
it says my ir remote address is 0xFD01 but your irlib says my ir remote address is 0X01!!!
but this is not all my problem
i have several nec ir remote and i was testing them on your radio and found odd problems!!!
as a said my first ir remote address was 0xfd01
my second ir remote  address is 0x03
my third ir remote address is 0x04
when i set your radio with my first remote can see conflict with my other remotes
for example if key 1 is set with address 0x01(0xfd01) and command 0x11
if my other remotes have a key with command 0x11 (with diffetrnt address)can trigger key 1 !!!

schreibfaul1, 2024-07-28:
The address and the command are each transmitted as 4 bytes. The first two bytes are the actual information and bytes 3 and 4 are their negation (0-&gt;F, 1-&gt;E, 2-&gt;D,...., F-&gt;0). Thus 0x00 is transmitted as 0x00FF and 0x16 as 0x16E9. Only the first two bytes are displayed.

Example: Address 0x40, command 0x01 is transmitted as 0x**40**BF**01**FE
[image]

valera39, 2024-07-30:
Dear Wolle!
After each firmware update I have to re-train the remote, in particular the digital buttons. Is this normal?

closed 2024-09-13 by github-actions[bot]