schreibfaul1/ESP32-MiniWebRadio #469: Remnants of a long title.

hevet, 2024-07-26:
If the information broadcast by the station is long, "garbage" remains at the bottom of the display. I am attaching a photo.
[image]

schreibfaul1, 2024-07-28:
Hi hevet,
it looks like it was caused by a character set with a big font. To find the cause I will create different "streamtitles". Unfortunately, that will take some time.

schreibfaul1, 2024-07-28:
I was able to reproduce this and it will be fixed in the next version.
[image]

schreibfaul1, 2024-07-28:
now it's better, the baseline for a Garamond font was too small and I added an additional margin for the textbox (bottom and right) [image]

hevet, 2024-07-29:
Hi Wolle. Thanks for solving the problem. :)

closed 2024-07-30 by hevet