schreibfaul1/ESP32-MiniWebRadio #470: No sound on some stations

valera39, 2024-07-29:
Dear Wolle!
After updating the audio library 3.0.12c, the sound on some stations disappeared. At the same time, the level indicator shows only on the left side. There are no errors in the port monitor. Please check for yourself
https://stream05.pcradio.ru/SSSR-med
https://live.radiospinner.com/clsscrdsc-96
https://stream05.pcradio.ru/rad_rdussrmscw-med
https://radio.voltagefm.ru:9009/VoltageUSSR_64_aac
http://pub0302.101.ru:8000/stream/pro/aac/64/161
[image]

schreibfaul1, 2024-07-29:
You have selected stations that transmit aacp or aac+.
My AAC decoder is too old for that, it can't do 'Parametric Stereo'. Thus aacp stations are decoded as mono.
But I was able to find an error that was hidden in the latest audio library. The left and right channels must always be controlled the same way.

P.S. If MISO is connected to your TFT display, you can create a hard copy simply click on the graphic.
[image]

valera39, 2024-07-29:
Dear Wolle!
I found another station that doesn't want to play with the updated audio library https://icecast-zvezda.mediacdn.ru/radio/zvezda/zvezda_128.
Here's the monitor log:
ESP32 Chip: ESP32-S3
Arduino Version: 3.0.2
ESP-IDF Version: 5.1.4
MiniWebRadio     Version 3.3c Jul 28/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 16777216 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 40 MHz
PSRAM total size: 8388608 bytes
RESET_REASON: Reset due to power-on event

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V3     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9488
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 15257.1 MB by 15279.0 MB free
[   760][I][esp32-hal-ledc.c:117] ledcAttachChannel(): LEDC attached to pin 10 (channel 0, resolution 8)
00:00:01 setup: ....  seek for stations.csv
00:00:01 setup: ....  stations.csv found
00:00:01 setup: ....  seek for WiFi networks
[  1806][I][WiFiMulti.cpp:86] addAP(): [WIFI][APlistAdd] add SSID: ASUS_AC66U     
00:00:07 setup: ....  6 WiFi networks found
00:00:07 setup: ....  TP-LINK_E345E0_RPT (-57)
00:00:07 setup: ....  ASUS_AC66U (-60)
00:00:07 setup: ....  ASUS_AC66U_RPT (-65)
00:00:07 setup: ....  TP-LINK_E345E0 (-81)
00:00:07 setup: ....  GG (-84)
00:00:07 setup: ....  Family (-91)
[ 14330][I][WiFiMulti.cpp:129] run(): [WIFI] scan done
[ 14336][I][WiFiMulti.cpp:134] run(): [WIFI] 7 networks found
[ 14342][I][WiFiMulti.cpp:249] run(): [WIFI] Connecting BSSID: 2C:FD:A1:C2:44:F0 SSID: ASUS_AC66U Channel: 2 (-61)
[ 14353][W][STA.cpp:537] disconnect(): STA already disconnected.
[ 14413][I][WiFiMulti.cpp:272] run(): [WIFI] Connecting done.
00:00:14 WiFI_info:   Connecting WiFi...
00:00:14 WiFI_info:   mDNS name: MiniWebRadio
00:00:14 setup: ....  connected to ASUS_AC66U, IP address is 192.168.1.206        
[ 14454][I][ArduinoOTA.cpp:129] begin(): OTA server at: MiniWebRadio.local:3232
00:00:14 ftpServer:   Buffers allocated: 4622 bytes
00:00:14 rtime_info:  Initializing SNTP
02:00:14 rtime_info:  Waiting for system time to be set... (1/10)
02:00:16 rtime_info:  Waiting for system time to be set... (2/10)
20:20:17 rtime_info:  20:20:17
20:20:17 audiotask:   is pinned to core 0
20:20:17 audiotask:   priority is 2
20:20:17 AUDIO_info:  commFMT = Philips
20:20:17 setup: ....  Number of saved stations: 93
20:20:17 setup: ....  current station number: 20
20:20:17 setup: ....  current volume: 3
20:20:17 setup: ....  last connected host: http://bookradio.hostingradio.ru:8069/fm
20:20:17 setup: ....  connection timeout: 2500 ms
20:20:17 setup: ....  connection timeout SSL: 3500 ms
20:20:18 action: ...  switch to station 20
20:20:18 AUDIO_info:  PSRAM found, inputBufferSize: 638965 bytes
20:20:18 AUDIO_info:  buffers freed, free Heap: 189756 bytes
20:20:18 AUDIO_info:  connect to: "icecast-zvezda.mediacdn.ru" on port 443 path "/radio/zvezda/zvezda_128"
[ 19817][E][ssl_client.cpp:36] _handle_error(): [ssl_starttls_handshake():313]: (-30592) SSL - A fatal alert message was received from our peer
[ 19831][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -30592
20:20:19 AUDIO_info:  Request https://icecast-zvezda.mediacdn.ru/radio/zvezda/zvezda_128 failed!
[ 19844][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 8 - ASSOC_LEAVE
20:20:19 AUDIO_info:  Request https://icecast-zvezda.mediacdn.ru/radio/zvezda/zvezda_128 failed!

valera39, 2024-07-29:
Is it no longer necessary to edit the configuration file in menuconfig?

valera39, 2024-07-29:
And I didn't find a "Run menuconfig" tab.

schreibfaul1, 2024-07-30:
https://icecast-zvezda.mediacdn.ru/radio/zvezda/zvezda_128 shows: 
[image]
I can't do anything about that, maybe a problem with the certificate, I'm not very familiar with SSL.

At the moment PIO is not able to provide the menuconfig for Arduino V3, I hope it will be possible again later. With the new version of the audioI2S lib the LWIP settings are not so important anymore.
I see you are using ESP32-S3, then you can extend the line in platformio.ini (for testing):
`https://github.com/schreibfaul1/ESP32-audioI2S.git#faad2-aac-decoder`
This replaces the aac decoder, then the stations listed above are in stereo.

valera39, 2024-07-30:
Dear Wolle!
I tested the audio library option you suggested. The above stations really started to work in stereo. But at the same time problems appeared on some other stations. I attach a part of the log:
ESP32 Chip: ESP32-S3
Arduino Version: 3.0.2
ESP-IDF Version: 5.1.4
MiniWebRadio     Version 3.3c Jul 28/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 16777216 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 40 MHz
PSRAM total size: 8388608 bytes
RESET_REASON: Reset due to power-on event

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V3     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9488
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 15257.1 MB by 15279.0 MB free
[   777][I][esp32-hal-ledc.c:117] ledcAttachChannel(): LEDC attached to pin 10 (channel 0, resolution 8)
00:00:01 setup: ....  seek for stations.csv
00:00:01 setup: ....  stations.csv found
00:00:01 setup: ....  seek for WiFi networks
[  1822][I][WiFiMulti.cpp:86] addAP(): [WIFI][APlistAdd] add SSID: ASUS_AC66U      
00:00:07 setup: ....  5 WiFi networks found
00:00:07 setup: ....  ASUS_AC66U (-60)
00:00:07 setup: ....  TP-LINK_E345E0_RPT (-63)
00:00:07 setup: ....  ASUS_AC66U_RPT (-67)
00:00:07 setup: ....  TP-LINK_E345E0 (-85)
00:00:07 setup: ....  ASUS_60 (-90)
[ 14347][I][WiFiMulti.cpp:129] run(): [WIFI] scan done
[ 14353][I][WiFiMulti.cpp:134] run(): [WIFI] 8 networks found
[ 14360][I][WiFiMulti.cpp:249] run(): [WIFI] Connecting BSSID: 2C:FD:A1:C2:44:F0 SSID: ASUS_AC66U Channel: 2 (-59)
[ 14371][W][STA.cpp:537] disconnect(): STA already disconnected.
[ 14440][I][WiFiMulti.cpp:272] run(): [WIFI] Connecting done.
00:00:14 WiFI_info:   Connecting WiFi...
00:00:14 WiFI_info:   mDNS name: MiniWebRadio
00:00:14 setup: ....  connected to ASUS_AC66U, IP address is 192.168.1.206
[ 14480][I][ArduinoOTA.cpp:129] begin(): OTA server at: MiniWebRadio.local:3232
00:00:14 ftpServer:   Buffers allocated: 4622 bytes
00:00:14 rtime_info:  Initializing SNTP
02:00:14 rtime_info:  Waiting for system time to be set... (1/10)
02:00:16 rtime_info:  Waiting for system time to be set... (2/10)
23:56:55 rtime_info:  23:56:55
23:56:55 audiotask:   is pinned to core 0
23:56:55 audiotask:   priority is 2
23:56:55 AUDIO_info:  commFMT = Philips
23:56:55 setup: ....  Number of saved stations: 93
23:56:55 setup: ....  current station number: 62
23:56:55 setup: ....  current volume: 1
23:56:55 setup: ....  last connected host: https://cast1.torontocast.com:1335/stream
23:56:55 setup: ....  connection timeout: 2500 ms
23:56:55 setup: ....  connection timeout SSL: 3500 ms
23:56:55 action: ...  switch to station 62
23:56:55 AUDIO_info:  PSRAM found, inputBufferSize: 638965 bytes
ms 18754
23:56:55 AUDIO_info:  buffers freed, free Heap: 189740 bytes
23:56:55 AUDIO_info:  connect to: "cast1.torontocast.com" on port 1335 path "/stream"
23:56:59 AUDIO_info:  SSL has been established in 3085 ms, free Heap: 148940 bytes
23:56:59 bitRate:     256Kbit/s
23:56:59 StationName: RMI Disco
23:56:59 icy-url: ..  http://cast1.torontocast.com:1335/stream
23:56:59 AUDIO_info:  MP3Decoder has been initialized, free Heap: 144424 bytes , free stack 2808 DWORDs
23:56:59 lastURL: ..  https://cast1.torontocast.com:1335/stream
23:56:59 AUDIO_info:  stream ready
23:56:59 AUDIO_info:  syncword found at pos 0
[... 139 lines cut ...]
00:40:17 AUDIO_info:  buffers freed, free Heap: 185524 bytes
00:40:17 AUDIO_info:  connect to: "91.189.162.134" on port 8008 path "/;"
00:40:17 AUDIO_info:  Connection has been established in 173 ms, free Heap: 185048 
bytes
00:40:17 StationName: Staroe Dobroe Radio
00:40:17 bitRate:     256Kbit/s
00:40:17 icy-url: ..  http://www.staroedobroe.ru/onair.php?ht=128
00:40:17 AUDIO_info:  MP3Decoder has been initialized, free Heap: 179440 bytes , free stack 3236 DWORDs
00:40:17 lastURL: ..  http://91.189.162.134:8008/;
00:40:17 AUDIO_info:  stream ready
00:40:17 AUDIO_info:  syncword found at pos 0
00:40:17 AUDIO_info:  Channels: 2
00:40:17 AUDIO_info:  SampleRate: 44100
00:40:17 AUDIO_info:  BitsPerSample: 16
00:40:17 AUDIO_info:  BitRate: 256000
00:40:18 Audiocodec:  MP3
00:40:19 AUDIO_info:  slow stream, dropouts are possible
00:40:20 AUDIO_info:  slow stream, dropouts are possible
00:40:21 AUDIO_info:  slow stream, dropouts are possible
00:40:22 AUDIO_info:  slow stream, dropouts are possible
00:40:23 AUDIO_info:  slow stream, dropouts are possible
00:40:25 AUDIO_info:  slow stream, dropouts are possible
What do you have to say about that?
I really hope you are able to resolve all the difficulties. For my part, I am ready to help to the best of my ability.
Thank you!

closed 2024-09-13 by github-actions[bot]