schreibfaul1/ESP32-MiniWebRadio #471: issue when playing FLAC via DLNA-Server

TubaFonse, 2024-07-29:
Dear Wolle - please help me: 
I cannot play a FLAC-File from my DLNA-Server (e.g. Win10-PC). 
Serial monitor (I deleted unnecessary infos ...): 
14:16:42 DLNA_server: title Am Böhmengässchen, childCount 0, itemSize 16991920, duration 0:03:02
...
14:17:25 AUDIO_info:  buffers freed, free Heap: 179344 bytes
14:17:25 AUDIO_info:  connect to: "192.168.178.29" on port 10243 path "/WMPNSSv4/3176339685/1_MTRfMjZiYjI2OWUtNDcxMTA"
14:17:25 AUDIO_info:  Connection has been established in 35 ms, free Heap: 179080 bytes
[683732][I][NetworkClient.cpp:565] connected(): Unexpected: RES: 0, ERR: 119
...
14:17:26 AUDIO_info:  Blechverr�ckt - Am B�hmeng�sschen
...
14:17:26 AUDIO_info:  Hostaddress is empty
14:17:56 AUDIO_info:  buffers freed, free Heap: 179340 bytes
14:17:56 AUDIO_info:  connect to: "192.168.178.29" on port 10243 path "/WMPNSSv4/3176339685/1_MTRfMjZiYjI2OWUtNDkwNTU"
14:17:56 AUDIO_info:  Connection has been established in 15 ms, free Heap: 179084 bytes
14:17:56 AUDIO_info:  FLACDecoder has been initialized, free Heap: 178148 bytes , free stack 2576 DWORDs
14:17:56 lastURL: ..  http://192.168.178.29:10243/WMPNSSv4/3176339685/1_MTRfMjZiYjI2OWUtNDkwNTU
14:17:56 Audiocodec:  FLAC
14:17:57 AUDIO_info:  stream ready, buffer filled in 1003 ms
14:17:57 AUDIO_info:  syncword found at pos 755
14:17:57 AUDIO_info:  syncword found at pos 0
[715499][E][Audio.cpp:1513] read_FLAC_Header(): Magic String 'fLaC' not found in header
[715513][I][NetworkClient.cpp:565] connected(): Unexpected: RES: 0, ERR: 119
14:17:57 AUDIO_info:  Channels: 2
14:17:57 AUDIO_info:  SampleRate: 44100
14:17:57 AUDIO_info:  BitsPerSample: 16
14:17:57 AUDIO_info:  BitRate: 288544
[715539][E][Audio.cpp:1328] readAudioHeader(): Processing stopped due to invalid audio header

Please let me know, if you need the unshortened trace to solve the issue. 
Many thanks, 
Alfons.

schreibfaul1, 2024-07-29:
Thanks for reporting. The last release of the audioI2S library was very extensive and there were some errors.
There is now a separate task for decoding, but it has to wait until the audio header has been read, otherwise there will be problems.
Please reload the libraries with 'Full Clean', then the DLNA files should work again.

TubaFonse, 2024-08-05:
Dear Wolle, above problem solved - many thanks!!!

closed 2024-08-05 by TubaFonse