schreibfaul1/ESP32-MiniWebRadio #473: Radio stops playing after round about 30min

HWH53, 2024-08-02:
Hallo Wolle,
mit der letzten Version v3.3c vom 29.07.2024 habe ich nun das Problem dass das Radio nach ca. 30 Minuten den Dienst einstellt. Gelegentlich lässt sich die Funktion über das Touchdisplay und die Senderumschaltung wieder reaktivieren. Das Display zeigt nur das Senderlogo, den Sendernamen, eine stehende Aussteuerungsanzeige und die Statusanzeigen am unteren Rand an. Seltsamerweise läuft die Uhr am oberen Rand weiter.
Weiterhin ist mir aufgefallen dass sich die Funktion DNLA nicht mehr bedienbar (ausschaltbar) ist sofern kein DNLA-Server konfiguriert ist und diese aus versehen aufgerufen wurde.

Konfiguration: ESP32-S3-DevKitC-1-N16R8, ILI9341, OTA....

 Ich hatte vermutet dass das Radio ungewollt in den sleepmode wechselt und daher die Funktion "sleep mode, show the time" aktiviert. Leider brachte dies auch keinen Hinweis.
Hast Du evtl. einen Tip für mich was ich prüfen sollte?

`
20:34:32 StreamTitle: ROCK ANTENNE - Rock Nonstop

20:34:37 StreamTitle: Fury In The Slaughterhouse - Won't forget these days
20:39:02 StreamTitle: Kid Kapichi - Can EU Hear Me
20:42:02 StreamTitle: Journey - Don't Stop Believin'
20:45:57 StreamTitle: A Life Divided - Best time
20:50:12 StreamTitle: ROCK ANTENNE - Rock Nonstop
20:50:17 StreamTitle: Rainbow - Long live rock'n'roll
20:54:17 StreamTitle: The Police - De do do do, de da da da
20:58:17 StreamTitle: Rock non stop - Doom tomb
20:59:12 StreamTitle: ROCK ANTENNE - Rock Nonstop
20:59:53 Time: ...... play Audiofile /voice_time/21_00.mp3
20:59:53 AUDIO_info:  buffers freed, free Heap: 193420 bytes
20:59:53 AUDIO_info:  Reading file: "/voice_time/21_00.mp3"
20:59:53 AUDIO_info:  MP3Decoder has been initialized, free Heap: 189016 bytes , free stack 4340 DWORDs
20:59:53 AUDIO_info:  Content-Length: 146378
20:59:53 AUDIO_info:  ID3 framesSize: 55
20:59:53 AUDIO_info:  ID3 version: 2.4
20:59:53 AUDIO_info:  ID3 extended header
20:59:53 AUDIO_info:  Audio-Length: 146323
20:59:53 AUDIO_info:  stream ready
20:59:53 AUDIO_info:  syncword found at pos 0
20:59:53 AUDIO_info:  Channels: 2
20:59:53 AUDIO_info:  SampleRate: 44100
20:59:53 AUDIO_info:  BitsPerSample: 16
20:59:53 AUDIO_info:  BitRate: 128000
20:59:54 AUDIO_FILE:  AudioCurrentTime 0:00s, AudioFileDuration 0:10s  
20:59:54 Audiocodec:  MP3
21:00:00 AUDIO_FILE:  AudioCurrentTime 0:06s, AudioFileDuration 0:08s  
21:00:01 AUDIO_info:  Closing audio file "21_00.mp3"
21:00:01 end of file: 21_00.mp3
21:00:01 AUDIO_info:  End of file "21_00.mp3"
21:00:01 AUDIO_info:  Connect to new host: "http://s5-webradio.antenne.de/rockantenne"
21:00:01 AUDIO_info:  buffers freed, free Heap: 193244 bytes
[29947905][E][NetworkManager.cpp:130] hostByName(): DNS Failed for 's5-webradio.antenne.de' with error '-54'
[29950418][I][NetworkClient.cpp:253] connect(): select returned due to timeout 2500 ms for fd 53
21:00:10 AUDIO_info:  Request http://s5-webradio.antenne.de/rockantenne failed!
[29950430][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 8 - ASSOC_LEAVE
21:00:10 AUDIO_info:  Request http://s5-webradio.antenne.de/rockantenne failed!
21:00:10 StationName:
21:00:10 StreamTitle:
`

schreibfaul1, 2024-08-03:
This is the DNS problem since Arduino V.3.0... I have implemented a workaround that keeps the DNS up to date. That helps.
[image]

HWH53, 2024-08-03:
Hallo Wolle,
danke für die schnelle Rückmeldung.
Ich habe gestern Abend und heute Morgen natürlich auch eine weitere Fehlersuche betrieben und folgendes festgestellt:
Sofern ich die Option "Time announcement on the hour" deaktiviere tritt das Problem nicht mehr auf. 
Das "time announcement" finde ich sehr interessant welches nach meiner Meinung zu jedem guten Rundfunkprogramm gehören sollte, aber von den Sendern leider nicht angewendet wird.
Eventuell hilft meine Info ja weiter.

mfg
Helmut

`09:59:53 Time: ...... Announcement at 10 o'clock is silent

10:03:16 StreamTitle: ROCK ANTENNE - Nachrichten
10:03:33 webSrv: ...  GetFolder "/"
10:05:26 StreamTitle: ROCK ANTENNE - Wetter
10:05:56 StreamTitle: ROCK ANTENNE - Rock Nonstop
10:06:36 StreamTitle: Van Halen - Why Can't This Be Love`

HWH53, 2024-08-03:
Hallo Wolle,
die von Dir angesprochene Änderung ist bereits in meinem Quellcode enthalten. Ich hatte die Version 3.3c kompl. neu gecloned und compiliert. Irgendwie differieren unsere Quellcode-Zeilennummern, ist dies korrekt?
Im 2. Anhang habe ich das "timeannouncement"  wieder aktiviert. Das Radio läuft weiter, aber die Uhrzeit steht. 
Eine Zeitansage erfolgte nicht. 
Kommt diese Ansage erst nach einem Reset? 
Nach einiger Zeit keine Anzeige mehr auf dem Display, Radio spielt weiter, Webserver arbeitet, aber keine Programmumschaltung möglich.

Falls ich irgendetwas falsch verstanden habe lasse es mich bitte wissen.

mfg
Helmut

[image]

[image]

schreibfaul1, 2024-08-03:
Hallo Helmut,
das ist genau der Fehler, WiFi vergisst nach wenigen Minuten die DNS Server. Wenn du dann auf eine andere Radiostation umschaltest oder die Verbindung neu aufbaust wird das Ziel nicht mehr gefunden. 
Du hast die Korrektur, aber sie wirkt nicht. Vielleicht was mit der Partitionstabelle?
Probier doch mal bitte einen "Erase Flash", dazu muss vorher der Monitor beendet werden, sonst funktioniert es nicht. 
[image]
Wenn du eine große Senderliste hast, mach vorher eine Sicherung, alles was im NVS steht ist nachher weg.

HWH53, 2024-08-03:
Hallo Wolle,
danke für deinen schnellen Hinweis. 
Ich habe den Flash gelöscht und anschließend den Upload durchgeführt. MWR konnte sich nicht mehr ins WLAN einloggen. Das Passwort wurde anscheinend nicht von der SD-Karte gezogen. Abhilfe schaffte nur die Hinterlegung von _SID und _PW in "common.h".  Anschließend habe ich die common.h wieder zurückgesetzt und das MWR loggte sich wieder normal ein.
Vorgang zur Verifikation wiederholt, Flash löschen, Code neu compilieren und hochladen. MWR loggt sich nicht ein.
Werden die Logindaten nach einem erfolgreichen Login irgendwo permanent im Flash abgelegt?
Erneuter Versuch:
Flash gelöscht, Code neu kompiliert und upgeloadet. Gleiche Erscheinung MWR loggt sich nicht ein. Nun die networks.csv auf der SD-Card bearbeitet, meine WLAN-Zugangsdaten an die 1. Position gesetzt (vorher an Position 4 hier waren die Positionen 1 - 3 mit * am Anfang auskommentiert. MWR loggt sich wieder ein!????
Erneuter Versuch: Flash löschen ...... MWR funktioniert.

Test über einige Stunden:
"timeannouncement" aktiviert. Zeitansage erfolgt wieder zur vollen Stunde. Der DNS-Reconnect funktioniert anscheinend auch wieder. Siehe beigefügtes Consolen Logfile.
Ich werde die Angelegenheit weiter beobachten und mich ggf. wieder melden.

danke nochmals mfg
Helmut
`
15:56:19 StreamTitle: ROCK ANTENNE - Rock Nonstop

15:59:34 StreamTitle: ROCK ANTENNE - Nachrichten
15:59:53 Time: ...... play Audiofile /voice_time/16_00.mp3
15:59:53 AUDIO_info:  buffers freed, free Heap: 187168 bytes
15:59:53 AUDIO_info:  Reading file: "/voice_time/16_00.mp3"
15:59:53 AUDIO_info:  MP3Decoder has been initialized, free Heap: 182760 bytes , free stack 4452 DWORDs
15:59:53 AUDIO_info:  Content-Length: 144525
15:59:53 AUDIO_info:  ID3 framesSize: 55
15:59:53 AUDIO_info:  ID3 version: 2.4
15:59:53 AUDIO_info:  ID3 extended header
15:59:53 AUDIO_info:  Audio-Length: 144470
15:59:53 AUDIO_info:  stream ready
15:59:53 AUDIO_info:  syncword found at pos 0
15:59:53 AUDIO_info:  Channels: 2
15:59:53 AUDIO_info:  SampleRate: 44100
15:59:53 AUDIO_info:  BitsPerSample: 16
15:59:53 AUDIO_info:  BitRate: 128000
15:59:54 AUDIO_FILE:  AudioCurrentTime 0:00s, AudioFileDuration 0:11s  
15:59:54 Audiocodec:  MP3
16:00:00 AUDIO_FILE:  AudioCurrentTime 0:06s, AudioFileDuration 0:08s  
16:00:01 AUDIO_info:  Closing audio file "16_00.mp3"
16:00:01 end of file: 16_00.mp3
16:00:01 AUDIO_info:  End of file "16_00.mp3"
16:00:01 AUDIO_info:  buffers freed, free Heap: 186796 bytes
16:00:01 AUDIO_info:  connect to: "s2-webradio.antenne.de" on port 80 path "/rockantenne"
16:00:01 AUDIO_info:  Connection has been established in 65 ms, free Heap: 186468 bytes
16:00:02 bitRate:     128Kbit/s
16:00:02 icy-descr:   ROCK ANTENNE
16:00:02 StationName: ROCK ANTENNE
16:00:02 icy-url: ..  http://www.rockantenne.de
16:00:02 AUDIO_info:  MP3Decoder has been initialized, free Heap: 185456 bytes , free stack 4452 DWORDs
16:00:02 lastURL: ..  http://s2-webradio.antenne.de/rockantenne
16:00:02 AUDIO_info:  stream ready
16:00:02 Audiocodec:  MP3
16:00:02 AUDIO_info:  syncword found at pos 0
16:00:02 AUDIO_info:  Channels: 2
16:00:02 AUDIO_info:  SampleRate: 44100
16:00:02 AUDIO_info:  BitsPerSample: 16
16:00:02 AUDIO_info:  BitRate: 128000
16:00:02 StreamTitle: ROCK ANTENNE - Nachrichten`

rjshiffer, 2024-08-03:
I think I having the same DNS problem, but it presents itself differently.
I noticed that running Version 3.3c, the radio takes much longer before the station plays.  I was seeing this in the log:
```
.
.
00:00:10 WiFI_info:   Connecting WiFi...
00:00:10 WiFI_info:   mDNS name: MiniWebRadio
00:00:10 setup: ....  connected to rigel, IP address is 10.0.0.119
[ 10785][I][ArduinoOTA.cpp:129] begin(): OTA server at: MiniWebRadio.local:3232
00:00:10 ftpServer:   Buffers allocated: 4622 bytes
00:00:10 rtime_info:  Initializing SNTP
18:00:10 rtime_info:  Waiting for system time to be set... (1/10)
18:00:12 rtime_info:  Waiting for system time to be set... (2/10)
18:00:14 rtime_info:  Waiting for system time to be set... (3/10)
18:00:16 rtime_info:  Waiting for system time to be set... (4/10)
18:00:18 rtime_info:  Waiting for system time to be set... (5/10)
18:00:20 rtime_info:  Waiting for system time to be set... (6/10)
18:00:22 rtime_info:  Waiting for system time to be set... (7/10)
18:00:24 rtime_info:  Waiting for system time to be set... (8/10)
18:00:26 rtime_info:  Waiting for system time to be set... (9/10)
18:00:28 rtime_info:  18:00:28
18:00:28 connection to NTP failed, trying again
...[ESPrestart]...
[repeats 1 or more times before successful
```

Thinking it must be my connection to time servers, I first attempted
to change time servers in platformio.ini without any resolution. 
After other attempts to fix the problem, I simply entered
the servers' IP address in platformio.ini which worked.  

Wolle, after seeing your comment about the DNS problem, I added the same code in RTIME::obtain_time in rtime.cpp (see below) and use the original time server names in platformio.ini.  

After adding "IPAddress DSN code" to RTIME::obtain_time:

```
00:01:07 WiFI_info:   Connecting WiFi...
00:02:05 WiFI_info:   mDNS name: MiniWebRadio
00:02:05 setup: ....  connected to rigel, IP address is 10.0.0.119
[ 10930][I][ArduinoOTA.cpp:129] begin(): OTA server at MiniWebRadio.local:3232
00:02:05 ftpServer:   Buffers allocadted: 4622 bytes
00:02:05 rtime_info:  Initializing SNTP
18:02:05 rtime_info:  Waiting for system time to be set... (1/10)
13:32:08 rtime_info:  13:32:08
13:32:08 audiotask:   is pinned to core 0
13:32:08 audiotask:   priority is 2
...normal operation...
```
```
boolean RTIME::obtain_time(){
    time_t now = 0;
    int32_t retry = 0;
    const int32_t retry_count = 10;
    while(timeinfo.tm_year < (2016 - 1900) && ++retry < retry_count) {
>>->>>> IPAddress dns1(8, 8, 8, 8); IPAddress dns2(8, 8, 4, 4); WiFi.setDNS(dns1, dns2);
        sprintf(sbuf, "Waiting for system time to be set... (%ld/%ld)", (long int)retry, (long int)retry_count);
        if (RTIME_info) RTIME_info(sbuf);
        vTaskDelay(uint16_t(2000 / portTICK_PERIOD_MS));
        time(&now);
        localtime_r(&now, &timeinfo);
    }
    localtime_r(&now, &timeinfo);
    strftime(strftime_buf, sizeof(strftime_buf), "%c", &timeinfo);
    if (RTIME_info) RTIME_info(strftime_buf);

    //log_i( "The current date/time in Berlin is: %s", strftime_buf);
    if(retry < retry_count) return true;
    else return false;
}
```

I don't know if that is the most elegant solution (workaround).  I leave it to
you for a final fix.

Thanks for all your work.

Richard

schreibfaul1, 2024-08-04:
Thanks for your tip. The problem with the DNS is annoying. Espressif released a new Arduino version the day before yesterday. Apparently the bug has been fixed. The new version is V3.0.4, you can change the platform.ini.
````
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.4
    platformio/framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.4/esp32-arduino-libs-3.0.4.zip
````

rjshiffer, 2024-08-04:
Hi Wolle,

Thanks for the super fast response.  I removed my change to rtime.cpp, updated platformio.ini as you provided, and uploaded.

No delay at startup and radio operating normally.  Great!

Thanks,
Richard

HWH53, 2024-08-05:
Hallo Wolle,
danke für die Info's!
Das DNS-Problem ist anscheinend mit dem Update der Arduino-Espressif Frameworks auf v3.04 behoben. Die stündliche Zeitansage funktioniert nun wieder zuverlässig ohne Verlust der eingestellten Station. Ebenfalls ist eine deutliche Geschwindigkeitsteigerung beim Start des MWR festzustellen.

nochmals vielen Dank für die gelungene Entwicklung von MWR!
mfg
Helmut

`--- More details at https://bit.ly/pio-monitor-filters

--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
15:59:53 Time: ...... play Audiofile /voice_time/16_00.mp3 
15:59:54 AUDIO_info:  buffers freed, free Heap: 191156 bytes
15:59:54 AUDIO_info:  Reading file: "/voice_time/16_00.mp3"
15:59:54 AUDIO_info:  MP3Decoder has been initialized, free Heap: 186752 bytes , free stack 4452 DWORDs
15:59:54 AUDIO_info:  Content-Length: 144525
15:59:54 AUDIO_info:  ID3 framesSize: 55    
15:59:54 AUDIO_info:  ID3 version: 2.4   
15:59:54 AUDIO_info:  ID3 extended header
15:59:54 AUDIO_info:  Audio-Length: 144470
15:59:54 AUDIO_info:  stream ready
15:59:54 AUDIO_info:  syncword found at pos 0
15:59:54 AUDIO_info:  Channels: 2
15:59:54 AUDIO_info:  SampleRate: 44100
15:59:54 AUDIO_info:  BitsPerSample: 16
15:59:54 AUDIO_info:  BitRate: 128000
15:59:54 AUDIO_FILE:  AudioCurrentTime 0:00s, AudioFileDuration 0:11s  
15:59:54 Audiocodec:  MP3
16:00:00 AUDIO_FILE:  AudioCurrentTime 0:06s, AudioFileDuration 0:08s  
16:00:01 AUDIO_info:  Closing audio file "16_00.mp3"
16:00:01 end of file: 16_00.mp3
16:00:01 AUDIO_info:  End of file "16_00.mp3"
16:00:01 AUDIO_info:  buffers freed, free Heap: 190980 bytes
16:00:01 AUDIO_info:  connect to: "s5-webradio.antenne.de" on port 80 path "/rockantenne"
16:00:02 AUDIO_info:  Connection has been established in 57 ms, free Heap: 190684 bytes
16:00:02 bitRate:     128Kbit/s
16:00:02 icy-descr:   ROCK ANTENNE
16:00:02 StationName: ROCK ANTENNE
16:00:02 icy-url: ..  http://www.rockantenne.de
16:00:02 AUDIO_info:  MP3Decoder has been initialized, free Heap: 189456 bytes , free stack 4452 DWORDs
16:00:02 lastURL: ..  http://s5-webradio.antenne.de/rockantenne
16:00:02 AUDIO_info:  stream ready
16:00:02 Audiocodec:  MP3
16:00:02 AUDIO_info:  syncword found at pos 0
16:00:02 AUDIO_info:  Channels: 2
16:00:02 AUDIO_info:  SampleRate: 44100
16:00:02 AUDIO_info:  BitsPerSample: 16
16:00:02 AUDIO_info:  BitRate: 128000
16:00:02 StreamTitle: ROCK ANTENNE - Rock Nonstop
16:00:03 StreamTitle: ROCK ANTENNE - Nachrichten
`

closed 2024-08-05 by HWH53

schreibfaul1, 2024-08-06:
Danke für Deine Rückmeldung. Die V3.0.4 scheint stabil zu sein. Ich habe die platformio.ini  angepasst.
vG Wolle