schreibfaul1/ESP32-MiniWebRadio #475: Some issues on an Ai-Thinker AudioKit ESP32 A1S board combined with an ST7796 display, and their fixes

frank-hx, 2024-08-05:
_Issue 1:_
On an Ai-Thinker AudioKit ESP32 A1S board combined with an ST7796 display I encountered these errors while booting and nothing else happened at all:
`rst:0x10 (RTCWDT_RTC_RESET),boot:0x3b (SPI_FAST_FLASH_BOOT)`
`invalid header: 0xffffffff`

Turns out IO12 is the bootstrap pin on the ESP32 A1S board, and being connected to an output of the ST7796 display causes the errors.

So I swapped IO12 and IO13 both physically and in common.h and now these errors are gone.

_Issue 2:_
Now my board booted, but unfortunately I got into these following runtime errors: 
`[E][Wire.cpp:449] endTransmission(): NULL TX buffer pointer`.

In order to get it working I had to modify `ES8388::write_reg`
from `return Wire.endTransmission() == 0;`
to `return 0 == _TwoWireInstance->endTransmission(true);` (Just like in `AC101::WriteReg`)

Just to be sure I also based `ES8388::read_reg` on `AC101::ReadReg`.

_Issue 3:_
By now my errors were gone, but audio still wasn't working.

So in `main.cpp` I modified `ES8388 dac(&i2cBusTwo);` to `ES8388 dac(&i2cBusOne);`

Now the ES8388 DAC is configured and working. 

_Issue 4:_
I also experienced that my ST7796 display sometimes wouldn't start correctly, the display then stays white. It was caused by a floating RES(ET) pin on the ST7796 which apparently has no pull-up resistor on it. So now that's also sorted.

_Boot speed improvement:_
In order to speed up the booting process I disabled the WiFi scan in `main.cpp`:

`/*int16_t n = WiFi.scanNetworks();`
`SerialPrintfln("setup: ....  " ANSI_ESC_WHITE "%i WiFi networks found", n);`
`for(int i = 0; i < n; i++){`
`    SerialPrintfln("setup: ....  " ANSI_ESC_GREEN "%s (%d)", WiFi.SSID(i).c_str(), (int16_t)WiFi.RSSI(i));`
`}*/`

And to disable waiting for NTP sync, also in `main.cpp`, I replaced `return obtain_time();` by `return true;`. Time now gets in sync just after the display is updated after finished booting.

Now it takes about 11 to 14 seconds from cold boot to the first music.

_Finally:_
Everything seems to work and now it's time to enjoy this nice WebRadio.

Although these issues might be specific to my setup I hope it can be helpful to improve the code or assist someone encountering similar issues.

schreibfaul1, 2024-08-06:
Hello,
thank you for your experience report. I only have the old board from AI-Thinker with the AC101. The ES8388 code is taken from a forum. If you can please provide the updated ES8388 files I will add them to the project.
I will look at your suggestions for changes in scanNetworks(); and NTP soon.

schreibfaul1, 2024-08-06:
thank you, these are good suggestions

frank-hx, 2024-08-06:
>[...] If you can please provide the updated ES8388 files I will add them to the project. [...]

Yes of course, I'm glad to help. 

I added `ES8388.cpp`, `main.cpp` and `common.h` in a zip file because these extensions aren't supported for upload: [Modified files for ES8388.zip](https://github.com/user-attachments/files/16514251/Modified.files.for.ES8388.zip)

There may still be leftover remnants of various attempts to fix the issues.

frank-hx, 2024-08-07:
In commit [b6b5a67](https://github.com/schreibfaul1/ESP32-MiniWebRadio/commit/b6b5a6709f7361e18b851bd66609ab6b81f68480) you accidentally added the wrong config in the updated document [AI Thinker A1S Board.pdf](https://github.com/schreibfaul1/ESP32-MiniWebRadio/commit/b6b5a6709f7361e18b851bd66609ab6b81f68480#diff-e98dc28656addd0402671411dbe979606ab3f5277066c3393377f46bca98a49a). It should be:

```
#ifdef CONFIG_IDF_TARGET_ESP32
    // Digital I/O used
        #define TFT_CS              22
        #define TFT_DC              5
        #define TFT_BL              -1 // at -1 the brightness menu is not displayed
        #define TP_IRQ              13
        #define TP_CS               12 // IO12 is bootstrap on Ai-Thinker AudioKit ESP32 A1S board. When connected to PEN-pin of the display it causes these boot-errors: "rst:0x10 (RTCWDT_RTC_RESET),boot:0x3b (SPI_FAST_FLASH_BOOT) #CR# invalid header: 0xffffffff"
        #define SD_MMC_D0           2 // cannot be changed
        #define SD_MMC_CLK          14 // cannot be changed
        #define SD_MMC_CMD          15 // cannot be changed
        #define IR_PIN              -1
        #define TFT_MOSI            23 // TFT and TP (VSPI)
        #define TFT_MISO            19 // TFT and TP (VSPI)
        #define TFT_SCK             18 // TFT and TP (VSPI)
        #define I2S_DOUT            26 // pin 25 AC101, pin 26 ES8388
        #define I2S_DIN             35 // pin not used
        #define I2S_BCLK            27
        #define I2S_LRC             25 // pin 26 AC101, pin 25 ES8388
        #define I2S_MCLK            0
        #define I2C_DAC_SDA         33 // some DACs are controlled via I2C
        #define I2C_DAC_SCL         32
        #define SD_DETECT           34 // some pins on special boards: Olimex, A1S ...
        #define HP_DETECT           39
        #define AMP_ENABLED         21
        #define BT_EMITTER_RX       -1 // TX pin - KCX Bluetooth Transmitter
        #define BT_EMITTER_TX       -1 // RX pin - KCX Bluetooth Transmitter
        #define BT_EMITTER_LINK     -1 // high if connected
        #define BT_EMITTER_MODE     -1 // high transmit - low receive
        #define BT_EMITTER_CONNECT  -1 // awake after shutdown
#endif

```

schreibfaul1, 2024-08-07:
Exactly, I was confusing ESP32-S3 with ESP32. Thank you for pointing this out.