schreibfaul1/ESP32-MiniWebRadio #479: brightness controll in v3.3

clear-sky, 2024-08-06:
Hi Dear Wolle!
thanks for v3.3
have you removed brightness control in v3.3?

schreibfaul1, 2024-08-06:
Hello clear-sky, you do not need to set the defines to -1. There is an automatic detection of what is connected to the I2C bus. If the bus remains free, this should not be a problem. Other devices, such as a capacitive touchpad or a rotary encoder, can be connected here later.
The Brightness button is the same as before
[image]
The regulation now makes a slider
[image]

clear-sky, 2024-08-06:
Hi Dear Wolle!
thanks for the answer-but some thing could be wrong!
my lcd does not have touch thus i edited setting.json file in sdcard as below:
`{"volume":2,"ringvolume":21,"alarmtime":180,"alarm_weekdays":0,"timeAnnouncing":true,"mute":false,"brightness":5,"sleeptime":0,"lastconnectedhost":"http://server4.streamserver24.com:26237/stream","station":5,"sumstations":14,"toneLP":0,"toneBP":0,"toneHP":0,"balance":0,"Timezone_Name":"Europe/Berlin","Timezone_String":"CET-1CEST,M3.5.0,M10.5.0/3","timeFormat":24}`

as you see brightness is set to 5 but my lcd brightness is max and seems radio does not reed brightness value from this file

schreibfaul1, 2024-08-06:
Indeed, there is a mistake.

clear-sky, 2024-08-07:
Hi Dear Wolle!
thanks problem solved!
what is your opinion to bring brightness control to web interface?

closed 2024-08-09 by clear-sky