schreibfaul1/ESP32-MiniWebRadio #480: DLNA Playlist don't work (bug - or feature request?)

TubaFonse, 2024-08-06:
Hallo Wolle, 
ich schaffe es nicht, eine Playlist von meinen DLNA-Medienservern am MWR abzuspielen.  
Ich habe es über Windows-PC, meine NAS bzw. FritzBox (Medienserver) vergeblich versucht. 
Gibt es da eine Möglichkeit bzw. Tricks, über DLNA eine .m3u oder .m3u8 Playlist ähnlich wie auf der SD-Karte zu nutzen? 

Vielen Dank und Grüße, 
Alfons

schreibfaul1, 2024-08-06:
Hallo Alfons,
eine Widergabeliste von Hand zu schreiben ist nicht ganz einfach. Aber du kannst den VLC Player dazu verwenden. Du gehst auf "Universal Plug & Play" und dann suchst du dir die Titel von der Fritzbox aus und fügst sie der Wiedergabeliste zu.
[image]
Dann auf Medien/Wiedergabeliste in Datei speichern als *.m3u.
Die m3u sieht ann in einem Editor etwa so aus
[image]
Und dann speicherst du die *.m3u auf der SD Karte.

TubaFonse, 2024-08-06:
Hallo Wolle, 
habe es gleich ausprobiert, so funktioniert es auch mit meinen Servern, danke! 
Grüße, Alfons

closed 2024-08-06 by TubaFonse