schreibfaul1/ESP32-MiniWebRadio #481: No Sound since update to Version 3.3g Aug 06/2024

HWH53, 2024-08-07:
Hallo Wolle,
ich habe auf die neue  Version 3.3g upgedatet. Seitdem habe ich keinen Sound mehr, Volume steht auf "0" und lässt sich weder über den Slider im Menue noch über die Weboberfläche verändern. Keine auffällige Meldung im Logfile. Ein komplettes Clonen des Repsositories erbrachte auch keine Veränderung.
Übersehe ich irgendeine Einstellung? Die SD-Karte habe ich noch nicht upgedatet, ist die erforderlich? 
Ich bin auf die Version 3.3c zurückgegangen, diese funktioniert mit dem Framework-Update bei mir zuverlässig.

mfg
Helmut

schreibfaul1, 2024-08-07:
Hallo Helmut,
ab jetzt können die Stufen für die Lautstärke zwischen 21 und 255 eingestellt werden. Damit das etwas komfortabel ist habe ich die Vol+, Vol- Buttons gegen Schieberegler ausgetauscht. Die Lautstärkestufen dürfen nicht kleiner als 21 werden, sind in deinem Fall wahrscheinlich 0. Das werde ich im nächsten Update prüfen und korrigieren. Gehe bitte auf die Einstellungen und setzt die Lautstärkestufen von Hand.

HWH53, 2024-08-07:
Hallo Wolle,
danke für die schnelle Rückmeldung! 
Es war wirklich so, die Lautstärkestufen standen in "settings.json" auf Null. Der neue Menuepunkt " Einstellungen wurde bei mir nicht angezeigt, da ich den Browser-Cash nicht geleert hatte und in der alten Einstellung war dies unter "i" nicht sichtbar.
Meistens sitzt das Problem vor dem Bildschirm!🤔

[image]

mfg
Helmut

closed 2024-08-07 by HWH53