schreibfaul1/ESP32-MiniWebRadio #482: Problem with playback of some stations on newer versions

valera39, 2024-08-08:
Hi Dear Wolle!
I upgraded to the new version and am having some problems playing stations that played flawlessly on previous versions of the audioI2S library
Please check these stations for yourself:  
http://91.189.162.134:8008/;
http://harddanceradio.ddns.is74.ru:8000/eurodisco
https://radiolegenda.ru:8001/64.aac
The entire log at http://harddanceradio.ddns.is74.ru:8000/eurodisco is full of these entries:
13:37:57 AUDIO_info:  syncword found at pos 0
13:37:57 AUDIO_info:  AAC decode error -6 : Channel coupling not yet implemented
13:37:57 AUDIO_info:  syncword found at pos 0
13:37:57 AUDIO_info:  AAC decode error -6 : Channel coupling not yet implemented
13:37:57 AUDIO_info:  syncword found at pos 0
13:37:57 AUDIO_info:  AAC decode error -6 : Channel coupling not yet implemented   
13:37:57 AUDIO_info:  syncword found at pos 0
13:37:57 AUDIO_info:  AAC decode error -32 : Bitstream value not allowed by specification
Part of the monitor log from another thread:
13:44:01 lastURL: ..  http://91.189.162.134:8008/;
13:44:01 AUDIO_info:  stream ready
13:44:01 AUDIO_info:  syncword found at pos 0
13:44:01 AUDIO_info:  Channels: 2
13:44:01 AUDIO_info:  SampleRate: 44100
13:44:01 AUDIO_info:  BitsPerSample: 16
13:44:01 AUDIO_info:  BitRate: 256000
13:44:01 Audiocodec:  MP3
13:44:01 webSrv: ...  File not found "/logo/.jpg"
13:44:03 AUDIO_info:  slow stream, dropouts are possible
13:44:20 AUDIO_info:  slow stream, dropouts are possible
13:44:21 AUDIO_info:  slow stream, dropouts are possible
13:44:22 AUDIO_info:  slow stream, dropouts are possible
13:44:26 AUDIO_info:  slow stream, dropouts are possible
13:44:27 AUDIO_info:  slow stream, dropouts are possible
13:44:28 AUDIO_info:  MP3 decode error -9 : INVALID_HUFFCODES
13:44:28 AUDIO_info:  syncword found at pos 551
13:44:28 AUDIO_info:  syncword found at pos 0
13:44:28 AUDIO_info:  MP3 decode error -9 : INVALID_HUFFCODES
13:44:28 AUDIO_info:  syncword found at pos 184
13:44:28 AUDIO_info:  syncword found at pos 0
13:44:28 AUDIO_info:  MP3 decode error -8 : INVALID_SCALEFACT
13:44:28 AUDIO_info:  syncword found at pos 98
13:44:28 AUDIO_info:  syncword found at pos 0
In general, the line "slow stream, dropouts are possible" became present much more often and on some other streams, although the wifi signal level did not change. 
I would also like to wish you to deal with threads that use SSL, such as https://icecast-zvezda.mediacdn.ru/radio/zvezda/zvezda_128.
And please explain what the line in the monitor log means during boot time when pressing reset on esp32: 
rst:0x1 (POWERON),boot:0x23 (DOWNLOAD(USB/UART0))
waiting for download
I have to click several times to load the radio normally.
Thank you!

schreibfaul1, 2024-08-08:
Hello valera39, 
the stream http://91.189.162.134:8008/; comes in badly, I don't think I can do much about it.

The other two stations show an error in the AAC decoder. This should not be the case. The new AAC decoder is the faad2, which has replaced the helix decoder. Faad2 is still used today in some Linux systems. I think I made a mistake when porting it to the ESP32. Troubleshooting will not be easy and will certainly take a lot of time.

We are now using the IDF Framework V5.1. I don't want to go back to V4... Unfortunately, PlatformIO does not currently allow "Arduino as a component" in this version. I hope this will soon be possible again so that the WiFi settings can be made in menuconfig. That would bring more speed. 

rst:0x1 (POWERON),boot:0x23 (DOWNLOAD(USB/UART0))
I don't know what the cause is. There is now a monitor filter that saves the log to a file. Maybe this is the reason? Please remove the entry `log2file` from platfornio.ini

valera39, 2024-08-08:
Thank you, I look forward to your corrections

schreibfaul1, 2024-08-08:
http://harddanceradio.ddns.is74.ru:8000/eurodisco
https://radiolegenda.ru:8001/64.aac
After "full clean" these streams work now

schreibfaul1, 2024-08-08:
https://icecast-zvezda.mediacdn.ru/radio/zvezda/zvezda_128
I can't fix the SSL problem, maybe it will be solved with the next Espressif update. Please use the URL without encryption until then
http://icecast-zvezda.mediacdn.ru/radio/zvezda/zvezda_128

closed 2024-09-22 by github-actions[bot]