schreibfaul1/ESP32-MiniWebRadio #483: Not a problem, just a question: about mqtt

Jefke1, 2024-08-09:
Dear All,
how can I connect the mqtt of the mini_web_radio with nodeRed. Maybe a stupid question for many, but I just started it and still want to give it a try. Any help is welcome, thanks in advance.

schreibfaul1, 2024-08-09:
There is a project that uses similar components to the MWR. The project is called [ESPuino](https://github.com/biologist79/ESPuino) MQTT is used there. The source code is available, you may find a good introduction to the topic. There is also a [Forum](https://forum.espuino.de/), which you can search for MQTT.

Jefke1, 2024-08-09:
> There is a project that uses similar components to the MWR. The project is called [ESPuino](https://github.com/biologist79/ESPuino) MQTT is used there. The source code is available, you may find a [...]

thanks.
grt Jef.

closed 2024-09-23 by github-actions[bot]