schreibfaul1/ESP32-MiniWebRadio #484: Radio is not working without Internet - Not an  issue - just a question.

FranZatGIT, 2024-08-10:
Hi Wolle,

after finishing the HW stuff now concentrate again more to the SW part. 

**What is the reason for the ESP reboot command when NTP is not found?  And why not starting with the Player screen without internet  (with AP running to setup internet)?**

just experimental - I comment out the ESP reboot when NTP is not found and I changed the return with a break after the AP function is called :
[image]

The radio now displays the AP screen for a few seconds and then switches to normal radio mode - playing MP3s from SD without Internet is possible as desired...

of course time is not correct and IP is shown as 0.0.0.0 but the AP works and setting up the correct WIFI  over WEB UI is still possible .

**Is there a reason why the Radio is limited to work with internet access only?**

Background in my case are also some additional functions I implemented ( Power MGMT, Speaker relay, Bluetoothmode,... ) They are not working when the programm is stuck in the setup phase and only AP WEB UI is working.

Thank you Franz

kerat71, 2024-08-14:
It's sounds very reasonable to me

schreibfaul1, 2024-08-14:
I have seen your suggestion, but it has been forgotten again. In many cases, the code is still from the early days. The attempt to connect to the WiFi network three times also seems outdated today. I will change this in the next update.

FranZatGIT, 2024-08-15:
Hi Wolle, I would have some other suggestion and ideas for the Radio. Would you like me to communicate that over the "issues"  here on Github or 1:1 e.g. via email....?

By the way - have you seen my question in Post https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/476

many Thanks Franz

schreibfaul1, 2024-08-16:
Sorry, I haven't seen #476 yet. Neat work, even with the solar cell! May I add some pictures to the gallery https://github.com/schreibfaul1/ESP32-MiniWebRadio/wiki/User-devices? I'm always interested in more ideas, feel free to post here in Github.

FranZatGIT, 2024-08-16:
sure, It would be honored if you would add the pictures to your gallery.

closed 2024-09-30 by github-actions[bot]