schreibfaul1/ESP32-MiniWebRadio #485: SD card connection is lost

valera39, 2024-08-12:
Hello Dear Wolle, hello everyone!
I noticed a strange peculiarity. 
After a long time in sleep mode, ESP as if loses connection with the memory card.
It is expressed in the fact that the time on the display freezes, the separator of hours and minutes does not blink. On the signal from the remote control turns on, but there is no station icon. Then if you turn the remote off again, the radio goes to sleep, but the clock
is no longer displayed on the full screen. At the same time, the radio wakes up normally on the alarm clock at the appointed time, but without the alarm clock and radio station icons.
If I turn the power off after this problem and then turn it back on, the screen shows that the SD card is not detected.
What I did: changed ESP to another, changed card readers with and without pull-up resistors, changed SD cards, formatted them and checked for errors, changed SD card frequency, soldered all connections.
But the problem occurs periodically. Log from the COM port is difficult to remove because of the design of the radio, but I will try to do it.
What can it be, any ideas?

schreibfaul1, 2024-08-12:
This is just a guess, but lower the frequency of the SD card.

valera39, 2024-08-12:
Yeah, sure, I've been down to 40MHz.

valera39, 2024-08-15:
For two days now, the radio has been in sleep mode as a clock while the memory card has not fallen off.

closed 2024-09-29 by github-actions[bot]