schreibfaul1/ESP32-MiniWebRadio #486: volume control

valera39, 2024-08-15:
Hello Dear Wolle!
In recent updates, you added the ability to change the volume level to 200.
I thought that for a smoother adjustment the volume would change logarithmically, for example like this: 1,2,3,4,5,6,7,8.......100. 
[image]
That is, at low volume levels to have a "long" adjustment. 
Is this possible?

schreibfaul1, 2024-08-15:
Hi valera39,
the volume controls on analog radios have a logarithmic characteristic, which would also work here. There are two curves, a quadratic one that is currently used and a logarithmic one. Now the log would be better. I'll turn it on in the next update. 
[image]

valera39, 2024-08-16:
Thanks, I'll look forward to updates

valera39, 2024-08-17:
Updated, the volume control works very well now.
Thanks!

closed 2024-08-17 by valera39