schreibfaul1/ESP32-MiniWebRadio #491: Multiple Feature request/ Issue - Player

FranZatGIT, 2024-08-16:
Here some Ideas to make the player more intuitive:

Feature:
1. The player should contain the "next" or "previous song" arrows buttons  also when playing a song.( Now the Player  have to be stopped to go to the next song)
2. The "list" button should be also visible while playing a song, to search/select  a next song while listening
3. The "Play all" songs of a folder should be enabled by default. 
4. The Player should remember the last played song/folder after power off/on 
5. I saw in serial monitor the passing time of a MP3 - it would be cool if that would be shown in display as well.
6. As mentioned in https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/484 The Radio should start also without internet (tested that in my customized code and it works) The Radio should directly start into the player mode (Radio mode makes no sense without Internet)
7. Play/Stop, Next, Previous Song would be cool to be on the IR remote control  and Web UI

Issue:
The radio crashes if no Audio file (just Folders) is in the "audiofile" folder

FranZatGIT, 2024-09-02:
Hi Wolle,

I saw in the new 3.4 version some changes in the player. However what do you think about my ideas? If something is unclear just let me know...

schreibfaul1, 2024-09-05:
changes: https://github.com/schreibfaul1/ESP32-MiniWebRadio/pull/503

closed 2024-10-21 by github-actions[bot]