schreibfaul1/ESP32-MiniWebRadio #493: WiFi errors attempting to use a BT Emitter

rjshiffer, 2024-08-22:
Hi Wolle and others,

I'm unable to get the radio to operate correctly when the BT Emitter is attached. I've tried with both an ESP32 and ESP32-S3.  I get numerous WiFi error messages and continuous re-boots. There seems to be a conflict between WiFi and the BT Emitter. I soldered the KCX_BT_Emitter to a PCB same as the one shown in a previous issue (see my picture). Only very, very, occasionally do I get the BT Emitter to connect to a remote speaker and then only for a few seconds.

Suspecting the Emitter/PCB soldering, etc., I checked with a multimeter to ensure proper connectivity with no solder bridges, etc. No problems are detected.

All my connections are as shown in the ESP32 and ESP32-S3 schematic drawings in the docs folder.

I'm at a loss as to my problem.   Could you please give me a pointer as to where I could look next.

Thanks.
Richard 

(I hope I did not overwhelm you with the volume of serial monitor output.)

**** ESP32 Monitor Output with BT Emitter Connected -- Failure *******

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4832
load:0x40078000,len:16440
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3504
entry 0x400805cc

ESP32 Chip: ESP32-D0WDQ6
Arduino Version: 3.0.4
ESP-IDF Version: 5.1.4
MiniWebRadio     Version 3.3h Aug 07/2024  
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 4194304 bytes, speed 40 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 80 MHz
PSRAM total size: 4194304 bytes
RESET_REASON: Reset due to power-on event

00:00:00    
00:00:00        *************************** 
00:00:00        *     MiniWebRadio V3     * 
00:00:00        ***************************
00:00:00 
00:00:00 setup: ....  Arduino is pinned to core
0 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 14052.4 MB by 15193.5 MB free
00:00:00 setup: ....  seek for stations.csv
00:00:00 setup: ....  stations.csv found
00:00:00 setup: ....  seek for WiFi networks
[  3706][E][WiFiMulti.cpp:306] run(): [WIFI] no matching wifi found!
00:00:00 WiFI_info:   WiFi credentials are not correct
00:00:00 RECONNECTION try 1
[  6113][W][STA.cpp:537] disconnect(): STA already disconnected.
[  7152][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 39 - TIMEOUT
[  7159][W][STA.cpp:537] disconnect(): STA already disconnected.
[  8240][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 2 - AUTH_EXPIRE
[  8247][W][STA.cpp:537] disconnect(): STA already disconnected.
[ 10445][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
[ 10453][W][STA.cpp:537] disconnect(): STA already disconnected.
[ 11139][E][WiFiMulti.cpp:293] run(): [WIFI] Connecting Failed AP not found.
00:00:00 WiFI_info:   WiFi credentials are not correct
00:00:00 RECONNECTION try 2
[ 11156][W][STA.cpp:537] disconnect(): STA already disconnected.
00:00:00 WiFI_info:   WiFi credentials are not correct
[... 330 lines cut ...]
[176610][E][NetworkClient.cpp:423] write(): fail on fd 55, errno: 113, "Software caused connection abort"
[177833][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 39 - TIMEOUT
[177840][W][STA.cpp:537] disconnect(): STA already disconnected.
[180240][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
[180247][W][STA.cpp:537] disconnect(): STA already disconnected.
[182446][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
[182453][W][STA.cpp:537] disconnect(): STA already disconnected.
[184653][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
[184660][W][STA.cpp:537] disconnect(): STA already disconnected.
[186859][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
[186867][W][STA.cpp:537] disconnect(): STA already disconnected.
[189265][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
[189273][W][STA.cpp:537] disconnect(): STA already disconnected.
15:56:18 WiFi         :  Reconnecting to WiFi...
[190174][W][STA.cpp:537] disconnect(): STA already disconnected.
E (201308) wifi:sta is connecting, return error
[191472][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
[191479][W][STA.cpp:537] disconnect(): STA already disconnected.
[193678][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
.[0;33m[193686][W][STA.cpp:537] disconnect(): STA already disconnected.
[194832][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 39 - TIMEOUT
15:56:32 WiFi         :  Reconnecting to WiFi...
[204173][W][STA.cpp:537] disconnect(): STA already
``` 
[image]

schreibfaul1, 2024-08-24:
BT and WiFi use the same frequency range. Where BT looks for a free channel before it transmits (frequency hopping). Nevertheless, influences can occur. Try to rotate the BT antenna 90° compared to the WiFi antenna, horizontally and vertically. And the distance between the two should be as large as possible. If it is not possible to shield the antennas from each other, try to wrap the BT module in an antistatic foil,
[image]

You can simply put such a bag over the BT module for testing. The BT range will still be sufficient after that.

rjshiffer, 2024-08-24:
Yes, that is what is happening.  I placed the BT Emitter in a antistatic bag, and re-orientated it as you suggested and was able to get more consistent connections and the streams play but with "slow stream, dropouts are possible."

Realizing that the work room I was in had other BT/WiFi interference possibilities (fluorescent lights, wireless controllers, another MiniWebRadio running in background, etc.) I moved the radio to my normal living space and it works fine with only minimal, barely noticeable dropouts over a period of an hour.

That's for your help.

closed 2024-08-24 by rjshiffer