schreibfaul1/ESP32-MiniWebRadio #496: some radio icons are not complet

maba64, 2024-08-27:
Thank you very much for the new version and your great work !!

Can you still show the current version on the info page ?

some radio icons are not complet: (on the display and also on the PC
[image]

[image]

station are from the sd imported

BR
Markus

schreibfaul1, 2024-08-27:
That is strange. Some icons should look different now. Perhaps the unpacking of the zip file onto the SD card did not work, or there is still old data in the browser cache. 
[image]

ve2cfb, 2024-09-01:
Same problem here. 
**On Web Page and Miniweb Radio.**
[image]

ve2cfb, 2024-09-02:
In my case, the downloaded image was a PNG 300x300 pixels. If i dont mistake and its an assumption on my part, Mini Web Radio probably change the size, so there might be a problem there. Wolle can confirm it.
changed the PNG to JPG and size to 128x128 pixels. 
Now its ok.

closed 2024-10-17 by github-actions[bot]