schreibfaul1/ESP32-MiniWebRadio #497: Connection issues

FranZatGIT, 2024-08-27:
Hi Wolle,

Today I loaded the latest version on my testboard and I'm having massive wifi connection issues with the new code. I downloaded a full fresh clone from Git (including the SD content) and the radio is pretty much not working at all. Ping only works occasionally. If I go back to my custom code version (based on a version from May), it works without problems. To rule out the HW, I also tried a new ESP - with the same result. Do you have any ideas? 

I would like to use your latest changes and add my code lines. But that makes no sense if it is not working.

Many thanks Franz

schreibfaul1, 2024-08-27:
Hi Franz,
before I published the version, I did exactly the same. I rewrote the SD card and cloned the Git repository into a new folder.
I rule out HW problems. What does the log say?
Please do a "Full Clean" before uploading. There may be something strange in settings.json or stations.json. You can delete these two files, they will be recreated.

FranZatGIT, 2024-08-27:
deleted the 2 files - no change. 

Here a screenshot of the output and below the full  test of another try:

[image]

ESP32 Chip: ESP32-S3
Arduino Version: 3.0.4
ESP-IDF Version: 5.1.4
MiniWebRadio     Version 3.4 Aug 23/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 16777216 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 40 MHz
PSRAM total size: 8388608 bytes
RESET_REASON: Reset due to power-on event

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V3     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9488
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 2.0 MB by 29.2 MB free
[   843][W][common.h:730] updateStationsList(): Stations: 26, favorites: 15
00:00:00 setup: ....  seek for WiFi networks
[  5221][W][STA.cpp:537] disconnect(): STA already disconnected.
00:00:00 WiFI_info:   Connecting WiFi...
00:00:00 WiFI_info:   mDNS name: MiniWebRadio
00:00:00 setup: ....  connected to skynet, IP address is 192.168.1.235, Received Signal Strength -39 dB
00:00:00 ftpServer:   Buffers allocated: 4622 bytes
00:00:00 rtime_info:  Initializing SNTP
00:00:00 audiotask:   is pinned to core 0
00:00:00 audiotask:   priority is 2
00:00:00 AUDIO_info:  commFMT = Philips
00:00:00 setup: ....  number of saved stations: 26
00:00:00 setup: ....  number of saved favourites: 15
00:00:00 setup: ....  current station number: 1
00:00:00 setup: ....  current volume: 21
00:00:00 setup: ....  volume steps: 21
00:00:00 setup: ....  volume after alarm: 12
00:00:00 setup: ....  last connected host: http://0n-70s.radionetz.de:8000/0n-70s.mp3
00:00:00 setup: ....  connection timeout: 2500 ms
00:00:00 setup: ....  connection timeout SSL: 3500 ms
00:00:00 action: ...  switch to station 1
00:00:00 AUDIO_info:  PSRAM found, inputBufferSize: 638965 bytes
00:00:00 AUDIO_info:  buffers freed, free Heap: 186744 bytes
00:00:00 AUDIO_info:  connect to: "0n-70s.radionetz.de" on port 8000 path "/0n-70s.mp3"
23:43:56 AUDIO_info:  Request http://0n-70s.radionetz.de:8000/0n-70s.mp3 failed!
[ 10431][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 8 - ASSOC_LEAVE
23:43:56 AUDIO_info:  Request http://0n-70s.radionetz.de:8000/0n-70s.mp3 failed!
23:43:56 StationName:
23:43:56 StreamTitle:
23:43:56 BT-Emitter:  KCX_BT_Emitter not found 
23:43:56 ftpServer:   Ftp server waiting for connection on port 21
23:44:00 AUDIO_info:  buffers freed, free Heap: 184180 bytes
23:44:00 AUDIO_info:  connect to: "0n-70s.radionetz.de" on port 8000 path "/0n-70s.mp3"
23:44:03 AUDIO_info:  Request http://0n-70s.radionetz.de:8000/0n-70s.mp3 failed!
[ 17575][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 8 - ASSOC_LEAVE
23:44:03 AUDIO_info:  Request http://0n-70s.radionetz.de:8000/0n-70s.mp3 failed!
23:44:03 StationName:
23:44:03 StreamTitle:
[ 18655][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 2 - AUTH_EXPIRE
[ 18663][W][STA.cpp:537] disconnect(): STA already disconnected.
23:44:17 AUDIO_info:  buffers freed, free Heap: 184012 bytes
23:44:17 AUDIO_info:  connect to: "0n-70s.radionetz.de" on port 8000 path "/0n-70s.mp3"
23:44:20 AUDIO_info:  Request http://0n-70s.radionetz.de:8000/0n-70s.mp3 failed!
[... 78 lines cut ...]
E (125090) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107
E (125097) diskio_sdmmc: sdmmc_read_blocks failed (263)
23:45:51 AUDIO_info:  file "/common/m/RSSI4.bmp" not found
[125668][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 2 - AUTH_EXPIRE
[125675][W][STA.cpp:537] disconnect(): STA already disconnected.
[126914][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 39 - TIMEOUT
23:46:03 WiFi         :  Reconnecting to WiFi...
[137552][W][STA.cpp:537] disconnect(): STA already disconnected.
[140382][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 205 - CONNECTION_FAIL
[140390][W][STA.cpp:537] disconnect(): STA already disconnected.
[142246][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 39 - TIMEOUT
23:46:17 WiFi         :  Reconnecting to WiFi...
[151559][W][STA.cpp:537] disconnect(): STA already disconnected.
[154389][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 205 - CONNECTION_FAIL
[154397][W][STA.cpp:537] disconnect(): STA already disconnected.
[155468][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 2 - AUTH_EXPIRE
[155475][W][STA.cpp:537] disconnect(): STA already disconnected.
[157068][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 39 - TIMEOUT
23:46:31 WiFi         :  Reconnecting to WiFi...
[165558][W][STA.cpp:537] disconnect(): STA already disconnected.
[168588][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 205 - CONNECTION_FAIL
[168596][W][STA.cpp:537] disconnect(): STA already disconnected.
[169702][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 2 - AUTH_EXPIRE
[169709][W][STA.cpp:537] disconnect(): STA already disconnected.
23:46:39 AUDIO_info:  buffers freed, free Heap: 185284 bytes

FranZatGIT, 2024-08-27:
compared the Common.h with the old one  and set all the BT Emitter pins to -1  ( BT Emittter is not used in my case ) and changed the SDMMC Frequency to 40MHz. It seems to work now - I will keep it open until tomorrow and watch if it is stable.
I think I found another thing while testing with the AP Mode. I will open another issue for that.

FranZatGIT, 2024-08-30:
Hi Wolle still have some issues with the connection - Some times it work some times not. I can not really reproduce it. Fact is that the old 3.1 release always works. Was there any major change with the Wifi part  in 3.4?

Shark2805, 2024-09-12:
hi friend. I really liked the project and I want to bring it to life. The board is stitched, but there is no Wi-Fi connection and does not create an access point. I tried all the options described in the problems, with no result. Help please.
ESP32 Chip: ESP32-S3
Arduino Version: 3.0.4
ESP-IDF Version: 5.1.4
MiniWebRadio     Version 3.4bd - Sep 09/202
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 16777216 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 40 MHz
PSRAM total size: 8388608 bytes
RESET_REASON: Reset due to power-on event

00:00:00    [0m
00:00:00 [33m       ***************************    [0m
00:00:00 [33m       *     MiniWebRadio V3     *    [0m
00:00:00 [33m       ***************************    [0m
00:00:00    [0m
00:00:00 setup: ....  Arduino is pinned to core [36m1[0m
[0;37m[  1220][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type SPI_MASTER_SCK (34) successfully set to 0x42129f18[0m
[0;37m[  1242][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type SPI_MASTER_MISO (35) successfully set to 0x42129e30[0m
[0;37m[  1255][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type SPI_MASTER_MOSI (36) successfully set to 0x42129d48[0m
[0;37m[  1268][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type SPI_MASTER_SS (37) successfully set to 0x42129c24[0m
[0;37m[  1281][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x42140560[0m
[0;37m[  1293][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 17 successfully set to type GPIO (1) with bus 0x12[0m
[0;37m[  1304][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 17 successfully set to type SPI_MASTER_SCK (34) with bus 0x1[0m
[0;37m[  1316][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x42140560[0m
[0;37m[  1328][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 2 successfully set to type GPIO (1) with bus 0x3[0m
[0;37m[  1339][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 2 successfully set to type SPI_MASTER_MISO (35) with bus 0x1[0m
[0;37m[  1350][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x42140560[0m
[0;37m[  1362][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 18 successfully set to type GPIO (1) with bus 0x13[0m
[0;37m[  1373][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 18 successfully set to type SPI_MASTER_MOSI (36) with bus 0x1[0m
[0;37m[  1385][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x42140560[0m
[0;37m[  1397][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 12 successfully set to type GPIO (1) with bus 0xd[0m
[0;37m[  1408][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x42140560[0m
[0;37m[  1420][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 8 successfully set to type GPIO (1) with bus 0x9[0m
00:00:00 tft_info: .  init [36mILI9341[0m
00:00:00 setup: ....  Init SD card[0m
[0;37m[  1551][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x42140560[0m
[0;37m[  1567][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 4 successfully set to type GPIO (1) with bus 0x5[0m
[0;37m[  1577][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x42140560[0m
[0;37m[  1589][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 11 successfully set to type GPIO (1) with bus 0xc[0m
[0;37m[  1600][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type SDMMC_CLK (38) successfully set to 0x420181dc[0m
[0;37m[  1613][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type SDMMC_CMD (39) successfully set to 0x420181dc[0m
[0;37m[  1625][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type SDMMC_D0 (40) successfully set to 0x420181dc[0m
[0;37m[  1638][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 11 successfully set to type INIT (0) with bus 0x0[0m
[0;37m[  1695][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 14 successfully set to type SDMMC_CMD (39) with bus 0x3fca1498[0m
[0;37m[  1707][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 13 successfully set to type SDMMC_CLK (38) with bus 0x3fca1498[0m
[0;37m[  1719][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 11 successfully set to type SDMMC_D0 (40) with bus 0x3fca1498[0m
00:00:00 [37msetup: ....  SD card found, 30413.8 MB by 30440.0 MB free[0m
[0;37m[  1774][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type LEDC (9) successfully set to 0x4207e868[0m
[0;37m[  1787][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 10 successfully set to type LEDC (9) with bus 0x3fced698[0m
[0;32m[  1798][I][esp32-hal-ledc.c:151] ledcAttachChannel(): LEDC attached to pin 10 (channel 0, resolution 8)[0m
00:00:00 setup: ....  seek for WiFi networks[0m
[0;32m[  2172][I][WiFiMulti.cpp:86] addAP(): [WIFI][APlistAdd] add SSID: ROSTELECOM_E328[0m
[0;32m[  2191][I][WiFiMulti.cpp:86] addAP(): [WIFI][APlistAdd] add SSID: ROSTELECOM_E328[0m
[0;32m[  2199][I][WiFiMulti.cpp:86] addAP(): [WIFI][APlistAdd] add SSID: ROSTELECOM_OAD7[0m
[0;37m[  2225][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 9 - WIFI_READY[0m
[... cut ...]
[0;37m[  1161][V][esp32-hal-uart.c:497] uartBegin(): UART0 not installed. Starting installation[0m
[0;37m[  1170][V][esp32-hal-uart.c:560] uartBegin(): UART0 initialization done.[0m

ESP32 Chip: ESP32-S3
Arduino Version: 3.0.4
ESP-IDF Version: 5.1.4
MiniWebRadio     Version 3.4bd - Sep 09/202
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 16777216 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 40 MHz
PSRAM total size: 8388608 bytes
RESET_REASON: Reset due to power-on event

closed 2024-10-27 by github-actions[bot]