schreibfaul1/ESP32-MiniWebRadio #498: AP Mode is not stable working after code change

FranZatGIT, 2024-08-27:
Hi Wolle,

As suggested (https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/484) the Radio is no longer stuck in the AP Mode screen when WIFI is not available. However I had troubles  to connect to the AP when the the stored credentials are wrong. Following the log the Radio enables the AP Mode but tries to connect to a Radio station and switches back and forth in the STA Mode:

[image]

[image]

sometimes AP Mode works but most time not. As mentioned in https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/491 The Radio should start into the player mode when internet is not available and should show the 192.168.4.1 as IP address.
of course a notification would be nice as well that AP Mode is active and a reboot is necessary for a Internet connect retry....

FranZatGIT, 2024-08-30:
Hi Wolle, any comment to that issue? Thanks Franz

schreibfaul1, 2024-08-31:
Hi Franz,
now I know what the problem is. I usually replace the networks.csv with my own, and then it always works. An incorrect PW was probably saved. Please close the serial monitor and click on **Erase Flash**. You can then reset the PW via the access point. I will fix this in one of the next updates.
Best regards
Wolle

FranZatGIT, 2024-09-02:
Hi Wolle,

Without Internet connection/WIFI the AP (WIFI: MiniWebRadio) is opened and the connection works stable. However the AP WIFI WEB page has some strange behavior. The drop-down list for selecting the WIFI changes from empty to occasionally filled.

Most time it looks like:

[image]

and sometimes the available WIFI appears:

[image]

This makes it a bit annoying if you change networks frequently. I also did a Flash Erase , removed WIFI credentials from Networks.csv - there was no difference

The logfile (screenshot in initial post) shows that the radio is still trying to connect to the WIFI. I don't know if a open AP and simultane STA Connection can cause troubles... 

Additional I saw during testing, that the connection to the the (disabled) WIFI is established as soon as the WIFI is enabled. That is ok but the AP is still active.

closed 2024-10-17 by github-actions[bot]