schreibfaul1/ESP32-MiniWebRadio #499: Stations.csv to Stations.json conversion

ve2cfb, 2024-08-28:
Hello Wolle,

thanks for the many changes with the new 4.3V. Do you ever get to sleep..?

Is there a way in ESP32-MiniWebRadio to convert or import from my Stations.csv to SDCARD Stations.json without doing it manually...?
Tried the web interface doing manually one by one (It takes a lot of time). 
Using a external utility ( **_Bulk Rename Utility_**)  is feasable but not  perfect.

Thanks again for your work.
Pierre

schreibfaul1, 2024-08-28:
Hi Pierre, the old method was not so good, especially as saving and loading was "strange".

Unfortunately I don't have a conversion tool. The json format is:
[["Fav","Cy","Name",URL"],[["Fav","Cy","Name",URL"],.....,["Fav","Cy","Name",URL"]]
You can use a word processor to replace all tabs with "," and all \n with "],[".
Square brackets on the very outside, then you're almost done.

ve2cfb, 2024-08-28:
Hi Wolle,
Its already done by doing something similar to your idea, but i like your idea.

By the way, is the random function for the MP3 files on your list to do...?

Thank again.

steve6375, 2024-08-28:
I don't think we want a random function because then it could play the same tune twice or frequently.
What is better is to have a shuffle function

closed 2024-10-13 by github-actions[bot]