schreibfaul1/ESP32-MiniWebRadio #500: No error, just a question to modify the setup

pisascha, 2024-08-30:
hello wolle,

thank you for this nice project. 

i'm new to this so sorry for the question but may you know it:

i want to built the webradio with built-in speakers. is it possible to do it with a PAM8406 amplifier? i found a sketch in the internet (but for the raspberry pi) where there is a PAM8406 wired to the PCM5102 and speakers to the PAM8406. is it possible do the same with your webradio?

[image]

if this is possible is it neccesary to modify your code? 

sorry for the trouble and for help in advance

schreibfaul1, 2024-08-30:
I have PAM8403 in use and it works. 
For the power supply you need at least 3A (2x 5W + 1A for the ESP32), I would dimension it a little larger, perhaps this:
[image]
Then you should make sure that the analogue part is positioned as far away as possible from the WiFi antenna of the ESP. Every cm of wire acts as an antenna and this can cause interference noise.

pisascha, 2024-08-31:
Thank you for your quick reply and the suggestion.

pisascha, 2024-09-06:
Hey Wolle, my parts arrived yesterday and now i want to start. Do I have to disactivate the hardware I don't use (e.g. ir) in common.h or is it done automatically?

schreibfaul1, 2024-09-07:
If you do not want to connect an IR receiver, it is better to set the pin to -1.
If you set a pin, it should not remain unconnected for an input.

pisascha, 2024-09-07:
Ok, thank you for your reply. That is what I thought. 

Thank you for your trouble, i appreciate your work

pisascha, 2024-09-08:
I checked my boxes of raspi -stuff and found a 1838-based ir receiver modul. Hope this will work, have a test in the next few days

pisascha, 2024-09-09:
Good morning wolle,

I want use my ESP32-s3 n16r8 with this microsd-modul

[image]

Due to your manual i didn't touch the resistor(s) on the modul cause i use the s3 model.

Now i get this error message 

[image]

What mistake am I making? Thank you for help!

pisascha, 2024-09-09:
Thank you very much for your help

schreibfaul1, 2024-09-09:
I can only advise against the Catalex adapter. The modification is difficult and there is no guarantee that it will work well. This board is intended for the old Arduino chips with 5V output. If you have an adapter for your SD card, simply solder 5 wires to it without any other components.
[image]

closed 2024-10-25 by github-actions[bot]