schreibfaul1/ESP32-MiniWebRadio #502: IR Remote Issue

TubaFonse, 2024-09-02:
Hallo Wolle, 
wenn ich mit der IR-Fernbedienung die Sender mit den Nummerntasten (0 ... 9) direkt umschalte, wechselt das Sender-Logo und der Sender-Name im Display nicht mit (auch nicht im Web-Interface). Wenn ich mit der Fernbedienung mit "Next Station" /  "Previous Station" umschalte, wird das aktuelle Sender-Logo und Sender-Name jedoch korrekt angezeigt. 
Kannst Du den Fehler reproduzieren?

Danke & Grüße, 
Alfons.

TedWillemsen, 2024-09-02:
Dear Wolle,
Same problem. IR number command (also in 2 or 3 digits) is correctly displayed, program logo, program name and small program number below left do not change, but the program content (Music, talking) changes to the selected program number, so the right URL is selected. Also additional program info belongs to the selected number. I suppose this will also work in TubaFonse issue. Remark: most of "frozen" data is stored on SD, except small program number left below.
Next and previous station via IR or touch works correct.  Problem witn 2 same projects ESP32S3 N16R8, PCM5102A, ILI9341 2.8", previous version 3.2h was OK

Kind regards,
Ted Willemsen

TedWillemsen, 2024-09-02:
Dear Wolle,

To be more complete, last updated version is 3.4a. Coming from 3.4 with the same behaviour.

Kind regards,
Ted

schreibfaul1, 2024-09-02:
Yes, there seems to be a mistake in it. I'll take a look at this.

schreibfaul1, 2024-09-02:
Hello Alfons, hello Ted, 
there have been some changes to the channel list recently. I did not take the IR remote control into account. But now the correct URL is called up.

Even if all changes to the channel list are automatically written to the SD card, it is better to save a backup on the PC for large lists. There is currently no "Undo" function. But that is being planned.

TedWillemsen, 2024-09-02:
Dear Wolle,

Great, src/main.cpp replaced by 3.4b version, problem is solved now.

Thank you very much for the quick reaction and solution
Ted

TubaFonse, 2024-09-03:
Herzlichen Dank Wolle für die schnelle und professionelle Lösung! 
Viele Grüße, 
Alfons

TubaFonse, 2024-09-03:
solved!

closed 2024-09-03 by TubaFonse