schreibfaul1/ESP32-MiniWebRadio #505: problem with time display in sleep mode

valera39, 2024-09-06:
Hello Dear Wolle!
Suddenly the time in sleep mode started to display like this.
[image]
It started when the clock froze at the same time in sleep mode. I turned on the radio, it played, the time in the upper right corner displayed correctly. I turned the radio off again, and this is the display.
This is not the first time this has happened. And what is interesting is that the alarm goes off normally, so the time is synchronized correctly, in working mode on the display everything is normal without any artifacts. This problem with the time display only occurs in sleep mode. I replaced the SD card, so far everything is fine, but I'm not sure if the problem will go away.
Do you have any insights on this issue. Why does this only happen in sleep mode? 
Can't read the files in the "digits" folder from the SD card?
I have changed the frequency of the SD card, I have set it to 80MHz and 40MHz
What else can be done about it? My head is already broken :)

valera39, 2024-09-06:
I'll add what I wonder: why is only the last digit of the minutes and the blinking colon of the hour and minute separator displayed? And this digit changes every minute and the separator blinks, so the time is set correctly, I understand. I don't know how commands are issued to output the time in sleep mode, as it is written in the code, and I probably won't understand.
I really hope for your understanding

valera39, 2024-09-07:
Got the old SD card back today, didn't do anything to it. The time is still displayed normally. 
And this problem is stable a couple times a month. It's a miracle.

schreibfaul1, 2024-09-09:
Strange, I couldn't reproduce it, but I know what the problem could be. In CLOCK mode the display is not constantly updated, but only the change is drawn, every second the colon, at the minute change the ones digit etc. Probably a variable is not set correctly. With the update I make sure that all segments are always drawn at the beginning.

valera39, 2024-09-09:
Dear Wolle, thank you for your attention to this annoying problem. It is really hard to catch and may not show up for weeks. I hope your changes will help to eliminate it. I will keep an eye on it. So you don't think the SD card has anything to do with it?

schreibfaul1, 2024-09-09:
Hi valera39,
this has nothing to do with the SD card. I suspect that the problem has been resolved, otherwise I will investigate further.

The audio library can now jump for all audio formats. If you enter an integer in the VS Code terminal and are in player mode, the system jumps by an offset. Example: -10 [Enter] jumps back 10 seconds. You cannot see the input in the terminal, but simply click on the terminal and enter -10 [Enter].

closed 2024-10-24 by github-actions[bot]