schreibfaul1/ESP32-MiniWebRadio #507: yet another ILI9341 touch varient

mdwsmith, 2024-09-11:
Hi thanks for the ESP-MiniWebRadio code (very impressive for a microcontroller), and the I2S audio libraries that are used by many projects.

I have a ESP32 audiokit with the AC101 DAC. 

I've connected it to a 2.8inch ILI9341 screen with touch from here;

https://www.aliexpress.com/item/4000219159401.html?spm=a2g0o.order_list.order_list_main.402.21ef1802yw2Xup

TFT MISO is not connected.

As per the issues noted on this tracker, I have tried various touch drivers, best luck with the RPIILI9341, I can alter the volume on the DLNA screen so its working somewhat, but to further diagnose need the screen coordinates of the touches.

Uncommenting the lines in main.cpp indicated by steve6375 here;

https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/262

eg this code fragment;

// Event from TouchPad
// clang-format off
void tp_pressed(uint16_t x, uint16_t y){
      SerialPrintfln("tp_pressed, state is: %i", _state);
      SerialPrintfln(ANSI_ESC_YELLOW "Touchpoint  x=%d, y=%d", x, y);

    if(_f_sleeping) return;  // awake in tp_released()

does not work - they do not display in the terminal - has there been changes that affect this?

I display the terminal by clicking the plug icon on the bottom bar of the PALMIO.

Any help or pointers appreciated - I'm a bit new to this!
Thanks

pisascha, 2024-09-13:
hey @mdwsmith 

it is still working on my machine. did you delete the slashes in rows 3011 and 3012 in main.ccp? i use platformio like @schreibfaul1 recommands. 

i had problems with the coordinates on my ili9341 no-name-aliexpress-extra-cheap-display too but setting the tp rotation in common.h from 0 to 3 like [steve6375](https://github.com/steve6375) recommanded worked out just fine. my setting:

[image]

mdwsmith, 2024-09-16:
@pisascha 
Thanks very much for your answer.

wrt "did you delete the slashes in rows 3011 and 3012 in main.ccp?"
Yes I did that as in the code in my first post, although in main.cpp at this time its lines 3004 and 3005.

wrt "i use platformio like @schreibfaul1 recommands."
Yes i do to - I have suceesfully compiled and uploaded the code as per the setup instructions.

wrt"setting the tp rotation in common.h from 0 to 3 like [steve6375](https://github.com/steve6375) recommanded worked out just fine"

I've tried several combinations of rotations and reflections and drivers - this is why I wanted to get the points of the screen co-ordinates to work out what is going wrong.

As Ive said in my first post, in the ILI9341 DLNA screen I seem to get some responce to screen touches - perhaps responces to touch presses in this screen are not going to the lines we are discussing in main.cpp, hence I'm getting no output in the terminal.  What ILI9341 screen are you in when you generate the touch presses?

Basically, I'm looking to work out which way up/around the screen touch pad is, and looking to calibrate it if needed.

Thanks very much for your help.

Best regards mdwsmith

closed 2024-10-31 by github-actions[bot]