schreibfaul1/ESP32-MiniWebRadio #508: OPUS decode error

valera39, 2024-09-13:
Hello Dear Wolle!
After a short playback time, the stream stops
ESP32 Chip: ESP32-S3
Arduino Version: 3.0.4
ESP-IDF Version: 5.1.4
MiniWebRadio     Version 3.4bc - Sep 03/202
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 16777216 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 40 MHz
TFT speed 40 MHz
PSRAM total size: 8388608 bytes
RESET_REASON: Reset due to power-on event

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V3     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9488
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 29790.5 MB by 29818.0 MB free
00:00:00 setup: ....  seek for WiFi networks
[  4977][W][STA.cpp:537] disconnect(): STA already disconnected.
00:00:00 WiFI_info:   Connecting WiFi...
00:00:00 WiFI_info:   mDNS name: MiniWebRadio
00:00:00 setup: ....  connected to Honor, IP address is 192.168.43.214, Received Signal Strength -28 dB
00:00:00 ftpServer:   Buffers allocated: 4622 bytes
00:00:00 rtime_info:  Initializing SNTP
00:00:00 audiotask:   is pinned to core 0
00:00:00 audiotask:   priority is 2
00:00:00 AUDIO_info:  commFMT = Philips
00:00:00 setup: ....  number of saved stations: 93
00:00:00 setup: ....  number of saved favourites: 93
00:00:00 setup: ....  current station number: 64
00:00:00 setup: ....  current volume: 5
00:00:00 setup: ....  volume steps: 21
00:00:00 setup: ....  volume after alarm: 0
00:00:00 setup: ....  last connected host: http://stream.polskastacja.pl/ps27_mp3
00:00:00 setup: ....  connection timeout: 2500 ms
00:00:00 setup: ....  connection timeout SSL: 3500 ms
00:00:00 action: ...  switch to station 64
00:00:00 AUDIO_info:  PSRAM found, inputBufferSize: 638965 bytes
00:00:00 AUDIO_info:  buffers freed, free Heap: 186348 bytes
00:00:00 AUDIO_info:  connect to: "stream.polskastacja.pl" on port 80 path "/ps27_mp3"
00:00:00 AUDIO_info:  Connection has been established in 157 ms, free Heap: 185932 
bytes
11:21:48 bitRate:     128Kbit/s
11:21:48 bitRate:     128Kbit/s
11:21:48 icy-descr:   Grupa Wirtualna Polska
11:21:48 StationName: 27_mp3
11:21:48 icy-url: ..  http://grupawp.pl
11:21:48 AUDIO_info:  MP3Decoder has been initialized, free Heap: 183564 bytes , free stack 4952 DWORDs
11:21:48 lastURL: ..  http://stream.polskastacja.pl/ps27_mp3
11:21:48 AUDIO_info:  stream ready
11:21:48 AUDIO_info:  syncword found at pos 0
11:21:48 AUDIO_info:  Channels: 2
11:21:48 AUDIO_info:  SampleRate: 44100
11:21:48 AUDIO_info:  BitsPerSample: 16
11:21:48 AUDIO_info:  BitRate: 128000
11:21:48 setup: ....  Ambient Light Sensor BH1750 found
11:21:49 Audiocodec:  MP3
11:21:49 ftpServer:   Ftp server waiting for connection on port 21
11:21:49 StreamTitle: Albert One - Secrets w Italo Disco
11:21:49 webSrv: ...  Download  "/stations.json"
11:21:55 DLNA_server: 0 media server found
11:22:01 Webpage:     index.html
11:22:02 webSrv: ...  GetFolder "/"
11:22:03 webSrv: ...  Download  "/stations.json"
11:22:07 Webpage:     index.html
11:22:08 webSrv: ...  GetFolder "/"
11:22:09 webSrv: ...  Download  "/stations.json"
11:22:16 AUDIO_info:  buffers freed, free Heap: 183376 bytes
11:22:16 AUDIO_info:  connect to: "radio.glafir.ru" on port 80 path "/easy-listen/paul-mauriat"
11:22:17 AUDIO_info:  Connection has been established in 380 ms, free Heap: 183116 
bytes
11:22:17 StationName: http://radio.glafir.ru/easy-listen/paul-mauriat
11:22:17 AUDIO_info:  chunked data transfer
11:22:17 bitRate:     64Kbit/s
11:22:17 icy-descr:   PAUL MAURIAT music.
11:22:17 StationName: PAUL MAURIAT on radio Glafir
11:22:17 icy-url: ..  http://radio.glafir.ru/easy-listen/paul-mauriat
11:22:17 lastURL: ..  http://radio.glafir.ru/easy-listen/paul-mauriat
11:22:17 webSrv: ...  File not found "/logo/.jpg"
11:22:17 AUDIO_info:  stream ready
11:22:17 AUDIO_info:  OPUSDecoder has been initialized, free Heap: 179916 bytes , free stack 4952 DWORDs
11:22:17 AUDIO_info:  syncword found at pos 0
11:22:17 AUDIO_info:  Paul Mauriat - Mull Of Kintyre
11:22:17 StreamTitle: Paul Mauriat - Mull Of Kintyre
11:22:17 AUDIO_info:  Channels: 2
11:22:17 AUDIO_info:  SampleRate: 48000
11:22:17 AUDIO_info:  BitsPerSample: 16
11:22:17 AUDIO_info:  BitRate: 160456
11:22:18 Audiocodec:  OPUS
11:23:33 AUDIO_info:  OPUS decode error -5 : SILK MODE UNSUPPORTED

schreibfaul1, 2024-09-13:
The OPUS decoder consists of two components, CELT and SILK. CELT is mainly used for music, SILK mainly for speech and low bandwidths. Many radio stations only use the CELT mode, these are played back properly. However, there are also stations that switch to Hybrid or Silk mode during pauses in music or speech. I don't have a good SILK decoder at the moment, in which case a message appears and the stream is cancelled. 
It will still take a long time to get the SILK decoder up and running, which will certainly not be until next year.

valera39, 2024-09-14:
Thank you, what I like about your answers is that you always answer any questions, even the simplest and most obvious ones, in detail and without arrogance. It's true that you are not always quick, but I suppose that's because of the lack of time. I would like to remind you about my question  #506 , What do you think?
Thanks again!

schreibfaul1, 2024-09-14:
Yes, OPUS has a wide range.
[image]
Celt is green and that has been realised.
Orange would be the SILK decoder, which is expected to arrive in a few months.
Yellow would be the hybrid mode, i.e. a mixture of CELT and SILK.

I will look at the problem with the time zones soon. The csv will probably be replaced with the json format. This has already been done for the "settings" and "stations". The json is easier to edit with JavaScript.

valera39, 2024-09-14:
Thank you! Is the station that doesn't work https://stream.zeno.fm/zuh2nqp5qzzuv already working from the audio library updated yesterday or not?

schreibfaul1, 2024-09-14:
The rejection (HTTP401) is triggered by a missing UserAgent. I have never had this before. Now a UserAgent is transmitted with the http request.

valera39, 2024-09-14:
Now the stream is playing normally
Thank you very much!

closed 2024-10-30 by github-actions[bot]