schreibfaul1/ESP32-MiniWebRadio #509: Balance slider reversed

TedWillemsen, 2024-09-16:
Dear Wolle,

In my opinion the balance slider in the equalizer works in the wrong direction. Slider to the right, only left channel, slider to the left, only right channel. Tested with a relaible L / R indicated headphone direct on PCM5102A phone jack output. However with the balance in mid position L/R channel routing is correct. Also tested with "generated" sound files via "Audacity". This issue is already seen from V2 onwards.

Kind regards,
Ted Willemsen

schreibfaul1, 2024-09-16:
Hi Ted,
you're right, I swapped the right/left balance in the audio library. Please reload the library with "Full Clean".

hevet, 2024-09-23:
@TedWillemsen I have the same problem, these crackling noises are very unpleasant.

TedWillemsen, 2024-09-23:
@Hevet Jacek
The problem occurs at the moment a selected station or file starts playing and does not appear anymore staying on the station or during the file playing.
Did you notice the problem earlier on previous LIB (audio ?) or SRC releases ?
In my case it happened after reloading the complete project status september 18. This on 2 identical projects
ESP32S3 N16R8, PCM5102A, TFT 2.8" ILI9341

Kind regards
Ted

hevet, 2024-09-24:
This wasn't a problem before. Unfortunately I don't know which library update this started happening. I'll have to look for that update.

TedWillemsen, 2024-10-02:
**Clicks after balance slider solution**

Busy with some other issues, disturbed IP adres on TFT and extended IR remote functions, (both solved). 
I come back to the issue of clicks after solving the balance slider issue.
In between, having 2 comparable project examples, I dit on one of them a complete new installation, new Visual Code install, erase of all belonging data via powershell, full flash erase on ESP.
After that some clicks on change between specific stations* are still there but less annoying, for me acceptable.
Swapping between SD audio files, clicks are completely gone.
Compared now on the onther set, station clicks are absolute more irritating and also present switching between SD audio files.
**So in my opinion if you do the right clean re-installation this issue can be closed.**

*Typical stations with still some clicks: 
0N 70s	0n-80s.radionetz.de:8000/0n-70s.mp3"	
0N 80s	0n-80s.radionetz.de:8000/0n-80s.mp3 "	
0N 90s	0n-90s.radionetz.de:8000/0n-90s.mp3"	**The most remarkable one !!!**
0N Charts 0n-charts.radionetz.de:8000/0n-charts.mp3"	
0N Dance 0n-dance.radionetz.de:8000/0n-dance.mp3"	
0N Disco 0n-disco.radionetz.de:8000/0n-disco.mp3"

All seems quite high level modulated (and compressed ?)!!
Above mentioned stations present in the example SD contents

Kind regards,
Ted

closed 2024-11-16 by github-actions[bot]