schreibfaul1/ESP32-MiniWebRadio #511: ESP32S3 OTA doesn't compile

HWH53, 2024-09-29:
Hallo Wolle,
ich habe heute auf die Version 3.5 upgedated. Diese Version lässt sich leider nicht für ESP32S3 OTA compilieren. Die Version ohne OTA compiliert fehlerlos. Bitte kontrolliere doch gelegentlich dieses Problem. 

`Warning! Arduino framework as an ESP-IDF component doesn't handle the `variant` field! The default `esp32` variant will be used.
Reading CMake configuration...
-- git rev-parse returned ''
-- git rev-parse returned ''
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
Processing 26 dependencies:

[1/26] chmorgan/esp-libhelix-mp3 (1.0.3)

[2/26] espressif/cbor (0.6.0~1)

[3/26] espressif/esp-dsp (1.4.12)

[4/26] espressif/esp-modbus (1.0.16)

[5/26] espressif/esp-serial-flasher (0.0.11)

[6/26] espressif/esp-sr (1.9.2)

[7/26] espressif/esp-zboss-lib (1.5.0)

[8/26] espressif/esp-zigbee-lib (1.5.0)

[9/26] espressif/esp_diag_data_store (1.0.1)

[10/26] espressif/esp_diagnostics (1.2.0)

[11/26] espressif/esp_insights (1.2.0)

[12/26] espressif/esp_modem (1.1.0)

[13/26] espressif/esp_rainmaker (1.5.0)

[14/26] espressif/esp_rcp_update (1.2.0)

[15/26] espressif/esp_schedule (1.2.0)

[16/26] espressif/esp_secure_cert_mgr (2.4.1)

[17/26] espressif/jsmn (1.1.0)

[18/26] espressif/json_generator (1.1.2)

[19/26] espressif/json_parser (1.0.3)

[20/26] espressif/libsodium (1.0.20~1)

[21/26] espressif/mdns (1.4.0)

[22/26] espressif/network_provisioning (1.0.2)

[23/26] espressif/qrcode (0.1.0~2)

[24/26] espressif/rmaker_common (1.4.6)

[25/26] joltwallet/littlefs (1.14.8)

[26/26] idf (5.1.4)

-- DEBUG: Use esp-modbus component folder: C:/Users/HWH/Documents/PlatformIO/Projects/Radio/ESP32-MiniWebRadio_V3_5/ESP32-MiniWebRadio/managed_components/espressif__esp-modbus.
-- Project sdkconfig file C:/Users/HWH/Documents/PlatformIO/Projects/Radio/ESP32-MiniWebRadio_V3_5/ESP32-MiniWebRadio/sdkconfig.esp32s3_OTA
-- Could NOT find Threads (missing: Threads_FOUND)
-- App "ESP32-MiniWebRadio" version: 1
-- Adding linker script C:/Users/HWH/Documents/PlatformIO/Projects/Radio/ESP32-MiniWebRadio_V3_5/ESP32-MiniWebRadio/.pio/build/esp32s3_OTA/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Users/HWH/.platformio/packages/framework-espidf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script C:/Users/HWH/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script C:/Users/HWH/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script C:/Users/HWH/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld       
-- Adding linker script C:/Users/HWH/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld       
-- Adding linker script C:/Users/HWH/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld      
-- Adding linker script C:/Users/HWH/.platformio/packages/framework-espidf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- DEBUG: Use esp-modbus component folder: C:/Users/HWH/Documents/PlatformIO/Projects/Radio/ESP32-MiniWebRadio_V3_5/ESP32-MiniWebRadio/managed_components/espressif__esp-modbus.
-- git rev-parse returned ''
-- git rev-parse returned ''
-- Configuring incomplete, errors occurred!
See also "C:/Users/HWH/Documents/PlatformIO/Projects/Radio/ESP32-MiniWebRadio_V3_5/ESP32-MiniWebRadio/.pio/build/esp32s3_OTA/CMakeFiles/CMakeOutput.log".
See also "C:/Users/HWH/Documents/PlatformIO/Projects/Radio/ESP32-MiniWebRadio_V3_5/ESP32-MiniWebRadio/.pio/build/esp32s3_OTA/CMakeFiles/CMakeError.log".

Compiler supported targets: xtensa-esp32s3-elf

fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
ESP Insights Project commit: HEAD-HASH-NOTFOUND
ESP RainMaker Project commit: HEAD-HASH-NOTFOUND
CMake Error at C:/Users/HWH/.platformio/packages/framework-arduinoespressif32/CMakeLists.txt:311 (message):
  esp32-arduino requires CONFIG_FREERTOS_HZ=1000 (currently 100)

==================================================== [FAILED] Took 13.11 seconds ====================================================
Environment    Status    Duration
-------------  --------  ------------
esp32s3_OTA    FAILED    00:00:13.105
=============================================== 1 failed, 0 succeeded in 00:00:13.105 ===============================================
 *  Der Terminalprozess "C:\Users\HWH\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload', '--environment', 'esp32s3_OTA', '--upload-port', 'COM7'" wurde mit folgendem Exitcode beendet: 1. 
 *  Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste. 
`
Danke für deine bisherige hervorragende Arbeit!!!

mfg
Helmut

maba64, 2024-09-29:
Hallo
habe das gleiche Problem
Gruss Markus

HWH53, 2024-09-30:
Danke für die superschnelle Korrektur!
ESP32OTA ist nun wieder wie erwartet.

mfg
Helmut

closed 2024-09-30 by HWH53