schreibfaul1/ESP32-MiniWebRadio #512: IP display

valera39, 2024-09-29:
Hello Dear Wolle!
This is how my IP is displayed on the latest version of the radio, ILI9488 display
[image]
And another problem, when trying to edit the remote's buttons in the configuration menu, there is no change on the page when the button on the remote is pressed

HWH53, 2024-09-30:
Same problem here!
Strange signs as ip-address.

mfg
Helmut

TedWillemsen, 2024-09-30:
Dear Wolle, Valera and Helmut,
Same problem after installing newest version. TFT ILI9341 2.8" But I think it is not a TFT problem because rest of display is OK
The position, content and length changes after each reset and power off-on, see fotographs.
The right IP is 192.168.178.227

Kind regards,
Ted
[image]
[image]

schreibfaul1, 2024-09-30:
the IP address is displayed correctly again.

TedWillemsen, 2024-09-30:
Hi Wolle
IP address issue solved, thanks a lot

Kind regards
Ted

HWH53, 2024-09-30:
Hello Wolle,
TFT_FREQUENCY should be set to 40000000 by default. With the 80000000 setting, my ILI9341 displays show garbled station logos and strange font artifacts with the small font for time and IP address.
mfg
Helmut

HWH53, 2024-10-01:
Hi Wolle,
problems with Ip-address und grafics are solved with this fix. 
Thanks for that!

Kind regards
Helmut

closed 2024-11-16 by github-actions[bot]