schreibfaul1/ESP32-MiniWebRadio #513: new remote control setup file

valera39, 2024-09-29:
Hello Dear Wolle!
I have the following problem: if I put the radio into sleep mode using the remote control button, I can only get out of sleep mode by pressing the timer button. no other buttons take the radio out of sleep mode. I have tried using these types of remotes
[image]
In version 3.4 the radio could be brought out of sleep mode with any button. I plan to use the remote shown in the photo on the left, as it is more functional and has a separate power button. It would be great if you made it possible to turn the radio on and off by pressing this button once, like in any regular remote for any equipment. This would be a very logical solution for this type of remote and for all similar ones with a separate power button.
Then the following problem occurred, when editing the buttons of the remote in the browser, I don't have the new settings of the ir_buttons.json file saved. I have to manually transfer this file to SD-card for the remote to work correctly

It would make sense to add another button to the remote's "save to" settings page. This way it would be clearer; the "save" button saves the changed settings in the ir_buttons.json file, the "save to" button saves the changed or original settings to a file on the computer, and the "load" button loads the settings file from the computer.
By the way, the same algorithm could be applied to the station editing tab.
Thank you for your hard work, I'm trying to help make the radio even better

TedWillemsen, 2024-09-30:
Dear Wolle and Valera

My experience with the new remote control is a slightly diffent but let me tell:
It looked for me as IR codes when saved via the browser only keeps working until a reset or power on-off happens.
After a new start IR remote is not working anymore until you do again "something" in the browser.
Opening the browsers IR settings you will see all codes correctly filled in, except the address
After the first touch the address appears and the key's are active again.
Is far I understand the ir... json file is directly written to SD on each activity in the IR browser page 
Save and load actions (Blue buttons below) are done to and from the P.C. download directory.
Looking in the the Ir....json files on SD card as well on PC download, I do not see the address recorded.
Is it still in NVM ? I suppose the address is needed for correct (and selective) decoding of the remote control.

Kind regards,
Ted

clear-sky, 2024-09-30:
Hi Dear Wolle! & Dear Valera!
thanks for new release and many many thanks for add ir buttons!
is there any restriction to add ir buttons numbers? or is restricted up to 22 ???

TedWillemsen, 2024-09-30:
Dear Wolle,
For IR remote I am using a "programmable" RC that's sending NEC codes on address 0x18, till yesterday without problems and much more comfortable / accurate than the cheap chinese foil key versions. 
Altough I have also such one and found that it is operating on address 0x00. And with this address RC keeps working after reset / power off-on. In contradiction to my 0x18 RC that only works in the "powered" period after browser configuraion.
My conclusion so far, after start/reset the IR sofware is default listening to address 0x00 only. Other addresses though visible in the browser are not used. As mentioned before IR adress is not stored on SD or PC .

Kind regards,
Ted

clear-sky, 2024-10-01:
Hi Dear Ted!
mentioned this issue due to #468

valera39, 2024-10-01:
The remotes I mentioned above work fine at a distance of about 4 meters. I only wish that a short press of a separate off button could be used to turn the radio off. Hopefully Wolle will make this available.

schreibfaul1, 2024-10-01:
@TedWillemsen You are right, the IR_Address must of course also be saved, that was the case so far, I will extend the ir_buttons_json 
@valera39 I will add the "awake" function.

clear-sky, 2024-10-01:
Hi Dear Wolle!
the ir remote that you recimmand for this project is not standard and does not have main keys of a standard remote
as Ted said and me due to 
https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/405#issuecomment-2081399363
standard minimum keys for a good remote are 27 as below:
0
1
2
3
4
5
6
7
8
9
power
mute
up
down 
left 
right
ok
menu
exit
red
green
yellow 
blue

as far as i know most satellite receivers use nec protocol for their remotes and most of us have such ir remotes

TedWillemsen, 2024-10-01:
Dear Wolle and valera39

Nice that you understood the real problem: not able to save address, so only listening to 0x00.
Structure of the NEC protocol was known by me.
I made more IR related projects and for my own "support" I made a small multi IR protocol analyzer based on the arduino library IRremote.hpp.
[image]
If interested I can send more information.
I am looking forward to your solution for the address storage. 
One remark on the "awake" issue of valera39. This shoud work already by the "off timer" key (key function 16).
Probably some other IR remarks of valera39 are also solved by adding address storage
You are doing a great job with your quick solutions and in depth knowledge of your project.

Kind regards,
Ted

schreibfaul1, 2024-10-01:
Hi Ted, hi Valera,
the IR address (referred to as UserCode in some specifications) is now evaluated correctly. The address is now also stored in the ir_buttons.json. If the value of a digit or function key changes (enter), a message appears in the upper right corner and the it_buttons.json is rewritten.
Another button can be assigned to position 17 to wake up.
[image]

The yellow background colour only indicates that an IR command is assigned multiple times, but that doesn't matter (I don't have another button free)

valera39, 2024-10-01:
How to make it possible to switch the radio on and off with one wake-up button (position 17) as it is done for example for a TV set?

TedWillemsen, 2024-10-01:
Dear Wolle, dear Valera, dear clear-sky

It works ! Amazing again how quick Wolle has adapted  6 involved files. Herewith my new ir_buttons.json file and a picture of the used Remote control [ir_buttons.json](https://github.com/user-attachments/files/17216138/ir_buttons.json)
[image]
System select (Miniwebradio): DSAT, commands all sent with address 0x18
Buttons used for "special functions": 
Teletekst (above Red) > Mode (radio<>clock) 0x52 15, 
Red > Off Timer 0x5b 16.
Green > Awake 0x4d 17.
Mute long press > Sleep 0x09 20

Some remarks:
Sometimes (special on Mode or Off Timer key) it seems there is no action, then you have pressed too long.
Do a short press and be patient for 1 or 2 seconds.
On the number keys you have a more or less immidiate feedback, pressing longer gives half numbers on the upper screen part
The explanation of edititing the browser IR page (Yellow fields and message upper right corner) is also valid for previous versions. For Valera it is now clear that the SD is written on the moment the  upper right message appears ?

May be an issue: TFT brightness value is read from SD on start/reboot, when SD not present or not recognized, screen keeps dark so you do not see the problem message. Probably use a default (50%) brightness before reading the SD.
For me it's not an important issue, you can also use the serial monitor dump.

Thanks again Wolle, and all paicipants in this great project for your contributions
Kind regards
Ted

TedWillemsen, 2024-10-02:
Dear Valera, dear Wolle

On/off  (in this project called Awake/Sleep) can eyasy be done by editing the same command* on pos. 17 and pos. 20.
Long press > sleep, short press same key > awake. *I suppose the command belonging to your power/stby key.
If you want a real toggle function like short press > sleep, short press awake, that has to be done in src/main.cpp in the IR processing area actual starting from line 2918 onwards. I wonder if Wolle can be motivated to implement that kind of taylor made requested key functions. 

Remark: be aware that Browser IR adapdions are only SD saved after the upper right message and some funtion not active before a new reboot.

Kind regards,
Ted

valera39, 2024-10-03:
Dear Ted, dear Wolle
You make a good point, I need a true toggle function such as short press to sleep, short press to wake. I think it would be useful not only for me. So I hope that  Wolle will read this message and make this function.

valera39, 2024-10-12:
Hello Dear Wolle!
After the last update, I have a very strange reaction to waking up from sleep using the remote control buttons.
If you press the MODE button in sleep mode, this image appears and the radio does not play.
[image]
If you press the MODE button in sleep mode, this image appears and the radio does not play.
The next time you press the MODE button, a list of stations appears on the screen. Use the volume up and down buttons to select the highlighted station from the list. Press the left and right buttons to scroll through the list. The next press displays the player, DLNA and clock screens. 
The next press displays the player, DLNA and clock screens.
[image]
If you press the left or right buttons in sleep mode, this image is displayed and the radio is played. but there is no additional information
[image]
If the timer button is pressed in sleep mode, the normal full screen image appears.
Please make a normal function to turn the radio on and off (wake up), one button 
[image]
with a short press.
Now it turns out that you can only turn on the radio by pressing the timer button.
I hope for understanding, with best wishes!

schreibfaul1, 2024-10-13:
Hello Valera,
thank you for your contribution. This is really strange behaviour and situations that I haven't tested yet.
Later, everything should be controllable via remote control, including the selection of audio files and DLNA entries. I'll have a look at this in the near future.

valera39, 2024-10-13:
Thank you! For me, the most important thing is to wait for full on and off with one button on the remote. I already have a new enclosure ready for your next radio.

schreibfaul1, 2024-10-14:
Waking up is now possible with any function key (not with the number keys 0...9). I hope the strange effects have now disappeared.
The function keys do not yet work with the PLAYER and DLNA mode, that will come later.

clear-sky, 2024-10-14:
Hi Dear Wolle!
thanks for this update! with mode key on ir remote i can access to station list -arrow keys work well bot ok key does not switch to selected station

valera39, 2024-10-14:
Hello Wolle!
Thank you, the latest update is a success.
Apparently there are some difficulties in realizing the switching of "sleep" mode (on/off) by a short press of a separate button?
What is the function of the "channel" button?
And one more question please, do you have any ideas of connecting any other devices via I2C bus?

valera39, 2024-10-14:
[clear-sky](https://github.com/clear-sky), It's the same with me. Wolle may not have finalized all the button functions yet

clear-sky, 2024-10-15:
Hi Dear valera!
in my opinion time out for station list (in ir remote select)is not necessary at all!
what is your opinion?

valera39, 2024-10-15:
Hi Dear [clear-sky](https://github.com/clear-sky)
Perhaps Wolle has its own preferences and is not able to fulfill all our wishes, as much as we would like to.
In the simple remote control, or the one I posted, the button labels are the closest to the functions of this now not just a radio, but as I believe a music center.
Other remotes, however, may mislead some users with incorrect button labels.
 I think the perfect remote would be from an old music center. But this is my personal opinion.
As for the time-out list, I think it is necessary, because the user needs time to make up his mind. As an option you can offer a list of favorite stations for quick selection.
I use DeepL translator, I hope you will understand me.

clear-sky, 2024-10-15:
HI Dear Wolle!
I understand that you want to use a remote that is cheap and available to everyone
Suppose you are scrolling up and down in the station list menu, but the sound of the radio bothers you. You should use the up and down buttons to reduce the volume of the radio, but this does not happen, so you have to change the channels and increase and decrease the volume. The sound should be distinct from the up, down, left and right keys
I have a idea for giving such remote to every one you want!
You can add a plugin in the web UI section that does a special job
In the hardware part, the user temporarily connects two pins of mcu to an infrared transmitter LED, then by pressing the buttons that are considered in the web part, the infrared codes that you have already embedded in the browser are sent to the transmitter LED.
Now the user takes the universal control which has the ability to record the codes in front of the  transmitter LED and the desired codes are transferred to his device.
In the last step, the LED is separated from the hardware
With this method, you can transfer the codes of a remote control with NEC protocol with any number of keys
universal remotes are available everywhere --see one of these
[image]

samand587, 2024-11-06:
Please upload your file ir_buttons  for the left remote control

closed 2024-12-24 by github-actions[bot]