schreibfaul1/ESP32-MiniWebRadio #515: Ethernet

valera39, 2024-10-01:
Hello Dear Wolle!
If I understand correctly ESP32-audioI2S library allows connection of wired internet for example via modules like w5550 or similar.
Is it possible to use this module in your radio or can there be any difficulties in the execution of this idea?

schreibfaul1, 2024-10-03:
The big advantage of an Ethernet module would be the much more stable Internet connection with sufficient data throughput. With Arduino V3 and the NetworkClient that is now available, this is certainly not difficult to realise.
Disadvantage: Additional GPIOs would be required. This could only be realised for the ESP32-S3.

valera39, 2024-10-03:
It would be really great if you could do it for ESP32-S3, I think most users use this microcontroller. And for wrower, maybe a pin expander would help, for example on PCF8574 https://www.ebay.de/itm/401631118140?srsltid=AfmBOoot7hpQlCcgtmKDaTuLwbvwjXuqzKUfzGCLq0WcAa9NCsbAb9vG (I was looking for an example for your country). 
I will be ordering the w5550 module and look forward to adding wired internet to your wonderful radio.
Thanks!

schreibfaul1, 2024-10-03:
That is difficult to say. The audioI2S library now also works with Ethernet, LAN7820 and W5500. 
A port expander will be too slow. The ESP32 has reached its limits and is not suitable for further developments.  I can now buy an ESP32-S3 (N16R8) for €8, which is often cheaper than an ESP32 board with PSRAM.
Ethernet would be something for next year, then the ESP32 will make way for the P4 which has no WiFi.

TedWillemsen, 2024-10-03:
Dear Valera, dear Wolle

I found an internet-radio project with both WiFi and Ethernet using  ESP32-S3 and a W5500
https://github.com/thieu-b55/The---I-want-this---ESP32-S3-Internet-radio
With the lack of nice graphics, clock and RC possibilities I do't feel me invited to try-out this project.
But I am also absolutly interested in the possibility of extending Wolle's project with ethernet.
Due to speed limitations of I2C bus, a device like PCF8574 for pin expansion is not a good idea.
May be good for "slow" I/O levels like buttons, potmeters led's and also frequently used between small (Alphanumeric) LCD displays but not for SPI speed.
Forget the Ethernet limitations on ESP32. Save the cost of a PCF8574, for only a few €€'s more, move to ESP32-S3

Kind regards
Ted

clear-sky, 2024-10-07:
Hi all!
some chineese company try to bring back wifi to esp32-p4
[image]

schreibfaul1, 2024-10-07:
An antenna connection can be seen on the board. We will wait and see how things develop.

closed 2024-11-21 by github-actions[bot]

clear-sky, 2025-01-07:
Hello everyone
Apparently the  esp32-p4 board is ready for sale on Alibaba's website
I live in Iran and I can't buy anything from there because of the sanctions.
https://www.alibaba.com/product-detail/Hot-ESP32-P4-WT99P4C6-S1-development_1601168765556.html?spm=a2700.details.you_may_like.4.3d28c4e1bCGkYg
https://www.alibaba.com/product-detail/WT0132P4-A1-ESP32-P4-SOM-Board_1601338078620.html?spm=a2700.galleryofferlist.normal_offer.d_image.1a9713a0tE1XT6
https://www.alibaba.com/product-detail/WT0132P4-A1-ESP32-P4-SOM-board_1601293888639.html?spm=a2700.galleryofferlist.normal_offer.d_title.1a9713a0Jv3kgG&selectedCarrierCode=SEMI_MANAGED_STANDARD@@STANDARD

schreibfaul1, 2025-01-08:
At the moment, the ESP32-P4 is not easy to obtain everywhere. Some boards have an Ethernet module because the P4 does not support WiFi. Other boards use an extra ESP32-C6 for WiFi. There are also P4 boards with an HMI display (800x480px). I think we'll wait and see how things develop.

clear-sky, 2025-01-08:
Hallo Dear Wolle!
I agree with you! specially well known companies like espressif and ai-thinker haven't introduced their own modules yet.