schreibfaul1/ESP32-MiniWebRadio #516: Only keys 1 to 9 work on the remote control.

hevet, 2024-10-02:
Only keys 1 to 9 work on the remote. All keys are assigned correctly. When I press the left arrow, for example, it detects its address and I can assign it to "left". Unfortunately, nothing else works except keys 1 to 9. Before these changes, the remote worked correctly. Is there anything else that needs to be changed somewhere? The remote on another radio also works correctly, so I rule out its damage.

TedWillemsen, 2024-10-03:
Dear Hevet

This can probably be related to issue #513
Two day's ago Wolle made a correction because in the IR adres visible and edited in the browser was not properly saved to SD and PC. 

Kind regards,
Ted

hevet, 2024-10-04:
I have installed the latest version with this fix, the address is correct, also on the card, but only keys 1 to 9 work. I use the remote control as shown in that post.

schreibfaul1, 2024-10-04:
Hi hevet, I can't reproduce the error. From now on the ir_json is updated more often, even if the value is written when clicking on an input field. Can you please check the following: please load the ir_buttons.json on your PC [image] 
and check with a text editor if the values in the file match the values in the table.

hevet, 2024-10-04:
@schreibfaul1 I checked and it looks ok
[ir_buttons.json](https://github.com/user-attachments/files/17256214/ir_buttons.json)
[image]

TedWillemsen, 2024-10-04:
Dear Hevet, dear Wolle,

IR values you read from the browser screen and from SD seems to be  equal to the default values given in the  ir_buttons.json from a **fresh** SD content list. 
You have to check/edit the browser IR overvieuw, key by key* and when all correspond with the recieved values in **C** do some action that generates a message in the upper right corner. At that moment the new values (belonging to **your specific RC**) are stored on SD. Clicking on the blue **save** field below wil write the actual showed values only to the json file on your PC.

*I expect only the 0-9 are already corresponding.

Looking forward to hear if this was helpfull
KInd regards,
Ted

TedWillemsen, 2024-10-04:
Dear Hevet,

Still one remark: 
After editing IR values you have to restart the radio to read new values from SD en being effective.

Kind regards,
Ted

TedWillemsen, 2024-10-06:
Dear Hevet,
What command code did you read in field C assinged to the key you intend to use as "awake" ?
Write down the codes from the keys not working and edit them manually in the SD Ir buttons json file, for example by a card reader.  The structure whitin the file is supposed to be self explaining. Save when ready and put it back in radio. Would that help ?

hevet, 2024-10-07:
@TedWillemsen  Awake I have it set to -1 because there are no free keys on the remote control.

hevet, 2024-10-08:
@schreibfaul1 , @TedWillemsen I found a solution, it is very simple, you just need to upload the latest version of the library to use the remote control. The remote control works immediately with default settings.

TedWillemsen, 2024-10-08:
@schreibfaul1 ,@hevet
Because I did a complete new installation, including all libraries, I could not reproduce this issue.

Kind regards,
Ted

closed 2024-10-08 by hevet