schreibfaul1/ESP32-MiniWebRadio #519: Radio Station Icon on Clock Screen

skassel2024, 2024-10-08:
Hallo Wolle,
in der Uhranzeige erscheint oben links das Stations Icon, wenn ich über die Weboberfläche ein Radiostation auswähle.
Auch kann ich nicht zurück zur Radioanzeige, PLayer oder DNLA. Im Monitor erscheint "unknown HTMLcommand change_state, param=RADIO (PLAYER/DNLA)". Nur am Radio direkt komme ich wieder zurück. Ist das so gewollt?

ESP32 Chip: ESP32-S3
Arduino Version: 3.1.0
ESP-IDF Version: 5.3.0
MiniWebRadio     Version 3.5b - Oct 06/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 16777216 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 40 MHz
TFT speed 40 MHz
PSRAM total size: 6015156 bytes

Für zukünftige Versionen wäre eine Batterieanzeige, wie "FranZatGit" es in seinem Projekt erwähnt hat schön.
Gruß Sören
[image]

schreibfaul1, 2024-10-09:
Hallo Sören,
das ist absolut nicht so gewollt. Ich sehe nicht immer alles, daher danke für deinen Hinweis. Ziel ist es, dass später die Anzeige auf der Webseite synchron zum Display ist und viele Funktionen mit der Fernbedienung  verwendbar sind. Die Senderliste geht bereits.
Die Batterieanzeige könnte ich machen. Aber wie groß soll die Batterie sein? Stromsparend ist das Projekt nicht.
viele Grüße
 Wolle

skassel2024, 2024-10-09:
Hallo Wolle, absolut tolles Projekt, Respekt!
Ich betreibe das Webradio zur Zeit mit  3xAA Ni-MH ca. 3h. 
Fernbedienung und BT habe ich noch nicht integriert, ist aber geplant.
Gruß Sören

AVRsteffen, 2024-10-10:
Ich betreibe das Radio mit einer 4-Zellen Varta LiIon Batterie mit integrierten BMS + Gauge + Schnittstelle zum auslesen aller Parameter der Batterie. Bei mir spielt das Radio locker 20 Stunden.

AVRsteffen, 2024-10-10:
Here two pic of Radio and Battery management
[image]
[image]

skassel2024, 2024-10-10:
Hallo Wolle,

in der aktuellen V 3.5d kann ich nun aus der Weboberfläche zurück zur Radioanzeige, PLayer oder DNLA, und "unknown HTMLcommand change_state, param=RADIO (PLAYER/DNLA)" erscheint nicht mehr im Monitor.

Das Stations Icon, wenn ich über die Weboberfläche ein Radiostation auswähle, ersheint aber immer noch in der Uhranzeige.

Sag Bescheid, wenn ich Dich nerve, aber ich denke das hilft , das Radio zu perfektionieren.
Gruß Sören

schreibfaul1, 2024-10-10:
Hallo Sören,
jetz weiß ich, was du meinst. Wenn im CLOCK Modus über die Weboberfläche die Station gewechselt wird tritt der Fehler auf. Ich hatte das erst nicht so verstanden. Der Fehler ist jetzt beseitigt.

closed 2024-11-24 by github-actions[bot]