schreibfaul1/ESP32-MiniWebRadio #521: ESP32 Wrover Compilation error

samand587, 2024-10-12:
I get an error when compiling: 
[image]

samand587, 2024-10-12:
Hi! What can I see?

samand587, 2024-10-12:
Hi! How can I fix this?

schreibfaul1, 2024-10-12:
I am not aware of the error. I suspect there are outdated files somewhere. Maybe "clear all" will help to recompile everything.
[image]
If it does not help, search for the .platformio folder on your PC (home directory in Linux, Windows??)
and delete the packages and platforms folders
[image]
Don't worry, the folders will be recreated.

samand587, 2024-10-13:
Hi! This is the error I get. How can I fix this?
[image]

schreibfaul1, 2024-10-13:
I can't answer that question, I've never had it.
Have you selected the correct environment? Then the toolchain should automatically create `flash_partitions.c.o` during compilation. You can then find this in the directory tree. You can also delete the build directory, it will be recreated during compilation. 
[image]

samand587, 2024-10-13:
What should I mark here for ESP32 Wrover?  Thanks! 
[image]

schreibfaul1, 2024-10-13:
If the error is fixed, if not you should look in the VS/PIO/Python area.

The settings in platformio.ini depend on your board. If the manufacturer specifies 4MB, you do not need to change anything. The size of the PSRAM is recognised automatically.

samand587, 2024-10-14:
Hi! Another mistake. How can I fix this?
[image]

schreibfaul1, 2024-10-14:
I have no idea. Over the years, countless users have cloned the project from Github and not received these errors. Maybe you should reinstall Git.
Here are the instructions again:
https://github.com/schreibfaul1/ESP32-MiniWebRadio/blob/master/docs/How%20to%20install.pdf 

If anyone reading here has had similar problems, please comment here.

samand587, 2024-10-14:
We managed to build the project on a laptop. Thanks!

samand587, 2024-10-14:
I connected it and got this . What can I see?
[image]

schreibfaul1, 2024-10-14:
You have selected a display with too high a resolution. Only the displays listed in #define TFT_CONTROLLER can be selected.
[image]

samand587, 2024-10-14:
My display   2,8" TFT 240xRGBx320  V1.1

samand587, 2024-10-15:
Вопрос решен! Спасибо автору!

closed 2024-10-15 by samand587