schreibfaul1/ESP32-MiniWebRadio #522: displaying the volume level on the screen

valera39, 2024-10-14:
Hello Dear Wolle!
Is it possible to make it so that when the volume buttons on the remote control are pressed, the display briefly shows the digital volume level? This way it would be visually apparent that the volume is increasing or decreasing. This is especially useful at low volume levels and when the remote control is pointed precisely at the receiver.
Do you think this would be useful?

valera39, 2024-10-21:
Dear Wolle, please note my question

schreibfaul1, 2024-10-22:
I have read your contribution. I'm still missing the right idea for realising it. I'll have a solution soon.

schreibfaul1, 2024-10-22:
With the current version, volume changes via the remote control are briefly displayed as follows:
 [image]
For this to work, the png and digits_small folders on the SD card must also be updated.

valera39, 2024-10-22:
That's great! Even better than I thought it would be. 
Thank you!

skassel2024, 2024-10-26:
Hi Wolle, Hi valera39
nur so als Anregung/Idee. Zeilen 2933 und 2938 in main.cpp eingefügt, um beim drücken del Vol+/Vol- Taste das Radio Submenue mit dem Lautstärkebalken aufzurufen. (funktioniert so aber nur im RADIO-Modus)
[image]
[image]

schreibfaul1, 2024-10-27:
Das ist eine gute Idee. Für RADIO und PLAYER haben wir die Einblendung mittels der Siebensegmentanzeige. Aber das lässt sich nicht überall so machen. In CLOCK wird jetzt der Lautstärkebalken angezeigt.
[image]

Mal außerhalb des Themas:
Auf großen Online-Plattformen wie Amazon oder Ali wird diese Fernbedienung zum kleinen Preis angeboten.
Ich vermute, dass diese FB weltweit beschaffbar sein wird.
[image]
Die Bezeichnung ist _ir Controller mit kd-Funktion kompatibel für x 96 mini x 96 x96w android tv box_ 
Vorteil: genau die Funktionstasten, die wir benötigen und größere Reichweite als die klassische Android Fernbedienung.
Damit lassen sich weitere Funktionen realisieren, wir im Titel springen, Home, Back, Pause/Resume, Stop ...

closed 2024-12-11 by github-actions[bot]