schreibfaul1/ESP32-MiniWebRadio #523: corrupted sound on some stations

clear-sky, 2024-10-14:
Hi dear Wolle!
in new update some stations have corrupted audio !
please check this stream
http://uk3.internet-radio.com:8405/live

clear-sky, 2024-10-15:
Hi Dear Valera!
can you compare audio quality in radio with your pc?

valera39, 2024-10-15:
Hello, [clear-sky](https://github.com/clear-sky), It's hard for me to judge the sound quality. because of different amplifiers and speakers, but my opinion is that the sound is ok.

clear-sky, 2024-10-15:
Hi Dear valera!
thanks for the answer-On my computer, the sound is noticeably better than the radio
It seems that there is a lot of disconnection and connection on the radio,If I want to explain better, the sound is scratchy
 let's see what is Mr. Wolle opinion

clear-sky, 2024-11-12:
Hi Dear Wolle!
 did you test this stream? (http://uk3.internet-radio.com:8405/live)

schreibfaul1, 2024-11-12:
The link works quite well for me. Do you have the possibility to send a ping to uk3.internet-radio.com. My guess is that the response time is greater than ~120ms, which is what it looks like for me:
[image]

Or TCP packets may get lost?

clear-sky, 2024-11-13:
Hi Dear Wolle!
Thanks for the hint! this is my ping result:

[image]

but there is a odd problem! this stream works quite well in my pc!!!!!

closed 2024-11-13 by clear-sky

schreibfaul1, 2024-11-14:
It's the same with me. With an Ethernet adapter, it also works under these conditions. It's clearly the WiFi. But I'm not very familiar with the WiFi settings.