schreibfaul1/ESP32-MiniWebRadio #524: Station problem causing disabling BT Button.

ve2cfb, 2024-10-15:
Hello Wolle,
Since Version 3.5e - Oct 10/2024 i cannot get in the BT. Previous version was ok.
Here is a picture of the failling menu + couple of files. The BT button is Violet (at least on my Display) and cannot access the next BT menu.
If i copy the Content_on_SD_Card.zip from your site, it works ok.
Did a format on a 32G and a 256G but still a problem.
Did a Erase Flash and Upload but still having problem.

Today, i found it.
One Station is causing the problem.
"Radio Deluxe Easy"
Have a look in **settings.json** you will see the problem for station 5.
By deleting "**settings.json** and deleting station No 5 (Radio Deluxe Easy) in **stations.json**
Everything works ok.

Re-installing Radio Deluxe Easy from the Web Interface, i get the same problem.

[settings.json](https://github.com/user-attachments/files/17377471/settings.json)
[stations.json](https://github.com/user-attachments/files/17377472/stations.json)
[image]

Thanks for your time...

valera39, 2024-10-15:
Confirmed, bluetooth is a problem.

schreibfaul1, 2024-10-16:
Hi Pierre,
I don't know what's causing the problem yet, but it's not the URL of "DELUXE MUSIC"
[image]

https://audiostreams.deluxemusic.tv/easy/mp3-128/audiostreams.deluxemusic.tv/
will be forwarded to
https://regiocast.streamabc.net/regc-deluxeeasy7122651-mp3-128-6100697?sABC=670so613%230%232p6q89s548q982rn9067ps93ro5r4042%23nhqvbfgernzf.qryhkrzhfvp.gi&aw_0_1st.playerid=audiostreams.deluxemusic.tv&amsparams=playerid:audiostreams.deluxemusic.tv;skey:1729082899

that's all right so far
If I understand correctly, the BT module is not recognised, I'll have a look at that soon.

ve2cfb, 2024-10-16:
Hello Wolle,
Yes i agree with you comment.
but here how the problem happen.

Here is the sequence you need to do.
you do not need Web interface.
-  From PC  delete settings.json on the SD card
- erase flash and upload
- Power off the Radio
- insert sdcard in Radio
- Power on Radio
- At this point, everything work ok.
- Accessing any stations will work.
- Able to access BT 1st Menu BT button
- Next, power off the Radio. 
-  wait 5-10 seconds before Power on. if < 2 seconds, problem will not happen.
- select any station from the list except station 5
- you should be able to access the BT menu from the radio, not from the web.(the web will give you different result)
- select station 5 from the list.
- try to access the BT menu. It should be a grayblue button.
- Will not be able to access the 2nd BT menu.
=======================================================

ve2cfb, 2024-10-16:
Wolle,
Thanks very much. I will do a full clean.

ve2cfb, 2024-10-19:
Hello again Wolle,
Same problem as before.
did a full clean and a erase flash and upload.
By the way, after a full clean, i get the following error when trying to do a erase flash and upload.
[Error Platformio.pdf](https://github.com/user-attachments/files/17446662/Error.Platformio.pdf)
This only happen one time.

Back to my problem:
Let me summarize the problem on a very simple test..
By using your latest version of stations.json and settings.json 
**everything works OK.**
but if i insert the following station in stations.json: 
**["*","D","Radio Deluxe Easy","https://audiostreams.deluxemusic.tv/easy/mp3-128/audiostreams.deluxemusic.tv/"]**
I do another erase flash and upload.
dont forget to do a POWER OFF.
and the problem is back.

if i remove that station from the stations.json, no more problem.

What am i doing wrong...?

Many thanks for your time.

schreibfaul1, 2024-10-19:
Hallo,

_Same problem as before.
did a full clean and a erase flash and upload.
By the way, after a full clean, i get the following error when trying to do a erase flash and upload.
[Error Platformio.pdf](https://github.com/user-attachments/files/17446662/Error.Platformio.pdf)
This only happen one time._
this happens when CMAKE is running before PIO has finished updating. I sometimes have this too, when the compilation is started a second time it works.

I have not yet been able to reproduce the actual problem with "RadioDeluxe Easy".
You do not need to edit the stations.json. Simply search for it and apply it using the two buttons.
[image]
The station always appears in the last column of the table.
[image]
And then you can cut the line and paste it into the desired position. The stations.json will then be changed automatically.

ve2cfb, 2024-10-21:
Hello Wolle,
Still a problem(only once and could not reproduce it because of the following problem) and another one with the Web Browser.

When trying your procedure to Access by COUNTRY and GERMANY, no list appear.
but if i choose any other country, no problem. 
Why a problem with Germany...?
Also when trying to access the home web page, It takes a lot of time to get the page and on the radio, the
LEVEL METER freeze up until web page is available. 
I disable all security, anti virus and so on.
Did a complete re install of vscode and platformio but still problems.

Use my neighbor Router and go rid of the Web Problem.
Now Germany works ok and did a search on deluxe and did your procedure and got the same error for the disabled BT button.

[image]

schreibfaul1, 2024-11-04:
Hello Pierre,
you've probably sent me the wrong zip.

On the page https://ttsmp3.com/ you can create mp3s
[image]

ve2cfb, 2024-11-04:
Sorry, my mistake.

[voice_time_Fr.zip](https://github.com/user-attachments/files/17623494/voice_time_Fr.zip)

schreibfaul1, 2024-11-06:
I have implemented it. This makes it easy to add other European languages later.
[image]

closed 2024-12-24 by github-actions[bot]