schreibfaul1/ESP32-MiniWebRadio #525: Playlist

samand587, 2024-10-15:
The playlist has been compiled, everything is working. I go into the browser, click "files" and my sheet is erased, a new sheet with three radios appears. Why is that?

schreibfaul1, 2024-10-16:
I assume you mean this table: 
[image]
This means that the file `stations.json` is corrupt or no longer exists. In this case, a standard table with three entries is created.
Errors cannot be completely ruled out. If you have created your own channel list, please save it as a backup on your PC using the blue button.

samand587, 2024-10-16:
Everything goes fine if you create a list of stations in the browser. But I made my list and inserted it into the SD card. It is not saved.

closed 2024-11-16 by samand587