schreibfaul1/ESP32-MiniWebRadio #526: Remote control

samand587, 2024-10-16:
I want to set up such a remote control, commands are not recorded, the test does not pass. What can I see?
[image]

schreibfaul1, 2024-10-16:
You do not need to do anything with the "Arduino" remote control. With the one in the photo, the commands will not match and you will have to reassign the buttons. If the value in the input field changes, the file ir_buttons.json should be rewritten. Unfortunately, this does not work with all browsers. Please enter the last assignment manually and confirm with ENTER. The web server then triggers a request and you can see in the terminal how the file is written. 
It looks like this:
[image]

samand587, 2024-10-16:
I'll give up the remote control. There will only be a TOUCH.

valera39, 2024-10-16:
У меня такой же, работает отлично

samand587, 2024-10-16:
Could you share the file ir_buttons ?

clear-sky, 2024-10-16:
When you buy these remotes, a piece of plastic is placed under the battery to prevent the battery from draining. Did you remove it?

samand587, 2024-10-17:
Yes, the remote control is working, it was used in another project.

samand587, 2024-11-16:
The question is very relevant, but they didn't answer me!

schreibfaul1, 2024-11-16:
[image]
**Prerequisite: The remote control must use the NEC protocol.**
When you press a button, the value in the _IR command_ field changes.
Then you assign the value to a function, e.g. if you have pressed "0", you assign it to ZERO. And you do this with all buttons.

samand587, 2024-11-16:
Thanks! Now everything is clear.

valera39, 2024-11-16:
I have the radio next to the TV and some of the buttons on the TV remote wake up the radio. How to eliminate this. I think a full implementation of the mode of switching the radio on and off with only one button and not several buttons for waking up would help here

schreibfaul1, 2024-11-16:
This is still an open problem. Each remote control always transmits an address code and a command.
[image]
I have not yet analysed the address code; this will be done soon with additional function keys.
However, if your remote control uses the same address code as the TV, there will always be interference.
The Arduino RC has the address code 0x00
[image]
The Android RC X96 (much better than a membrane keyboard) has the address code 0x01
[image]

valera39, 2024-11-16:
Android RC X96 also causes interference, but to a lesser extent. Explain please, I don't understand why you don't want to use one button to wake and sleep the radio. Is it very difficult to implement? In this case it would be an excellent, albeit partial, solution to the problem

schreibfaul1, 2024-11-16:
Yes, now, if the address code is evaluated and the TV has the address code 0x00, for example, then no longer.

valera39, 2024-11-16:
I wouldn't want to change my TV with a different remote :)
So what about the one button?

valera39, 2024-11-17:
Back to the wish of putting the radio to sleep and waking up with one button. This option would drastically reduce the probability of interference from another remote control, because now the radio is also awakened by other function keys, which leads to false activations from another remote control. 

The option with one button is extremely important for me, I hope not only for me.

Dear Wolle, please answer me specifically: do you realize this function of switching on/off the radio, which is logical for all devices with remote control, or this function will never be in the project.

I really hope for your attention and answer!

schreibfaul1, 2024-11-17:
Hello valera39,
the remote controls will no longer influence each other if they have different address codes. The IR library must be reloaded for this.
I have also created more space to add further functions. Also an on/off button if this is available.

schreibfaul1, 2024-11-17:
I had to correct another error (the IR array was too small), and now there is an on/off function.
[image]

valera39, 2024-11-18:
Hello Dear Wolle!
Thank you so much for the separate off button! But I discovered a problem: the radio turns on and off with one button perfectly. But if you press any button in the off state, for example, up, then the screen appears on the static image.
[image]
The image can be different, depending on the command given.
Is it possible to fix this, so that in the off state the radio would not react to other commands of the remote control?

schreibfaul1, 2024-11-18:
The error has been eliminated and further IR functions are activated.

clear-sky, 2024-11-18:
Thanks Dear Wolle!
for expanding IR functions

schreibfaul1, 2024-11-19:
Thanks, the aim is to be able to control most of the functions with the remote control, but that will take some time.

samand587, 2024-12-21:
Thanks for the answers. The issue is closed!

closed 2024-12-21 by samand587