schreibfaul1/ESP32-MiniWebRadio #527: setSocketOption(): fail on 0, errno: 9, "Bad file number"

Jimbobnz, 2024-10-16:
What do I need to do to fix this?

`08:53:44 StationName:
08:53:44 AUDIO_info:  buffers freed, free Heap: 142575 bytes
08:53:44 AUDIO_info:  connect to: "digitalstreams-edge-lb.mediaworks.nz" on port 443 path "/edge_net/playlist.m3u8"
08:53:44 AUDIO_info:  SSL has been established in 546 ms, free Heap: 142547 bytes
08:53:44 lastURL: ..  https://digitalstreams-edge-lb.mediaworks.nz/edge_net/playlist.m3u8
08:53:45 AUDIO_info:  connect to: "https://digitalstreams-edge-lb.mediaworks.nz/edge_net/playlist.m3u8?listeningSessionID=66ea2c7b0168ff6f_2489181_MNwY5eX7_MTcyLjIzLjE2LjYy_0000000sNDn&downloadSessionID=0"
**[718741][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"**
08:53:45 AUDIO_info:  The host has disconnected, reconnecting
08:53:47 AUDIO_info:  media sequence number: 216138991
08:53:47 StreamTitle: Eminem & Jelly Roll - Somebody Save Me - Eminem & Jelly Roll - Somebody Save Me
08:53:47 AUDIO_info:  connect to: "https://digitalstreams-master-lb-cf.mediaworks.nz/iZLupWgkWAkp-216138991-8000.aac"
08:53:47 StreamTitle: **HTTP/1.1 404 Not Found**
08:53:47 StationName:
08:53:47 StationName:`

schreibfaul1, 2024-10-18:
With `Accept-Encoding: identity;q=1,*;q=0` I inform the server that I want to receive unpacked data packets. The server then sends HTTP 404. 
With `gzip` the server would send me data packages that I cannot unpack fast enough on the ESP32.
https://digitalstreams-edge-lb.mediaworks.nz/edge_net/playlist.m3u8 is one of the stations that does not work here

Jimbobnz, 2024-11-18:
Could there be a code enhancement better handle the 404 http as it locks up the web interface.

closed 2025-01-04 by github-actions[bot]