schreibfaul1/ESP32-MiniWebRadio #528: Problem mit dem TFT 9488

bodo328, 2024-10-20:
Hallo zusammen 
erstmal ein Daumen hoch für die tolle Arbeit.
Ich habe mir das Radio nachgebaut mit einem 2,8 Zoll TFT das läuft auch wunderbar. mit dem alter werden die Augen schlechter und ich habe mir ein 3,5 zoll Display ILI 9488 gekauft und angeschlossen. Die Pinbelegung ist ja identisch mit dem 9341 Display, also Displays getauscht und Programm auf das neue Display eingestellt und Hochgeladen.
Leider bleibt das Display dunkel grau wobei die Touchfunktion gegeben ist.
Das Display ist in Ordnung an einem anderem Esp32 dev Modul geht es mit der Bodmer's TFT_eSPI.

Welche einstellungen muss ich ändern damit es Funktioniert? In der Common.h ist das Display ILI 9488 angewählt also Typ 4

Ich bitte um Hilfe

bodo328, 2024-10-20:
Habe ich glatt vergessen das ganze ist am ESP32 S3 angeschlossen

schreibfaul1, 2024-10-20:
Manche dieser Displays haben einen Designfehler, [Bodmer eSPI](https://github.com/Bodmer/TFT_eSPI/discussions/898)
Du kannst einfach MISO des Displays unbeschaltet lassen https://github.com/schreibfaul1/ESP32-MiniWebRadio/blob/master/docs/ILI9488_pins.jpg , dann funktioniert 
 immer noch alles außer Hardcopies, und ich glaude das Display kann max 40MHz
Das beste Display ist nach meiner Einschätzung das 4" ST7796.

bodo328, 2024-10-20:
Problem gelöst

Es lag an der Verdrahtung, ich habe gelesen das man bei einigen Modellen die Verbindung zum TFT MISO nicht benötigt wird und Fehler verursacht. Was hab ich gemach ich habe die Verbindung ganz weggelassen. 
Auf jeden Fall Funktioniert es jetzt. Werde mal das große 4" Display ausprobieren. Es ist schon erstaunlich was so ein halbes Zoll ausmacht es ist schon viel angenehmer damit zu arbeiten.
Vielen Danke Wolle

closed 2024-10-20 by bodo328