schreibfaul1/ESP32-MiniWebRadio #529: Sleep Modus IR Button

skassel2024, 2024-10-22:
Hallo Wolle,
habe da noch ein Problem mit dem "falling_asleep". Mir ist nich klar, woher die long pressed key nr: "29" kommt. das sollte doch eigentlich "20" sein. Ich habe mir dadurch geholfen, dass ich in main.cpp  key==20 durch key==29 ersetzt habe. Aber alle Änderungen, welche ich selbst vorgenommen habe, sind beim nächstern Update wieder verloren. So habe ich auch Out 05 (AMP_ENABLED) als LED Standby im sleep Modus verwendet. (siehe Bilder im Anhang, Funktion muss ich noch umkehren).
Mein Radio habe ich in ein ausgedientes Gehäuse (Ipod-Docking) von eyeTEC eingebaut und läuft mit dem dazugehörigen Röhrenverstärker wunderbar. (ESP trifft Röhre)
Gruß Sören
[image]
[image]
[image]

schreibfaul1, 2024-10-23:
Hallo Sören,
ein tolles Gerät hast du gebaut, das würde ich niemals schaffen. :-))

Nun zum Problem:
IR_button  29 gibt es (eigentlich) nicht. Was steht in der ir_buttons.json? Ist es da schon falsch?
Die json wird beim setup() in defaultsettings() geladen. Dafür gib es eine eigene Klasse.
Kannst du mal bitte in der common.h die beiden auskommentierten Zeilen aktivieren.
[image]

das Ergebnis sollte nach einem Neustart etwa so aussehen:
[image]

Das wäre die Zuordnung der Tasten für die IR Bibliothek. 
Ich möchte gerne wissen, wo das Problem liegt.

skassel2024, 2024-10-23:
Hallo Wolle,
danke für die schnelle Rückmeldung, dafür kann ich nicht programmieren ;-) 
Gruß Sören

Der Inhalt von ir_buttons.json:

[{"A":"0x0e","label":"IR address"},
{"C":"0x14","label":"IR command"},
{"0":"-1","label":"ZERO"},
{"10":"0x14","label":"MUTE"},
{"20":"0x00","label":"SLEEP"},
{"1":"-1","label":"ONE"},
{"11":"0x0d","label":"ARROW RIGHT"},
{"21":"-1","label":"CANCEL"},
{"2":"-1","label":"TWO"},
{"12":"0x0c","label":"ARROW LEFT"},
{"22":"-1","label":"-"},
{"3":"-1","label":"THREE"},
{"13":"0x0a","label":"ARROW DOWN"},
{"4":"-1","label":"FOUR"},
{"14":"0x09","label":"ARROW UP"},
{"5":"-1","label":"FIVE"},
{"15":"0x10","label":"MODE"},
{"6":"-1","label":"SIX"},
{"16":"0x19","label":"OK"},
{"7":"-1","label":"SEVEN"},
{"17":"0x00","label":"WAKE UP"},
{"8":"-1","label":"EIGHT"},
{"18":"-1","label":"-"},
{"9":"-1","label":"NINE"},
{"19":"-1","label":"-"}]
 
die Ausgabe nach Neustart:
[image]
[image]

hier noch ein Bild mit dem zugehörigen Röhrenverstärker
[image]

schreibfaul1, 2024-10-23:
Das Bild sieht stark aus., einfach perfekt. Das würde ich gerne auf der Seite der selbstgebauten Radios veröffentlichen, wenn ich darf. Ich habe früher selbst Röhrenradios gebaut, ich glaube die Anzeigeröhre ist eine EM84 und davor vermutlich zwei Trioden.

Was mir aufgefalleen ist, die Hardcopy der IR-Webseite passt nicht zur ir_buttons.json. Ich habe das "WAKE UP" weil eine Doppelbelegung der Tasten nicht gut funktioniert hat. WakeUp ist jetzt jede Funktionstaste. Du hast die Zifferntasten nicht zugewiesen. Das ist nicht das Problem, aber ich kann mir vorstellen, dass es auf deiner Fernbedienung Ziffern gibt. 

Bitte führe einen "Full Clean" aus 
[image]

damit die IR-Lib neu geladen wird.
Und dann weise die Tasten neu zu. Also auf der IR-Webseite eine Taste drücken und dann das entsprechende Input-Feld anklicken, bis alle Tasten neu zugewiesen sond. Die ir_buttons.json wird automatisch neu erzeugt. Keine Ahnung ob es hilft, aber der Fehler deutet darauf hin, dass er in der IR-Library liegt

skassel2024, 2024-10-23:
Hallo Wolle,
gerne kannst Du die Bilder verwenden. Das "Magische Auge" ist eine 6E2 (EM84), die beiden Doppeltrioden 6N1 (ECC85) 
Full Clean hatte ich schon ausgeführt und neu hoch geladen. Also ich finde keinen Unterschied von ir_buttons.json zu Hardcopy Website, ausser dem Label WAKE UP. Die Doppelbelegung für SLEEP 0x00 und WAKE UP 0x00 funktioniert einwandfrei. Die eyeTEC Fernbedienung hat keine Zifferntasten. Ansonsten immer noch bei IR address 0x0e, IR command 0x00 long pressed key nr: 29. 
In deinem Beispiel "ESP32-IR-Remote-Control" geht der Index der Buttons von 0-29, wobei nur 0-20 verwendet werden. 
Gruß Sören

schreibfaul1, 2024-10-24:
Hallo Sören,
die Bilder sind hier veröffentlicht https://github.com/schreibfaul1/ESP32-MiniWebRadio/wiki/User-devices

Ich glaube, das Problem ist jetzt gelöst. Bisher wurden alle Tasten mit 0x00 initialisiert. Da deine Fernbedienung aber das Kommando 0x00 für eine Taste verwendet gibt es davon mehrere und die letzte wird ausgewählt. Damit das nicht mehr passiert werden jetzt alle Tasten mit -1 (0xFFFF) initialisiert.
Meine Fernbedienung benutzt 0x00 nicht, daher habe ich den Fehler nicht erkannt.

beste Grüße
Wolle

skassel2024, 2024-10-24:
Hallo Wolle,
danke der Ehre mit den Bildern.
Der "Fehler" scheint behoben zu sein. Hast Du dazu Änderungen in der IR Library vorgenommen, da dort alle Buttons mit 0xFFFF vordefiniert sind?
[image]
Habe auch key==29 wieder auf key==20 geändert.
[image]

Gruß Sören

schreibfaul1, 2024-10-24:
Prima, dann war meine Vermutung richtig. Alle Tasten mit 0 zu initialisieren, auch die unbenutzten, ist nicht gut wenn 0 ein gültiges IR Kommando sein kann.
Dann kann ich das Thema jetzt schließen.
vG Wolle

closed 2024-10-24 by schreibfaul1